tsduck/tsduck #69: "pcr-smooth" plugin

lars18th, 2018-06-22:
Hi,

After read the wiki about MPTS muxing:
https://github.com/tsduck/tsduck/wiki/Muxing-use-case:-create-live-streams-from-files
I noted that the problem with the PCR jitter can be resolved with a pcr-smooth plugin.

What's the problem? The problem is when the incoming stream have inconsistent PCR timestamps, because a pid filtering or imprecise encoding muxing.

So, one solution is to do some _PCR Smooth_ processing. This process consists on:
- Load the stream in a de-jitter buffer.
- Search for PCRs inside the buffer and consume the packets between two consecutive PCR marks based on the interval of these marks.

This processing will reconstructes the original bitrate of the stream based on time.

See here for an explanation of how Cisco's DCM handlens this:
https://www.cisco.com/c/en/us/support/docs/video/digital-content-manager-dcm-software/200167-Troubleshoot-DCM-PCR-Dejitter-Buffer-Res.html

What you think?

lars18th, 2018-06-22:
Hi,

I feel this patch for FFMPEG can resolve the problem with non CBR muxing:
https://ffmpeg.org/pipermail/ffmpeg-devel/2016-March/192040.html

So, you can remove the parameter `-muxrate` (in FFMPEG) when generating the program prior to use the "mux" TSDuck plugin.

lelegard, 2018-06-22:
Which problem are you trying to solve precisely ?

lars18th, 2018-06-22:
Hi @lelegard ,

Don't worry at time! I'm doing some testing with MUXING multiple VBR streams using other tools and TSDuck at the same time. I'll do a more tests with the "mux" plugin, and the "pcrbitrate" that I feel is the module that I'm demanding.

Thank you for your support!

closed 2018-06-25 by lars18th