tsduck/tsduck #70: Support for DTA-2137?

rezna11, 2018-06-22:
Hello Thierry. Do you think DTA-2137 would work with TSDuck as both dual DVB-S2 demodulator and dual ASI output? It has DTAPI support but TSDuck manual only mentions DTA-1x devices.

lelegard, 2018-06-22:
Hi Jan,

All Dektec devices haven't been tested with TSDuck, of course. That being said, the Dektec software API is extremely consistent. By experience, if a function works on one Dektec device, it will work on another device the same way.

So, we should rather think in terms of _functions_ (ASI input, ASI output, DVB-S output, etc) rather than _devices_.

I just had a look at the DTA-2137 on the Dektec site. The 2 ASI output should work with TSDuck. ASI input and output on Dektec cards was the very first reason why I started a small unofficial project in 2005 which became TSDuck later. So, yes, ASI output should work. Let us know if this is not the case.

However, I never took the time to implement the Dektec _demodulators_ (reception). Why? Because this is not a use case for TSDuck. These devices are precise monitoring devices with specific analysis software. You can't really take advantage of them using TSDuck. TSDuck simply reads an input TS. Why purchasing a € 1000+ professional monitoring receptor to receive a TS while a € 50 generic USB or PCI tuner can do the same?

Note that Dektec demodulators use the specific Dektec API, not the standard system API for DVB tuners (DirectShow or LinuxTV), because they serve a different purpose (signal analysis vs. watching TV).

rezna11, 2018-06-23:
I asked because there's one on ebay for quite cheap but after a closer look, I see it's a version without "C" which is supposed to be without GS support and GS with BBFRAMES was the reason why I considered getting this. Don't ask me why, I don't know what I would use it for right now but I just see that there are some "GS" feeds on satellite that can't be decoded with normal equipment - but 2137 without "C" can't do it anyway and Dektec call it "low cost". So I'll pass on this, especially if it's not in TSDuck roadmap to add support.

lelegard, 2020-08-23:
> I would like to reopen this ticket and request support for DekTec demodulators in TSDuck.

It has already been done for some time now. I still have no Dektec demodulator but other users did tests on DTA-2137C and DTA-2138B. Please have a look at the user's guide, `dektec` plugin. If anything is not working properly, please report it here.

rezna11, 2020-08-23:
sorry, I completely forgot, and I confirm DTA-2137C works

closed 2020-08-23 by rezna11