tsduck/tsduck #71: "http" input

lars18th, 2018-06-22:
Hi,

I'm trying to use TSDuck with a SAT>IP server that supports HTTP transport (mainly, a simple URL that you use with a SAT>IP request format that does a continuous streaming). It's very useful for use with a laptop, as you don't need any internal tuner.

However, the problem is that TSDuck doesn't supports the HTTP input. I try to compensate with the PIPE support and CURL. For example with: `curl http://saptip:8080/... | tsp -O hides ...`. However I have a lot of problems:

- The system isn't reliable. After some time, and when a error is found, the full process kills and this error appears: `Error: file: synchronization lost after 805,251 packets, got 0xFF instead of 0x47`.
- Is not possible to "reconnect".

So, I request: Any option? It will be usefull a simple "HTTP receiver" input? Any alternative?

The idea is to have a simple and _reliable_ option for receiving the TS from a remote HTTP server. I appreciate all you can comment.
Thank you!

lelegard, 2018-06-22:
Hi,

Well, this is typically a request for a new input plugin. However, to make it part of TSDuck, we must have a good reason to find it generic enough. A precise specification is necessary. Is `tsp` the TCP server or a TCP client? What is the encapsulation protocol? What is the HTTP request?

And is the content really a transport stream? IP-TV systems use SPTS over UDP/IP, typically multicast. But HTTP over TCP (unicast by definition) is typically used by OTT, VoD and the format is DASH or HLS or similar. In the case of HLS, the transport stream format is used, but using a manifest and chunks. With DASH, the TS format is not even used.

So, is your transmission really a pure transport stream without encapsulation?

Is this a standard or a specific protocol?

rezna11, 2018-06-23:
At some point I also wanted to use HTTP stream on TSDuck input. I think mptsd worked (receiving http, outputting UDP for TSDuck) so that was good enough for me. In my case, my HTTP stream was pure TS packets, it only started with HTTP header. What I used this for was an HTTP server that I wrote for remote channel tuning and streaming, so when a client sent a request, the server moved the dish, tuned in a transponder, set filters, did de-encapsulation (T2MI, TDT) and took care of decryption and started sending out the TS. At the receiving end, I wanted to use this as input for TSDuck.

lars18th, 2018-06-23:
Hi @lelegard ,

I'll try to answer to your requests:

- The module `http-input` is _only_ for INPUT (aka receiver role).
- The transport protocol is plain HTTP.
- Mime type is `video/mp2t`.
- The content is a pure MPEG-2 Transport Stream.
- The requested URL is irrelevant, but configurable as one parameter of the module.

The difference with a plain UDP/Unicast or Multicast is the transport protocol. In this case HTTP over TCP, instead of UDP. And the semantic for the HTTP request is a simple "infinite download", that's a continous (non-stop) downloading of the TS until the connection is closed.

This _type_ of transporting a TS over HTTP is very common in a lot of tools. The advantage over UDP/Multicast is mainly that the connection is "on-request", instead of "passive receiver". Yes, I know that with IGMP is possible to enable/disable the server. But with HTTP is more easy to initiate a processing when the connection is stablished. This is the key point! Futhermore, the TCP _reliable_ transport can be an advantage over the _unreliable_ UDP if you use a fast network in a local environment.

I hope you get the idea. :smile:

lars18th, 2018-06-24:
> So, to be clear, tsp would act as a TCP client, connecting to a TCP server using HTTP protocol.
Am I right?

Yes! :+1:

lelegard, 2018-06-25:
OK, good idea. An `http` input plugin has been added in commit 10fecf2cc5046cfc8b7dca2e5f9bfed7c268744d.

You have to pull and rebuild to test it.

lars18th, 2018-06-25:
> OK, good idea. An http input plugin has been added in commit 10fecf2.
You have to pull and rebuild to test it.

Thank you! I'll check it...

lars18th, 2018-06-26:
Hi @lelegard ,

Compiled and testing it... 
commands like `tsp -I http "http://..." -P analyze -O drop` are working now. Great! A big improvement!
For example, now I can get the stream directly from a SAT>IP server with HTTP transport support (not all have this function, but RTSP; however using a software SAT>IP client/server as a proxy I can reach all). So DVB internal tuners are now retired. :+1: 

The only thing I'm worried about right now is the behaviour when errors come: disconnections, 188 miss-align, corrupted packets, etc.
I'll do more tests and I'll post here the results.

Thank you!

lars18th, 2018-06-26:
Hi @lelegard ,

First suggestion to improve this great `http` input module:

- Add the option to "reconnect" after disconnection (or any error) in the socket.

At time, when the socket closes the entire process ends. In a lot of scenarios this will be the correct behaviour. But, in some cases it's interesting to reconnect. For example, you can support:

`tsp -I http -t | --reconnect-time [value]` where [value] is by default 0 (=no reconnect) and when >0 is the number of seconds (or miliseconds) to wait until reconnect.

You agree with that?

lelegard, 2018-06-26:
There are already options like `--repeat` or `--infinite`. But they stop in case of error. I can add some `--ignore-errors` option to reconnect also in case of error.

lars18th, 2018-06-26:
> There are already options like --repeat or --infinite. But they stop in case of error. I can add some --ignore-errors option to reconnect also in case of error.

Great idea! Thank you! :+1:

lars18th, 2018-06-26:
Hi @lelegard ,

Second suggestion:

- Add the option for selecting the receiving timeout.

I see this error time to time: `ERROR: [http_request ...] receiving timeout`.
I prefer if I can control this value.
You agree?

lars18th, 2018-06-26:
Hi @lelegard ,

Third comment:

- How to control the `http` buffer input?

In the current input module `ip udp` exists the option `--buffer-size`. It will be interesting to have too this option in this module... as not all incomming streams have good timings. In fact, I'm using something like `tsp -I http URL ... -P pcrbitrate -d | ` and I need to increase the buffer to obtain stable results.

Please, can you consider this too?
Thank you!

lelegard, 2018-06-26:
The HTTP implementation is not TSDuck specific. It uses the system built-in libraries: WinInet on Windows and libcurl on Unix. We need to check the tuning capabilities of these libraries.

lars18th, 2018-06-26:
> The HTTP implementation is not TSDuck specific. It uses the system built-in libraries: WinInet on Windows and libcurl on Unix. We need to check the tuning capabilities of these libraries.

This is for the "timeout" of for the "buffer"?

And regarding the "buffer", it exists any generic method for increase the _input_ buffer (not limited to this plugin)?

Thank you @lelegard for your support! This project is amazing! :smile:

lelegard, 2018-06-26:
An update with all requested features has been pushed.

Considering the global buffer, the  `tsp` option `--buffer-size-mb` can do the job I think. But take care that increasing the buffer also increases the latency, which can be a problem with real time streams.

lars18th, 2018-06-26:
Thank you @lelegard ! :+1: 

I'll test it asap. But I feel all is now complete for this plugin. So I'll close this issue if all works right.

Regarding the "global" option `--buffer-size-mb` I think it's that I need it. The user case is: get an HTTP stream, smooth it with the plugin `-P pcrbitrate -d`, and then process it (mainly remuxing). So in this case, the _incoming buffer_ is needed to fix jitter errors. Obviously this increases the latency... but the idea is to do it only for the input stream... not for the entire chain. So, in this scenario what you think: I need to use the option `max-queue` in the `http` input, or the `buffer-size-mb`?

Regards.

lars18th, 2018-06-29:
> If you mean regulating the transmission speed without loosing input packets, then both options can make it, but probably http --max-queue more reliably because this queue if fully dedicated to the input plugin while the global buffer is shared by all plugins.

Just that! And for instance I'm already using `http --max-queue` with an increased buffer.

So, I feel we can close this issue. And you can potentially release new binaries.
Thank you! :+1:

closed 2018-06-29 by lelegard