tsduck/tsduck #73: "switch" plugin

lars18th, 2018-06-25:
Hi,

I don't know how to do this, so I request the creation of another plugin (sorry for doing a lot of requests!).

I need to _switch_ between two different inputs (=streams). And I like to _dynamically_ change from one to another. I know the requirements about compatible PSI/SI tables and other relevant stuff. But, the problem is do all the processing inside the `tsp` chain without interruption. For example, I can use three different `tsp` processes (A,B,C): the last one reads from a multicast address, and the other two write to the same address. But, this implies to start/stop the process A,B in sync... and I found it very difficult.

So, instead of this complex model, I prefer to use a `switch` plugin for something like this:
```
tsp \
...
    -P swith -src "tsp ..." -src "tsp ..." -src "tsp"  \
...
```
More or less similar to the `merge` plugin in the concept of reading from a secondary tsp chain. But instead of "merge", it will "replace" the _entire_ stream with an alternative input. So, the _source 0_ is the "regular" stream from the processing chain. And _source 1_ is the first one; the _source 2_ the following; etc.

In fact, this will be related to the `fork` plugin (with similar parameters), as it does the `merge` one.

The only thing to resolve is _how to select inputs in realtime_. For this functionality, several different alternatives are possible: use one file (and checking changes), receive one UDP message, use a broker protocol, etc. I don't care about the best option. I suggest to use the most simple one. In any case, the trigger can be a simple `NUMBER` that selects the input (between 0 and 255), that the plugin needs to receive/check at _some interval_.

However, the key point is that the real swtich between current input and the new input it has to be smooth. So, only after starting to receive packets from the new input, the other one is discarded; and the switching is done at 188-packet boundaries.

What you think? This has sense?

lelegard, 2018-06-26:
I have some difficulties to find a use case or a rationale.

All new features in TSDuck started with a real use case, a practical need for a good reason. This need is then generalized and turned into a generic tool. Here, my feeling (but I may be wrong) is that we try to do something "because it's possible" (hello Sheldon, Leonard, Raj and Howard :laughing:) rather than because there is a need for it.

lars18th, 2018-06-26:
Hi @lelegard ,
BBT alegory! :smiley: 

Now seriously. I feel you know the (now ancient) OpenCaster from Avalpa. https://github.com/aventuri/opencaster

What I request is a plugin similar to the OpenCaster's `tsinputswitch`: https://github.com/aventuri/opencaster/blob/master/code/tools/tsinputswitch/tsinputswitch.c

The user-case is quite simple: At some point of your processing chain now you "merge" some input. But you need to "select" this input from more than one stream. That's all. Why would you want to do that? For example, because you like to change from one live stream to some recorded one (for example a "no input signal" stream).

To control this "switch", I propose the simple technique used in the `spliceinject` plugin. That's use `--file` or `--udp`. So, for example with a simple `echo 1 >/dev/udp/127.0.0.1/4444` the input 1 is selected. 

I feel now is more clear.
You agree now?

rezna11, 2018-06-26:
I don't want to hijack this thread, just share my opinion on what this could be useful. I myself can see 2 scenarios here:

1) redundancy - for example if you have 2 encoders, main and backup and if you want TSDuck to process the stream from main encoder and if goes down, then (automatically) start processing the backup stream, but I don't know a real use case for this, apart from if someone wants to use TSDuck in productive environment 

2) this could be used as "remote control" to "switch channels", while getting one consistent output from TSDuck - for example, have 10 HTTP sources on standby as TSDuck inputs, only use one as live input. Then, the remote controller (any custom application or script) tells TSDuck to take another source and output it. This would be useful to save bandwidth and avoind the need to mux.

lars18th, 2018-06-26:
Hi @rezna11 ,

Just two of the user cases I'm thinking about! :smile: 

I hope @lelegard accept this suggestion.

lelegard, 2018-08-17:
Having a plugin which completely replaces the stream does not seem a good idea. A packet processing plugin gains control only when packets are present upstream. What you need is to switch inputs.

I think the best way would be to allow more than one input plugin on the command line. Only one input plugin is active at a time and the switch feature is managed at `tsp` level, not in a plugin. I will think about it, but this is significant work.

lars18th, 2018-09-20:
> I think the best way would be to allow more than one input plugin on the command line. Only one input plugin is active at a time and the switch feature is managed at tsp level, not in a plugin. I will think about it, but this is significant work.

Great! But please take your time to think about it. There is no high priority at time. :wink: 

In any case, if you ever going to implement it, don't forget to add a method to make the switching externally (using one UDP message or a monitoring a file).

Regards.

lelegard, 2018-09-25:
Hi @lars18th 

As you noticed, I am working on something in the background (not much time to work on it, so each commit is a small increment only). This will not be a `tsp` plugin for many technical reasons but a `tsswitch` command which can be piped into `tsp` if you like or to any other output. The utility `tsswitch` will reuse the same input and output plugins as `tsp`.

With bash, there is no need to use netcat. The bash shell understands redirections to TCP or UDP. For instance, if `tsswitch` runs on the same host and listens to UDP commands on port 4444:

~~~
$ echo >/dev/udp/127.0.0.1/4444 0
$ echo >/dev/udp/127.0.0.1/4444 4
$ echo >/dev/udp/127.0.0.1/4444 prev
$ echo >/dev/udp/127.0.0.1/4444 next
$ echo >/dev/udp/127.0.0.1/4444 exit
~~~

I will let you know when something will be ready for testing.

lelegard, 2018-09-25:
Just to give an idea of what will be possible (hopefully) :

[image]

lars18th, 2018-09-25:
> Just to give an idea of what will be possible (hopefully) :

Amazing! :astonished: 

I feel with this processing chain will be possible at the same time to:

- Select on-demand any of the different inputs.
- Automatically switch to other input when an error or missing stream appears.

So a very useful tool for a robust input! :wink: 

Tip: Perhaps you like to support too some kind of "reliable output", that's: when all inputs are wrong, then (optionally) add to the output some null packets at defined bitrate. So, if the tsswitch process is live then it will generate a valid stream; and only if the output is zero then you can assume that the processing is completly dead.

Tip2: Supporting the status of the tsswitch process can be interesting too. For example, emitting one UDP message when a switch is completed, or when an error is detected, or when all inputs fail, etc.

They're just suggestions.

lelegard, 2018-09-25:
> Select on-demand any of the different inputs.

Done through UDP messages.

> Automatically switch to other input when an error or missing stream appears.

Already done. When an input terminates or fails, it automatically switches to the next one.

In fact, based on options, when an input terminates or fails, you can either terminate or switch to next input. When all inputs have terminated (one complete cycle), you can terminate or restart a given number of complete cycles (including an infinite number).

Additionally, the default is to start and stop input plugins when they are selected. So, the input is restarted from the beginning at each cycle. But you can also start all inputs in parallel at the beginning and never stop them. The selected input is the one from which the output grabs packets. In the meantime, all other input plugins run and overwrite again and again their input buffer. When you switch to an input, the latest packets are already there, immediately available. This is fine only for live streams which do not use the same physical device (eg. tuner).

> Tip: Perhaps you like to support too some kind of "reliable output", that's: when all inputs are wrong, then (optionally) add to the output some null packets at defined bitrate.

You can do it yourself. Add your "default" input as any command (null at given bitrate if you like) as the last input plugin. If all other inputs fail, the last one will be implicitly selected.

> Tip2: Supporting the status of the tsswitch process can be interesting too. For example, emitting one UDP message when a switch is completed, or when an error is detected, or when all inputs fail, etc.

Hey, come on, be reasonable, why not full SNMP control ? :wink:

lelegard, 2018-09-25:
> Tip3: Support of pre-buffering for a smooth switching when using "on-demand" starting inputs. So, when you have multiple inputs from sources like HTTP, then complete the switching only when the new input is stablished and packets are comming.

This is what I explained in my previous reply: 

> But you can also start all inputs in parallel at the beginning and never stop them. The selected input is the one from which the output grabs packets. In the meantime, all other input plugins run [...]

rezna11, 2018-09-25:
Sorry to step in but I have a question. Will this new tool work in my scenario where I have a single tuner receiver, capable of streaming just one channel (transponder) at a time? You're mentioning a "parallel model" which can be switched off. So if it's off, how would the switch between 2 streams look like?

is it A)
- streaming source 1
- command to switch to source 2 arrives
- immediately activate source 2

or is it also possible to do B)
- stream source 1
- command to switch to source 2 arrives
- `tsswitch ` tool WAITS for defined time to give enough time to source 1 to be switched off and free up the tuner
- then source 2 is activated (and with the option to retry if the waiting period was too low)

lars18th, 2018-09-25:
Hi @rezna11 ,

As far as I know, this isn't finished yet.

However, I suggest to support **both** modes _when parallel mode is disabled_:

- Fast switch: disable current source, then start new source (good for your environment: one tuner).
- Delayed switch: start new source, and when the new input receives data, then switch the inputs and after close the old input.

I hope @lelegard will aggree with support both modes (plus the parallel mode).

lars18th, 2018-09-25:
Hi @lelegard ,

My suggestion is to support:

- `default mode`: First stop the current input, after closing it then start the new input.
- `--fast-switch`: All inputs in parallel, immediately switch.
- `--delayed-switch`: First start the new input, after _receive_ data from it (you can add some timeout in case for error) then complete the switching; and finally close the old input.

The third (new) mode has sense when you use "on-demand" sources (HTTP, for example). So, the input isn't connected if you don't use it. This it's not a problem for the wasted resources where the tsp process runs; but has sense for not connect to the source _until_ the input is active. Imagine one HTTP source that is in fact a remote server with multiple TUNERs. When you start one HTTP connection, then you're locking one tuner; so if you request the same service over two HTTP connections then you're using two tuners... when in fact you only need.

I hope you can consider this "third" mode.

Regards.

rezna11, 2018-09-25:
>  Imagine that you have only one physical tuner, input 1 uses the input plugin dvb on frequency 1, input 2 uses dvb on frequency 2 (same tuner), etc. 

This is exactly what I wanted to hear :) Thanks

joolzg, 2018-09-25:
Will there be auto restart of the main stream.

You have 2 streams, 1 from multicast, 1 from a file on loop.

multicast stops, file starts playing

multicast restarts and this input starts playing.

lelegard, 2018-09-25:
> Will there be auto restart of the main stream.

There is no _main_ stream, just streams. Currently, all input plugins are considered equal. No priority is given. Many different switching policies can be implemented. Let's start with something simple. Not so simple actually, see the previous comments.

In the very specific case you describe, how would you write a technical specification ? What does mean _multicast stops_ ? I mean at the level of the `tsswitch` process. The input plugin `ip` waits for incoming UDP datagrams. Unlike TCP, if the UDP multicast server crashes, there will be no information sent. So, we would need to first modify the `ip` plugin to use some reception timeout terminating the plugin in error. This is the only way to make `tsswitch` aware of something unusual in one of its plugins.

joolzg, 2018-09-25:
Main stream in my case would be the 1st, and they go down in priority.

Your correct, how do you know when the stream has stopped. A timeout would be the only way i guess.

A larger case would be a main stream, a backup stream and a file where the system would switch to a lower stream when the higher priority stream stops, but would switch to the highest when it restarts. In this example, both main and backup stop, so the switch would switch to the file stream, main starts again so switch would switch to main, backup starts again but nothing changes as we are on main.

I hope you understand my ramblings.

lars18th, 2018-09-26:
Hi @lelegard ,

> Let's start with something simple. Not so simple actually...

I agree!
However, in the meantime "another" simple suggestion:

- Please, support to skip received messages for switching when the current input is the one in the message.

As UDP is not reliable, and I like to use a distributed environment, my idea is to send messages twice with a difference of 500ms or 1 second. So, the `tsswitch` process needs to be transparent to _duplicated_ messages.

Of cource, I presume this will be trivial, as you can check the current input when receive a message. However, I prefer to comment it.

Thank you for your good job!

lelegard, 2018-09-26:
> Please, support to skip received messages for switching when the current input is the one in the message.

Already done !

lelegard, 2018-09-28:
Hi @lars18th, @rezna11, @joolzg,

I have a first version to test with the three modes (default, fast and delayed). Please read the user's guide (command tsswitch and example 5.1.7). 

There is no main/backup stream for now. All streams are considered equel. I will see later but I have limited time now. First, you need to define what means "main stream is back" after a failure. At minimum, you must attempt again and again to restart the main plugin while another one is running. And even if the start operation is correct, this does not mean that the stream is back (for instance, starting UDP always succeeds since it is mainly setting up a socket but the stream may not be back).

Testing was minimal. But since you seem interested in the feature, I push the test burden on you :wink: 

Please report your feedback here.

rezna11, 2018-09-29:
I did a quick test with the default mode. I tried using DVB tuner as 3 sources and it works as expected - UDP messages switched between channels from different transponders on the same tuner. I tried switching about 20 times and it was pretty good. The output stream wasn't without errors but that might be because of something else down in the chain. I guess it will be better to `merge` the tsswitch output in another `tsp` instance to keep constant bit rate. I will have to check.

Then I tried to use http sources but that didn't work. The first (default) was OK. Also switching to next was OK. But then I could never switch to 3rd. It looks like it takes too much time to switch off the active http source. Is there a specific parameter to add to force http source to quit as fast as possible?

rezna11, 2018-09-29:
here is a debug log, in here I tried to switch between 3 http inputs - switching to 1 and 2 worked, back to 0 did not...

```
C:\Users\Administrator>tsswitch --debug=1 -v -i -r 10.0.0.11:4444 -I http http://192.168.30.248:6999/1:1:V:S2:3895:20200:70:0:0:0:0 -I http http://192.168.30.248:6999/1:1:V:S2:3895:20200:200:0:0:0:0 -I http http://192.168.30.248:6999/1:1:V:S2:3895:20200:300:0:0:0:0 -O ip 239.0.0.11:5555
debug level set to 1
input buffer: 512 packets, max input: 128 packets, max output: 128 packets
trying to load C:\Program Files (x86)\TSDuck\bin\tsplugin_ip.dll
symbol tspNewProcessor not found in C:\Program Files (x86)\TSDuck\bin\tsplugin_ip.dll
trying to load C:\Program Files (x86)\TSDuck\bin\tsplugin_http.dll
symbol tspNewOutput not found in C:\Program Files (x86)\TSDuck\bin\tsplugin_http.dll
symbol tspNewProcessor not found in C:\Program Files (x86)\TSDuck\bin\tsplugin_http.dll
* Debug: debug level set to 1
* Debug: http[0]: debug level set to 1
* Debug: http[1]: debug level set to 1
* Debug: http[2]: debug level set to 1
* Debug: ip: debug level set to 1
* Debug: setting socket reuse address to 1
* Debug: binding socket to 10.0.0.11:4444
* Debug: http[0]: received start request, current: true
* Debug: http[2]: input thread started
* Debug: http[2]: waiting for input session
* Debug: UDP server thread started
* Debug: ip: output thread started
* Debug: http[0]: input thread started
* Debug: http[0]: waiting for input session
* Debug: http[0]: starting input plugin
* Debug: http[0]: input plugin started, status: true
* Debug: setting event: WAIT_STARTED, 0, true
* Debug: http[1]: input thread started
* Debug: http[1]: waiting for input session
* Debug: http[0]: internal push-input thread started
* Debug: http[0]: HTTP header: HTTP/1.0 200 OK
* Debug: http[0]: HTTP header: Connection: Close
* Debug: http[0]: HTTP header: Content-Type: video/mpeg
* Debug: http[0]: HTTP header: Server: stream_enigma2
* http[0]: downloading from http://192.168.30.248:6999/1:1:V:S2:3895:20200:70:0:0:0:0
* http[0]: MIME type: video/mpeg, expected size: unknown
* Warning: http[0]: MIME type is video/mpeg, maybe not a valid transport stream
* Debug: setting event: WAIT_INPUT, 0, false
* received command "1", from 10.0.0.11:55271 (2 bytes)
* Debug: switch input 0 to 1
* Debug: enqueue action STOP, 0, false
* Debug: enqueue action WAIT_STOPPED, 0, false
* Debug: enqueue action SET_CURRENT, 1, false
* Debug: enqueue action START, 1, true
* Debug: enqueue action WAIT_STARTED, 1, false
* Debug: setting event: NONE, 0, false
* Debug: executing action STOP, 0, false
* Debug: http[0]: received stop request
* Debug: executing action WAIT_STOPPED, 0, false
* Debug: executing action SET_CURRENT, 1, false
* Debug: executing action START, 1, true
* Debug: http[1]: received start request, current: true
* Debug: executing action WAIT_STARTED, 1, false
* Debug: not ready, waiting: WAIT_STARTED, 1, false
* Debug: http[1]: starting input plugin
* Debug: http[1]: input plugin started, status: true
* Debug: setting event: WAIT_STARTED, 1, true
* Debug: executing action WAIT_STARTED, 1, false
* Debug: http[0]: stopping input plugin
* Debug: http[1]: internal push-input thread started
* Debug: http[0]: Web transfer is interrupted by application
* Debug: http[0]: internal push-input thread completed
* Debug: input 0 completed, success: true
* Debug: setting event: WAIT_STOPPED, 0, false
* Debug: http[0]: waiting for input session
* Debug: http[1]: HTTP header: HTTP/1.0 200 OK
* Debug: http[1]: HTTP header: Connection: Close
* Debug: http[1]: HTTP header: Content-Type: video/mpeg
* Debug: http[1]: HTTP header: Server: stream_enigma2
[... 925 lines cut ...]
* Debug: executing action START, 0, false
* Debug: executing action SET_CURRENT, 1, false
* Debug: executing action WAIT_STARTED, 0, false
* Debug: enqueue action WAIT_STARTED, 2, false
* Debug: input 0 completed, success: true
* Debug: http[2]: starting input plugin
* Debug: http[0]: starting input plugin
* Debug: http[2]: input plugin started, status: true
* Debug: enqueue action WAIT_STARTED, 1, false
* Debug: executing action SET_CURRENT, 1, false
* Debug: http[2]: stopping input plugin
* Debug: enqueue action WAIT_STARTED, 0, false
* Debug: enqueue action SET_CURRENT, 2, false
* Debug: http[2]: starting input plugin
* Debug: setting event: WAIT_STOPPED, 2, false
* Debug: executing action SET_CURRENT, 0, false
* Debug: http[0]: stopping input plugin
* Debug: http[2]: starting input plugin
* Debug: http[0]: starting input plugin
* Debug: http[1]: received start request, current: false
* Debug: http[2]: starting input plugin
* Debug: http[2]: starting input plugin
* Debug: ^C
C:\Users\Administrator>
```

rezna11, 2018-10-01:
Thanks! Now it's possible to switch to a previously used source, so there's definitely an improvement.
One thing I'm not sure about, with the `http` plugin as source, is the way how the http connection is ended when I switch to another plugin. It seems to me that `tsswitch` stops reading data but the socket is not closed. I don't have a detailed log yet to confirm this but this suspicion is based on the fact that my http server normally closes the client connection immediately if I push STOP in VLC or switch to another channel in Enigma2, for example. But when the stream goes through `tsswitch`, the http server doesn't detect that the connection ended and keeps it open for some time until it detect a timeout.

lelegard, 2018-10-01:
Considering your previous debug log, I assume that you are using Windows. In that case, TSDuck uses the builtin library WinInet for HTTP access (on Unix systems, it uses libcurl). VLC is probably using its own HTTP library, maybe libcurl, even on Windows. This may explain the difference on the disconnection behaviour.

The detailed behaviour of WinInet is not always well documented. Building a test configuration to reproduce your problem would take me some time. So, could you make some additional tests?

First, just to check, try to remove the flag `INTERNET_FLAG_KEEP_CONNECTION` in file `src/libtsduck/windows/tsWebRequestGuts.cpp`, line 222. The exact effect of this flag is not clear. There is a possibility that it keeps the connection with the server for some time, even if we close the URL. Removing the flag here is not the right final solution. But I would like to know if it has an influence on your delayed disconnection.

Second, only if the previous test did not success, please post a debug log of the sequence of an input switch.

rezna11, 2018-10-01:
Yes, I use Windows. I tried to remove the flag but get the same result. On the other hand, if I run `tsp` with `http `as input and then I quit tsp, the connection to the http server is ended immediately, so this behavior of not cutting the connection is related to `tsswitch` only.

Debug log with the `INTERNET_FLAG_KEEP_CONNECTION` flag removed:

```
C:\Users\Administrator>tsswitch --debug=1 -v -i -r 10.0.0.11:4444 -I fork "tsp --bitrate 100 -I null -P regulate" -I http http://192.168.30.248:6999/1:1:V:S2:3895:20200:70:0:0:0:0 -I http http://192.168.30.248:6999/1:1:V:S2:3895:20200:200:0:0:0:0 -I http http://192.168.30.248:6999/1:1:V:S2:3895:20200:300:0:0:0:0 -O ip 239.0.0.11:5555
debug level set to 1
input buffer: 512 packets, max input: 128 packets, max output: 128 packets
* Debug: debug level set to 1
trying to load C:\Program Files (x86)\TSDuck\bin\tsplugin_ip.dll
symbol tspNewProcessor not found in C:\Program Files (x86)\TSDuck\bin\tsplugin_ip.dll
trying to load C:\Program Files (x86)\TSDuck\bin\tsplugin_fork.dll
symbol tspNewOutput not found in C:\Program Files (x86)\TSDuck\bin\tsplugin_fork.dll
trying to load C:\Program Files (x86)\TSDuck\bin\tsplugin_http.dll*
Debug: fork[0]: debug level set to 1symbol tspNewOutput not found in C:\Program Files (x86)\TSDuck\bin\tsplugin_http.dll

symbol tspNewProcessor not found in C:\Program Files (x86)\TSDuck\bin\tsplugin_http.dll
* Debug: http[1]: debug level set to 1
* Debug: http[2]: debug level set to 1
* Debug: http[3]: debug level set to 1
* Debug: ip: debug level set to 1
* Debug: setting socket reuse address to 1
* Debug: binding socket to 10.0.0.11:4444
* Debug: UDP server thread started
* Debug: ip: output thread started
* Debug: fork[0]: received start request, current: true
* Debug: fork[0]: input thread started
* Debug: fork[0]: waiting for input session
* Debug: fork[0]: starting input plugin
* Debug: fork[0]: creating process "tsp --bitrate 100 -I null -P regulate"
* Debug: http[1]: input thread started
* Debug: http[1]: waiting for input session
* Debug: http[2]: input thread started
* Debug: http[2]: waiting for input session
* Debug: http[3]: input thread started
* Debug: http[3]: waiting for input session
* Debug: fork[0]: input plugin started, status: true
* Debug: setting event: WAIT_STARTED, 0, true
* received command "1", from 10.0.0.11:50031 (2 bytes)
* Debug: switch input 0 to 1
* Debug: enqueue action STOP, 0, false
* Debug: enqueue action WAIT_STOPPED, 0, false
* Debug: enqueue action SET_CURRENT, 1, false
* Debug: enqueue action START, 1, true
* Debug: enqueue action WAIT_STARTED, 1, false
* Debug: setting event: NONE, 0, false
* Debug: executing action STOP, 0, false
* Debug: fork[0]: received stop request
* Debug: executing action WAIT_STOPPED, 0, false
* Debug: executing action SET_CURRENT, 1, false
* Debug: executing action START, 1, true
* Debug: http[1]: received start request, current: true
* Debug: executing action WAIT_STARTED, 1, false
* Debug: not ready, waiting: WAIT_STARTED, 1, false
* Debug: http[1]: starting input plugin
* Debug: http[1]: input plugin started, status: true
* Debug: setting event: WAIT_STARTED, 1, true
* Debug: executing action WAIT_STARTED, 1, false
* Debug: http[1]: internal push-input thread started
* Debug: http[1]: HTTP header: HTTP/1.0 200 OK
* Debug: http[1]: HTTP header: Connection: Close
* Debug: http[1]: HTTP header: Content-Type: video/mpeg
* Debug: http[1]: HTTP header: Server: stream_enigma2
* http[1]: downloading from http://192.168.30.248:6999/1:1:V:S2:3895:20200:70:0:0:0:0
* http[1]: MIME type: video/mpeg, expected size: unknown
* Warning: http[1]: MIME type is video/mpeg, maybe not a valid transport stream
* Debug: setting event: WAIT_INPUT, 1, false
* received command "0", from 10.0.0.11:50031 (2 bytes)
* Debug: switch input 1 to 0
* Debug: enqueue action STOP, 1, false
* Debug: enqueue action WAIT_STOPPED, 1, false
* Debug: enqueue action SET_CURRENT, 0, false
* Debug: enqueue action START, 0, true
* Debug: enqueue action WAIT_STARTED, 0, false
* Debug: setting event: NONE, 0, false
* Debug: executing action STOP, 1, false
* Debug: http[1]: received stop request
* Debug: executing action WAIT_STOPPED, 1, false
* Debug: executing action SET_CURRENT, 0, false
* Debug: executing action START, 0, true
* Debug: fork[0]: received start request, current: true
* Debug: executing action WAIT_STARTED, 0, false
* Debug: not ready, waiting: WAIT_STARTED, 0, false
* Debug: executing action WAIT_STARTED, 0, false
* Debug: http[1]: stopping input plugin
* Debug: http[1]: error pushing packets
* Debug: http[1]: Web transfer is interrupted by application
* Debug: http[1]: internal push-input thread completed
* Debug: input 1 completed, success: true
* Debug: setting event: WAIT_STOPPED, 1, false
* Debug: http[1]: waiting for input session
* tsp: user interrupt, terminating...
```

lelegard, 2018-10-01:
OK, I can reproduce the issue on Windows. I first setup an Apache server on a Linux VM with a cgi-bin script which streams TS with a command like `tsp -I file -i ... -P regulate`. Thus, I have a cheap HTTP streamer. Then, I use a `tsswitch` command with two URL's on this server.

When the `tsswitch` clients runs on Windows (WinInet library), after a `next` command, the previous socket on the server remains in state `ESTABLISHED`. After some time, it goes into `FIN_WAIT1` and then disappears after a while.

When the `tsswitch` clients runs on Linux or macOS (libcurl library), the previous socket immediately disappears on the server.

There is clearly an issue with the way WinInet closes its sockets. I will investigate.

In the meantime, I suggest to apply the universal remedy in computing: avoid Windows and use Linux or macOS :wink:

rezna11, 2018-10-01:
Thanks for investigating the issue!
Unfortunately I rely on Dektec, so I can't use a Mac. And I was never able to make all of my DVB tuners work on Linux - one worked, the other did not. Never managed to get more than one or two to work (driver issues I guess). I've tried many Linux kernel versions and various linux TV drivers, no luck, just tens of hours wasted...

lelegard, 2018-10-01:
Don't worry, it was a joke. Almost...

I find the issue and fixed it on Windows. There is no more pending socket on the web server side, the disconnection is immediate, as it was with tsswitch on Linux.

rezna11, 2018-10-01:
I'm running into another issue now. And this worked before the last update... the `fork` source (the 1st in my command) seems to get stuck and won't let me switch to other sources. Is it possible that the http fix broke something else? 

The reason I use this fork source is not to stream from the http server at times when no stream is needed.

Log:

```
C:\Users\Administrator>tsswitch --debug=1 -v -i -r 10.0.0.11:4444 -I fork "tsp --bitrate 100 -I null -P regulate" -I http http://192.168.30.248:6999/1:1:V:S2:3895:20200:70:0:0:0:0 -I http http://192.168.30.248:6999/1:1:V:S2:3895:20200:200:0:0:0:0 -I http http://192.168.30.248:6999/1:1:V:S2:3895:20200:300:0:0:0:0 -O ip 239.0.0.11:5555
debug level set to 1
input buffer: 512 packets, max input: 128 packets, max output: 128 packets
* Debug: debug level set to 1
trying to load C:\Program Files (x86)\TSDuck\bin\tsplugin_ip.dll
symbol tspNewProcessor not found in C:\Program Files (x86)\TSDuck\bin\tsplugin_ip.dll
trying to load C:\Program Files (x86)\TSDuck\bin\tsplugin_fork.dll
* trying to load C:\Program Files (x86)\TSDuck\bin\tsplugin_http.dll
Debug: fork[0]: debug level set to 1
symbol tspNewOutput not found in C:\Program Files (x86)\TSDuck\bin\tsplugin_http.dll
symbol tspNewProcessor not found in C:\Program Files (x86)\TSDuck\bin\tsplugin_http.dll
* Debug: http[1]: debug level set to 1
* Debug: http[2]: debug level set to 1
* Debug: http[3]: debug level set to 1
* Debug: ip: debug level set to 1
* Debug: setting socket reuse address to 1
* Debug: binding socket to 10.0.0.11:4444
* Debug: UDP server thread started
* Debug: fork[0]: received start request, current: true
* Debug: http[2]: input thread started
* Debug: http[2]: waiting for input session
* Debug: http[3]: input thread started
* Debug: http[3]: waiting for input session
* Debug: ip: output thread started
* Debug: fork[0]: input thread started
* Debug: fork[0]: waiting for input session
* Debug: fork[0]: starting input plugin
* Debug: fork[0]: creating process "tsp --bitrate 100 -I null -P regulate"
* Debug: http[1]: input thread started
* Debug: http[1]: waiting for input session
* Debug: fork[0]: input plugin started, status: true
* Debug: setting event: WAIT_STARTED, 0, true
* received command "1", from 10.0.0.11:49906 (2 bytes)
* Debug: switch input 0 to 1
* Debug: enqueue action STOP, 0, false
* Debug: enqueue action WAIT_STOPPED, 0, false
* Debug: enqueue action SET_CURRENT, 1, false
* Debug: enqueue action START, 1, true
* Debug: enqueue action WAIT_STARTED, 1, false
* Debug: executing action STOP, 0, false
* Debug: fork[0]: received stop request
* Debug: executing action WAIT_STOPPED, 0, false
* Debug: not ready, waiting: WAIT_STOPPED, 0, false
* received command "2", from 10.0.0.11:49906 (2 bytes)
* Debug: switch input 0 to 2
* Debug: enqueue action STOP, 0, false
* Debug: enqueue action WAIT_STOPPED, 0, false
* Debug: enqueue action SET_CURRENT, 2, false
* Debug: enqueue action START, 2, true
* Debug: enqueue action WAIT_STARTED, 2, false
* Debug: executing action WAIT_STOPPED, 0, false
* Debug: not ready, waiting: WAIT_STOPPED, 0, false
* tsp: user interrupt, terminating...
```

lelegard, 2018-10-02:
Hi @rezna11 

This should be fixed with the latest commit.

The actual problem was in `tsp` and plugin `regulate` in your forked command. The process `tsp` terminates, among other conditions, when the output fails (which is the case when you switch input and break the pipe of the `fork` plugin). This termination condition is propagated backward to all plugins, down to the input plugin. But, for a given plugin, the internal mechanics of `tsp` checked this termination condition only when the plugin needed more packets from the previous plugin.

In the case of `tsp --bitrate 100 -I null -P regulate`, the input plugin immediately saturated the large internal buffer (16 MB by default, ~90000 packets) with null packets and waited for more space. The plugin `regulate` immediately got all these packets and let them pass very slowly, at 100 b/s. And the termination condition was checked only when the plugin needed more input packets, 1 million seconds later (16 MB at 100 b/s). Since `tsp` terminates when all its plugins are terminated, the `tsp` process apparently hanged.

I fixed this and now the termination condition is checked more frequently.

Now, concerning `tsswitch`, in default mode, an input is first stopped, wait for stop completion (in the case of the `fork` plugin without `--nowait` option, this means termination of the `tsp` process) and then start the next input. So, in practice, this never switched within the next 1 million seconds.

Finally, you mentioned that this did not happen previously. Well, previously, `tsswitch` did not wait correctly the completion of the input stop...

lelegard, 2018-10-04:
Hi,

I have added  two options to `tsswitch`: `--receive-timeout` and `--primary-input`.

With `--receive-timeout`, if the current input plugin does not receive any packet within the specified timeout, we automatically switch to the next input. By default, it only switches to the next input when the current input terminates or returns an error. But there are cases where the packet source dies and the input plugin has no way to detect that the source died and the input plugin waits forever (`ip` for instance which uses unconnected UDP reception).

With `--primary-input`, we address @joolzg's concern. We define one of the sources as the primary one. This source is always started, never stopped (a bit like `--fast-switch` but only this input, others are started and stopped on demand). When no packet is received on the primary source, we switch to the next one. As long as there is no packet on the primary source, all remote commands are executed. But as soon as packets re-appear on the designated primary input, we automatically switch back to it.

Note that `--primary-input` automatically enables `--receive-timeout`.

Minimal testing was done (but including Windows this time). I use `tsswitch` with 4 different sources: `ip`, `http`, `fork` and `file`, switching back and forth between them. When tested with `--primary-input`, `ip` was the primary input because it is easy to start and stop the packet source without disturbing the input plugin.

Please do more testing and provide feedback.

I think that `tsswitch` seems feature-complete now.

joolzg, 2018-10-05:
Thanks for the primary switching. I will try and give it a test over the weekend.

Again THANKS for this great tool.

lelegard, 2018-10-17:
Hi @lars18th, @rezna11, @joolzg,

Any feedback or issue with `tsswitch` ? I plan to release version 3.15 some time soon, but not with pending issues in `tsswitch`. So, I am more or less waiting for your green-light.

lars18th, 2018-10-18:
Hi @lelegard ,

I'll test soon and I'll post results.
Thank you!

lars18th, 2018-10-22:
Hi @lelegard ,

After initial tests all seems to work. Thank you! :wink: 

However, some recomendations:

- I use this simple test line, and I recommend to share it in the documentation example 5.1.7:
```
tsswitch -v -m -r 127.0.0.1:4444 -i \
 -I ip 239.0.0.1:12345 -I ip 239.0.0.2:12345 -O ip 239.0.0.3:12345
```
- The parameter `-i|--infinite` seems to be required _all the time_, and this isn't advised! With the previous example, if I omit the `-i` then after the `echo >/dev/udp/127.0.0.1/4444 0` **after** a previous ` echo >/dev/udp/127.0.0.1/4444 1` the process quits. This isn't the expected behaviour. The idea is to switch between any input until a quit/exit message is received. However, without the `-i` this is false. Please, advise it or modify the default behaviour.

I feel you can release this version. But I'll keep checking to see if there's a bug.
Regards.

lelegard, 2018-10-22:
> The parameter -i|--infinite seems to be required all the time, and this isn't advised!

The documented default behaviour is to run all input plugins in sequence until the last one and then terminate (in the absence of remote commands). From `tsswitch` documentation: 

> Running all input plugins in sequence, from 0 to n-1, is called a _cycle_. By default, only one cycle is executed before `tsswitch` terminates. Using the option `--cycle`, it is possible to execute a given number of cycles. With the option `--infinite`, `tsswitch` runs endlessly.

lars18th, 2018-10-22:
OK. So from my point of view the tool is working well.

> Any feedback or issue with tsswitch ? I plan to release version 3.15 some time soon, but not with pending issues in tsswitch. So, I am more or less waiting for your green-light.

Green light to the release.

rezna11, 2018-10-23:
It's now working for me too. I tried using default mode with DVB tuner and `--fast-switch` with IP cameras and both worked as expected.

lelegard, 2018-10-23:
OK, thanks for your feedback. I close this one. Please open a new issue if you encounter a specific proble with `tsswitch`.

closed 2018-10-23 by lelegard

kirillbushev, 2021-08-06:
So, first of all i would like to thank all participants of this thread, and sure almighty @lelegard.
You've done a great job, after reading up to the 377 out of 506 pages of tsduck documentation (i'll finish it, i hope), i was able to produce the following one-liner:

 tsswitch --verbose -f --debug=1 --remote 4444 -m -I http http://10.5.110.100/sdi4/mpegts -I fork --nowait 'tsp --control-port 8888 -I file --infinite blackout.ts -P pcrbitrate -P regulate' -O ip localhost:6668

(later udp mpegts from :6668 is grabbed by ffmpeg and feeded via rtmp, outofscope)

The blackout.ts is produced by:
tsp -I ..input (the same as first one in tsswitch) -P until --seconds 10, then it is
tssfixcc'ed -c -d1 for discontinuity fix, and i suppose to use -P regulate cause bitrates are different (that's was not obvious for me, like many other things still..)

remote 4444 is used to echo "next" | nc -uw 1 localhost 4444 (ubuntu doesn't have /dev/udp from the scratch at least), sources are switching ALMOST fine (maybe some 1 frame jitter..)!

The inner forked control-port 8888 is used to CHANGE blackout file on the fly.
So i use
tspcontrol -t 127.0.0.1:8888 list -v

Executable: /usr/bin/tsp

 0: -I file --infinite blackout.ts
 1: -P pcrbitrate
 2: -P regulate
 3: -O file

Then tspcontrol -t 127.0.0.1:8888 restart 0 --infinite blackout2.ts  (new file ..2.ts)

And...

Eventually this is not working as expected. It doesn't switch right away, during playback. It seems it really switch only in some circumstances: like ending the initial file playback, or when i issue multiple "next" commands, and only then i am able to see the new blackout file instead of old blackout.

I'll appreciate any help.

If fact, i would like to share my use case:

So we are one of the major federal TV broadcasting company, the digital part of it.
We have many live streams, which shows many different content : news, shows, serials, etc.
Some of content is produced internally,  some are purchased from different production companies - that is when "license polices" arise. For example, some piece of content could NOT be shown on some digital surface, according to contract. So we have to put a "blackout" (looped video "this content is unavailable on this surface" in a live stream for this period. This should sound pretty familiar for you. We have a cms, with "digital rights", so we know "time" and duration, when we should inject a blackout in a live stream. Sure, it's not frame accurate. Currently we use a custom Wowza Streaming Server java-plugin (it ask a DRM system every minute and switches inputs).

It was ok until we want to use SCTE-35. I am still not sure if it fits for our use case, because we currently don't have much markers in input stream from big TV. Only a few, for regional ad breaks. If anyone could share how markers could be inserted with frame accuracy (pts..? where i supposed to get it?), then analyzed and for example, if it's an adbreak - server-side insert ad, if it's non ad break, but we do not have rights to show - embed a blackout... 

I am fan of  opensource software, all technologies might be already available on some proprietary hardware, but i suppose they all just have tsduck inside it )

I hope the last 129 pages of documentation will help me..

lelegard, 2021-08-06:
Hi @kirillbushev,

You should really open a new issue for that kind of question. Appending to a discussion which was closed almost three years ago, there is a high risk to get unnoticed.

> remote 4444 is used to echo "next" | nc -uw 1 localhost 4444 (ubuntu doesn't have /dev/udp from the scratch at least),

The syntax `>/dev/udp/host/port` is a special feature of the bash shell. It works on all OS, including Windows! At system level, there is no such file, `ls /dev/udp` does not show anything. You cannot open a file with that name. Only the bash redirection works. Just try...

> It doesn't switch right away, during playback. It seems it really switch only in some circumstances: like ending the initial file playback, or when i issue multiple "next" commands, and only then i am able to see the new blackout file instead of old blackout.

My guess is that this is just a buffering effect.

Your command is `tsp -I file --infinite blackout.ts -P pcrbitrate -P regulate`. The input plugin reads input as fast as the input comes (at disk speed in case of a file). TS packets are stored in the tsp global buffer (16 MB by default) as long as there is free space in the buffer. The plugin `regulate` slows down the output of the buffer. When you restart the input plugin with a new file, new input packets will immediately be read from the new file. _However_, there are still 16 MB of content from the previous file in the global buffer which is output as the regulated speed. If the bitrate of the TS is 8 Mb/s, that will take 16 seconds. The first packets of the second file will then reach the output only 16 seconds after being read.

In short, you act on input but you observe output. There can be some delay between the two.

For such a `tsp` command, you don't need a large buffer. I would suggest to start with `--buffer-size-mb 1` (1 MB buffer). If you want to reduce it more, use `--buffer-size-mb 0.1` for instance. Thus, there are less previous content to flush before the new file and the switch will appear faster.

kirillbushev, 2021-08-09:
@lelegard buffer-size-mb 0.1 helped, thx for quick reply! (bash trick also)
I'll return later in separate thread for scte-35.