tsduck/tsduck #74: "svrename" plugin doesn't update EIT

lars18th, 2018-06-26:
Hi,

I know that the EIT info is very different to the rest of PSI/SI tables. However, when "renaming" the ID of a service is required to update too the id in the EIT. The problem in this case it that the processing of the EIT pid could be huge. So I suggest to add a new parameter in the `svrename` plugin to achieve this task:

- `svrename -e | --process-eit` will instruct to replace the id in the ETI table.

This will be a first improvement before the development of the future `eitmerge` plugin described at #62.
Wihtout this change will be impossible to "merge" a MPTS with valid EIT info.

What you think, @lelegard ?

lelegard, 2018-06-26:
Yes, definitely a valuable future enhancement

lelegard, 2018-07-03:
Fixed in plugin 753ce15c4d68b30da4292cc214665673e418d584

In addition to `svrename`, the plugins `tsrename` and `svremove` are similarly impacted.

For consistency with BAT and NIT, the EIT's are updated (or removed) by default and the option `--ignore-eit` is used to leave them unchanged.

closed 2018-07-03 by lelegard

lars18th, 2018-07-04:
Hi @lelegard ,

Thank you! It works perfect.
This is the first action before the `eitmerge` plugin (#62).