tsduck/tsduck #75: Build error

Philip-Wells, 2018-06-26:
I am seeing a build error when building the default target (ie make in top directory).
I installed all the pre-requisite packages listed in the build documentation.  

Linux phlp 4.17.2-200.fc28.x86_64 #1 SMP Mon Jun 18 20:09:31 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

The tail end of the log reads
  ...
  [DTAPI] using ../../dektec/LinuxSDK/DTAPI/Lib/GCC4.4/DTAPI64.o
  [CC] release-x86_64/tsduck.so
/usr/bin/ld: skipping incompatible /usr/lib/gcc/x86_64-redhat-linux/8/../../../libcurl.so when searching for -lcurl
/usr/bin/ld: skipping incompatible //lib/libcurl.so when searching for -lcurl
/usr/bin/ld: skipping incompatible //usr/lib/libcurl.so when searching for -lcurl
/usr/bin/ld: cannot find -lcurl
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:124: release-x86_64/tsduck.so] Error 1

Here's the libcurl images that I have 

[phil@phlp tsduck]$ find /usr/lib -name libcurl*
/usr/lib/libcurl.so
/usr/lib/vmware/lib/libcurl.so.4
/usr/lib/vmware/lib/libcurl.so.4/libcurl.so.4
/usr/lib/vmware-ovftool/libcurl.so.4
/usr/lib/pkgconfig/libcurl.pc
/usr/lib/libcurl.so.4
/usr/lib/libcurl.so.4.5.0

lelegard, 2018-06-26:
Hi,

It looks like you have a system configuration issue. Given your `uname` output, I assume that you have a 64-bit Fedora 28 system.

Here is my current Fedora 28 config:
~~~
$ uname -a
Linux vmfedora 4.17.2-200.fc28.x86_64 #1 SMP Mon Jun 18 20:09:31 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
$ find /usr/lib* -name libcurl*
/usr/lib64/libcurl.so.4
/usr/lib64/pkgconfig/libcurl.pc
/usr/lib64/libcurl.so
/usr/lib64/libcurl.so.4.5.0
$ rpm -qa libcurl*
libcurl-devel-7.59.0-4.fc28.x86_64
libcurl-7.59.0-4.fc28.x86_64
$ 
~~~

You can see that my `libcurl` files are `/usr/lib64` while yours are in `/usr/lib`. I wonder if you installed the 32-bit `libcurl` library. Could you post the output of `rpm -qa libcurl*`?

Philip-Wells, 2018-06-26:
libcurl-7.59.0-4.fc28.x86_64
libcurl-7.59.0-4.fc28.i686
libcurl-devel-7.59.0-4.fc28.i686

lelegard, 2018-06-26:
OK, so you do not have the 64-bit version of `libcurl-devel`. This explains the error.

Philip-Wells, 2018-06-26:
Installed libcurl-devl.x86_64 and recompiled and build completed successfully.  

Thanks

closed 2018-06-27 by Philip-Wells