tsduck/tsduck #76: Regulate plugin at output speed

lars18th, 2018-06-27:
Hi @lelegard  ,

Sorry if I doing a lot of requests! But, now I’m starting to work with TSDuck and each time I find something interesting then I want to share it. :wink: 

This time I like to share a suggestion about a "simple" (I hope) improvement with a lot of powerful.

As I’m working on replacing the "mptsd" muxer with TSDuck (however, I feel some improvements in the library are pending to complete this task), I see this thing:

- When you like mux multiple inputs and create one MPTS stream (from scratch), the first plugin to use is a simple "null packet generator" (`-I null`). Then you need to adjust the bitrate for the resulting mux. So typically you apply a global value (`tsp --bitrate [value]`) and you add after the null generator a filter to regulate the bitrate (`-P regulate`). However, the [value] of the bitrate needs to be pre-calculated. In this case, if you like to use a modulator for the output then you could use the tool `tsterinfo` and pass to it the modulation parameters. The results of this tool include the “Transport stream bitrate” that’s the number to use as bitrate [value].

So, why not do this automatically? That’s the idea:

- Add to the plugin `regulate` the parameter `-a | --at-output-speed`. And this automatically gets the "bitrate" from the output and apply it.

As I feel, the entire chain of tsp shares global info. So, it’s possible to set values from the end of the chain to the rest of modules. Furthermore, if the output doesn’t calculate any bitrate value then an error is raised. But, you can fix this indicating a max bitrate value with a general `tsp --bitrate [max]`.

At the end the suggestion is to support this:

- `tsp --bitrate 30000000 -I null -P regulate -a ... -O hides -b 8 -c QPSK -f 474000000 -g 1/32 -h 3/4 -t 8K `

And the bitrate is adjusted to 9,048,128 b/s automatically.

What you think?

lars18th, 2018-06-27:
Hi,

After some testing, I need to modify my proposal...

The plugin to change should be another different to `regulate`, as the function of this module isn't to modify the bitrate (but to force it). Perhaps the best semantic is to use a global parameter like in this example:

- `tsp --bitrate 30000000 --at-output-speed -I null -P regulate ... -O hides -b 8 -c QPSK -f 474000000 -g 1/32 -h 3/4 -t 8K`

In this case, the sintax is more clear.
You agree?

lelegard, 2018-06-29:
I will try to find a good and generic solution for this question.

Currently, the bitrate information is propagated from input to output all along the chain. Maybe we could also propagate the same information backward. Then, at each plugin in the chain, `tsp` will have two bitrate information, one coming forward from the input plugin and one coming backward from the output plugin. And `tsp` will select the "most relevant" information to present to the plugin.

To be able to select a "most relevant" information, we need to evaluate and propagate a level of confidence with a bitrate information. In decreasing order of confidence in the bitrate value, we have: ASI in/out, modulator out, tuner in, PCR evaluation, nothing.

This is a generic solution which will solve your specific question since the level of confidence in the bitrate coming forward from `null` is zero while it is very high with the one coming backward from `hides` or `dektec`.

I will think about it.

lars18th, 2018-06-29:
> I will think about it.

Good! This is only an idea for a more friendly use of modulators. In any case, don't worry a lot! Just put it in the TODO.

Regards.

lars18th, 2021-02-25:
Hi @lelegard ,

You think it's time to implement the functionality of `tsp --bitrate-at-output-speed -I null -P regulate ... -O hides ...` ?

closed 2026-04-23 by github-actions[bot]