tsduck/tsduck #81: Multicast IP input/output flexibility

zuzzurro, 2018-07-03:
I think, but I may be wrong as I have been in the past about tsduck, that there is currently no way to directly select which interface to receive/send multicast streams and people have to rely on OS level routing.
Would it be possible to add code to let the user control it directly? In particular I'm thinking at the way multicat does it (that from a comment in their code seems to have been borrowed from DVBlast) where one can add options at the end of the address such as:

Options include:
 /ifindex=X (binds to a specific network interface, by link number)
 /ifaddr=XXX.XXX.XXX.XXX (binds to a specific network interface, by address)
 /ttl=XX (time-to-live of the UDP packet)
 /tos=XX (sets the IPv4 Type Of Service option)
 /tcp (binds a TCP socket instead of UDP)
 /srcaddr=XXX.XXX.XXX.XXX (source address for raw packets)
 /srcportr=XX (source port for raw packets)

lars18th, 2018-07-03:
Hi @zuzzurro ,

Regarding the IP (OUTPUT) from the current documentation:

```
-l address
--local-address address
When the destination is a multicast address, specify the IP address of the outgoing local interface.
It can be also a host name that translates to a local address.
```

So you can use `tsp -O ip -l 127.0.0.1 239.0.0.1:10000` and then you will send it over the loopback.

And regarding the IP (INPUT):

```
-f
--first-source
Filter UDP packets based on the source address. Use the sender address of the first received
packet as only allowed source.
This option is useful when several sources send packets to the same destination address and
port. Accepting all packets could result in a corrupted stream and only one sender shall be
accepted.
To allow a more precise selection of the sender, use option --source. Options --first-source
and --source are mutually exclusive.

-l address
--local-address address
Specify the IP address of the local interface on which to listen. It can be also a host name that
translates to a local address. By default, listen on all local interfaces
```

So you can do similar `tsp -I ip -f 192.168.1.100 -l 192.168.1.101 239.0.0.1:10000` and receive only over the interface with address 192.168.1.101 and only from source 192.168.1.100.

And in both modules it's also the `--ttl` option. However, in any case I agree with **adding the parameter `tos` for the output module.**

@lelegard , you think it's good to support `tos` in output?

lelegard, 2018-07-03:
For input `ip`, the strategy is the following:

- By default, listen on all interfaces, sending join requests on all of them. This is brutal but efficient. The user does not have to worry about routing tables.
- With option `--default-interface`, let the system select the right interface based on routing table (and consequently send one single join request).
- With option `--local-address`, the local interface is manually selected.

For output `ip`:

- By default, let the system select the right interface based on routing table.
- With option `--local-address`, the outgoing local interface is manually selected.

The options `/srcaddr` and `/srcport` you mention are used for IP spoofing. I don't really like this idea.

The source selection options in the input plugin (`--source` and `--first-source`) serve a different purpose. They are used to avoid mixing streams from different sources, accidentally sending to the same destination.

A `--tos` option can be added to the output plugin. Be aware that, as far a I read on the net, this requires some registry settings on Windows.

lelegard, 2018-07-03:
Option `--tos` has been added to output `ip`. It was really straightforward.

I let you test it and post results here. It will probably work on Linux and will probably do nothing on Windows.

lars18th, 2018-07-04:
Thank you!

I'll check it.

lelegard, 2018-08-06:
Can we close this one?

closed 2018-09-16 by lelegard