tsduck/tsduck #82: DSM-CC Object Carousel

piotrserafin, 2018-07-03:
I would like to introduce DSM-CC Object Carousel support for tsduck.  I have created DSM-CC engine in the past. Submitting this issue for upvotes if you are interested.

lelegard, 2018-07-03:
Feel free to submit code contributions on this using pull requests.

I suppose that you should restrict to some forms of carousels. There are many different applications of DSM-CC, from SSU to MPE and many other usages and formats.

lars18th, 2018-07-04:
Hi @piotrserafin ,

I agree with your offering! I understand that your objective is a plugin (or output module) for storing (receive) DSM-CC Objects. Right?
This will be useful for testing MHP/HbbTV Apps.

I hope after your contribution someone will help to add support for "injecting" DSM-CC Objects in the Carousel.

piotrserafin, 2018-07-05:
Hi @lars18th

Yes the module I have created in the past was used for downloading HbbTv Apps. It will take some time for rewriting it but I'm starting my work.

I'm not sure but possibly OpenCaster can inject and/or generate carousel.

lars18th, 2018-07-06:
Hi @piotrserafin ,

Thanks a lot for your work! I can't wait to see it.

Regarding the "injection" of the carousel with OpenCaster... well, it's completly abandon-ware. I still use some parts, but only because I don't have a current alternative. The TSDuck is be far a more **robust** and **well designed** project. So I prefer to use it. Perhaps in the (not near) future when GET (aka download) functions of DSM-CC are integrated in TSDuck, someone will try to port/implement DSM-CC INJEST functions to it.

Regards.

piotrserafin, 2020-07-29:
Hello @lelegard

I would like to come back to this topic, due to other activities (wedding, house building) I had no time but finally I can start some work.

I'm planning to approach this in two steps:

1. Tables, Descriptors, BIOP Messages
2. DSMM plugin which will allow to analyze, download, and save locally carousel content.

I've started some work and it can be followed here:
https://github.com/piotrserafin/tsduck/tree/dsmcc

Example from table plugin:

```bash
* DSM-CC UNM, TID 59 (0x3B), PID 272 (0x0110)
  Version: 0, sections: 1, total size: 115 bytes
  - Section 0:
    TIDext: 1 (0x0001) - DSI
    DSM-CC Message Header
    protocolDiscriminator: 0x11 (17)
    dsmccType: 0x03 (3)
    messageId: 0x1006 (4102)
    transactionId: 0x80110001 (2148597761)
    adaptationLength: 0x00 (0)
    messageLength: 0x005B (91)
    serverId: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    compatibilityDescriptorLength: 0x0000 (0)
    privateDataLength: 0x0043 (67)
    IOP::IOR
    type_id_length: 0x00000004 (4)
    typeId: 73 72 67 00                                            srg.
    taggedProfilesCount: 0x00000001 (1)
    profileIdTag: 0x49534F06 (1230196486)
    profileDataLength: 0x0000002B (43)
    Extraneous 47 bytes:
    0000:  00 02 49 53 4F 50 0D 00 01 00 01 00 02 01 00 04  ..ISOP..........
    0010:  53 47 57 00 49 53 4F 40 12 01 00 00 00 16 00 1C  SGW.ISO@........
    0020:  0A 00 01 80 25 01 E1 00 6B 0E 40 00 00 00 00     ....%...k.@....
```
Of course this is work in progress.

I have a question related how to structurize the code. I would like to have dsmcc support as a core part of tsduck, but if you prefer this can be done also as an extension. Current flat tsduck structure brings some complexity because it is required to put everything in one place (eg. descriptors folder with dozens of files). Im using tsDSMCC prefix for all files to group it but maybe it is allowed to introduce more hierarchical folder structure (tables/dsmcc , descriptors/dsmcc)?

lelegard, 2020-08-14:
Hi @piotrserafin 

Thanks for working on this. Concerning the structure, the base classes should be in `libtsduck` (data and utility classes) to be used by other applications when needed. The plugins should be in `tsplugins`.

Now, inside `libtsduck/dtv`, there are many ways to groups classes. It could have been by standards (MPEG, DVB, ATCS, etc), by data type (tables, descriptors), by usage (PSI, SI, MPE, T2-MI, SimulCrypt, DSM-CC, etc) No solution was perfect. Standards and usages often overlap between each other and I sometimes don't know if a class should fall into one category or another. Similarly, users don't always know to which specific standard or usage a class belongs to and it would take time to locate a specific class.

Data types (character sets, tables and descriptors) or very specific usages (DVB SimulCrypt since they are head-end protocols) are rather unambiguous and were isolated in subdirectories. Using `descriptors/dsmcc` would mean recreating a usage-based classification inside `descriptors` and would normally mean that all descriptors should be classified in other subdirectories as well.

So, I simply suggest to leave all descriptors altogether. Prefixing by `tsDSMCC...` is fine as long as the class is named accordingly (`ts::DSMCC...`).

Concerning the implementation of the serialization and deserialization of tables and descriptors, I have started to migrate the cumbersome code using static functions such as `GetUInt16()` and others, explicitly managing addresses and pointers. The code is ugly and error-prone. Instead, I created high-level buffer classes `Buffer` (generic) and its subclass `PSIBuffer`  (more specialized to analyze PSI/SI structures). Have a look at the protected parts of classes `AbstractTable` and  `AbstractDescriptor` for more details. For descriptors, it is possible to replace `serialize()` and `deserialize()` by `serializePayload()` and `deserializePayload()`. They are easier to write and less error-prone (error management is detected in the buffer classes and handled automatically in the caller). You can have a look at a few descriptors which have been migrated, especially `ISO639LanguageDescriptor` or `ServiceDescriptor`. The benefit is obvious. I do not think I will ever migrate all existing descriptors to this new scheme but new descriptors and tables should use it.

Additionally, to debug tables and descriptors and to make sure that no regression is ever introduced, I strongly suggest to add a dedicated test script in the [TSDuck test suite](https://github.com/tsduck/tsduck-test). See for instance `test-063` which tests all ISDB tables and descriptors. The [script](https://github.com/tsduck/tsduck-test/blob/master/tests/test-063.sh) is trivial. The [XML input file](https://github.com/tsduck/tsduck-test/blob/master/input/test-063.xml) can be built incrementally, descriptor after descriptor, with all meaningful structures.

jxxp9, 2023-07-02:
Hey guys, any update on this?

jxxp9, 2023-07-03:
Yeah I'd really like to see that implemented. I'm analyzing some streams looking for Targeted Advertising implementation and looks like SCTE-35 messages can be also embedded in DSM-CC carousels. Implementing them would enable rmsplice and all the other plugins to also parse those SCTE messages (besides all the other interesting info in DSM-CCs)

piotrserafin, 2024-11-08:
Hello @lelegard,

After a period of inactivity, I'm pleased to share that I've resumed work on DSM-CC functionality. Here’s a summary of the progress so far:

# Completed

1. **Added DSM-CC U-N Message Specific Descriptors**

_Tags 0x00-0x0B - Allocated by DVB
Tags 0x70-0x7F - Reserved for DVB MHP_

- type_descriptor (0x01)
- name_descriptor (0x02)
- info_descriptor (0x03)
- module_link_descriptor (0x04)
- CRC32_descriptor (0x05)
- location_descriptor (0x06)
- est_download_time_descriptor (0x07)
- group_link_descriptor (0x08)
- compressed_module_descriptor (0x09)
- SSU_module_type_descriptor (0x0A)
- subgroup_association_descriptor (0x0B)
- label_descriptor (0x70)
- caching_priority_descriptor (0x71)

2. **Implemented DSM-CC Download Data Message (DDB) Support**

I am currently able to gather all DDB sections into a module, though parsing of BIOP (directory, file, etc.) messages is still pending.

3. **Added DSM-CC User-to-Network Message Support**

Both DSI and DII messages are now supported, with parsing for all necessary Object Carousel structures.

Examples:
`tstables -p 0x76A ../../../streams/hotbird_11642H.ts`
[tstables_hotbird11642H.txt](https://github.com/user-attachments/files/17680066/tstables_hotbird11642H.txt)

`tstables -p 0x76A ../../../streams/hotbird_11642H.ts --xml-output hotbird_11642H_dsmcc.xml`
[hotbird_11642H_dsmcc.xml.txt](https://github.com/user-attachments/files/17680096/hotbird_11642H_dsmcc.xml.txt)

And dump from DVBInspector for comparision:
[dvbinspector_hotbird_11642h.txt](https://github.com/user-attachments/files/17680104/dvbinspector_hotbird_11642h.txt)

# Questions

- I’d like to process DDB messages after collecting all sections. My understanding is that in buildXML, tables are complete (with all sections gathered), but in serializePayload, I need to verify that all sections are collected. Should I use table.isValid() for this check?

# To-Do
**Part 1 - Initial PR: Tables/Descriptors Support**

- [ ] Refactor tsDSMCCUserToNetwork: The current structure is a placeholder to validate parsing. I’ll add tests and refactor it into a more OOP-based structure.
- [ ] BIOP Message Parsing and Object Hierarchy Tree: Implement parsing for BIOP messages to build a directory and file structure.
- [ ] Add tsduck-test Tests: Extend test coverage.
- [ ] Robustness: Improve handling, particularly for cases like Object Carousel with SGI vs. SSU Data Carousel with GroupInfo.

**Part 2 - DSM-CC Plugin Support**

- [ ] Enable downloading, optional decompression, and saving of DSM-CC Object Carousel/Data Carousel content to disk.
- [ ] Add carousel search capability based on parameters (PID, Association Tag).
- [ ] Detect carousel type (SSU vs. OC) based on DataBroadcastIdDescriptor in PMT.

# WiP
Progres can be tracked here: https://github.com/piotrserafin/tsduck/tree/feature/dsmccoc

I will keep you informed , feel free to comment but this is still early work in progress stage.

lelegard, 2024-11-09:
Hi @piotrserafin 

You opened this thread six years ago. I have enormous respect for your tenacity! 

> I’d like to process DDB messages after collecting all sections. My understanding is that in buildXML, tables are complete (with all sections gathered), but in serializePayload, I need to verify that all sections are collected. Should I use table.isValid() for this check?

When a table is serialized, either in XML or binary format, the C++ table object (as a subclass of `ts::AbstractTable`) is _always_ valid. Serializing an invalid table always fails in the first place. Your `buildXML()` and `serializePayload()` will never be called. There is no need to check.

There are so many tables and descriptors (several hundredths) that the idea is to group everything that is common code in superclasses and let subclasses (the specific tables and descriptors) focus on their implementation and nothing else. And by the way, you were right to wait six years, the structure was not so clean back in 2018.

So, when your `buildXML()` and `serializePayload()` are called, your table is valid and clean as a C++ object. Just focus on serializing it.

The difference between XML and binary is elsewhere. An XML structure can be arbitrarily large and you can have a straightforward code which generate everything in a clean structure of nested loops for nested structures. Easy.

Binary tables, on the other hand, are structured in sections. And sections have a maximum size. Additionally, not all sections have the same maximum size. Typically, MPEG-defined PSI sections are limited to 1024 bytes while so-called "private" SI sections can grow up to 4096 bytes. And other sections can use specific limitations (implement the virtual method `maxPayloadSize()` if you have a non-standard limitation).

Therefore, when your `serializePayload()` is called, your must be careful to never overflow each section. The superclass cannot do that for you because you are the only one to know how to handle section boundaries in your table, where you should stop a section.

The idea is that you receive a write-buffer with the maximum size of your section. You write into that buffer, taking care of not overflowing the buffer (if you do, this is a bug and the superclass will invalidate the binary table). When you compute that the next data or substructure would overflow the buffer, call `addOneSection()` and continue. The method `addOneSection()` is a magic bullet which tells the superclass "take what's in the buffer, create a new section from it into the final binary table, cleanup the buffer, and give it back to me". This means that, upon return from `addOneSection()`, the buffer is emptied and you may continue serializing data into it, which will go into the next section. You may consider `addOneSection()` as some kind of coroutine.

One more important point. Most tables have a sort of common prolog at the beginning of each section payload (a couple of service id, transport stream id, etc) which are repeated into each section. There is a specific trick for this. At the beginning of your `serializePayload()`, you serialize that common prolog only once and you "push the buffer state" (there is a stack of saved states in a buffer). Then, only focus on the rest. When you call `addOneSection()`, the buffer is not completely emptied, it is restored in the pushed state. Therefore, the common prolog of the next section is already there. No need to serialize it again, just continue to generate payload data.

More details can be found in [ts::AbstractTable::serialize()](https://github.com/tsduck/tsduck/blob/0f22eee316ef7afe09fbb1a02d66ef5bdf1b19c0/src/libtsduck/dtv/tables/tsAbstractTable.cpp#L186).

I have added a section in the developer's guide about [adding PSI/SI tables or descriptors](https://tsduck.io/download/docs/tsduck-dev.html#addpsi). You may also have a look at it.

You probably already did that but you may also look at the code of existing tables. A sufficiently complicated, but not-too-complicated, multi-section table is the [DVB UNT](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/tables/dvb/tsUNT.cpp). If you want to look at a big monster, the [DVB SAT](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/tables/dvb/tsSAT.cpp) is probably the most complicated table. It has been implemented by @paulhiggs. Paul is now quite experienced in implementing tables and descriptors.

Good luck. Let us know your progress.

lelegard, 2024-11-09:
Hi @piotrserafin,

I had a look at your attached text files. Nice job. Please continue. Looking forward to merging this in the baseline.

Did you implement the descriptors as ["table-specific descriptors"](https://tsduck.io/download/docs/tsduck-dev.html#_table_specific_descriptors)?

lelegard, 2024-11-09:
> Did you implement the descriptors as ["table-specific descriptors"](https://tsduck.io/download/docs/tsduck-dev.html#_table_specific_descriptors)?

Answering my own question after looking at your code (I should have started with that), yes you did.

@piotrserafin, just one point, about your code: Please use your name in the copyright header of the new files you created, not mine. This is your code, not mine. You deserve credits for it, not me.

The rule:
- Each source file which is created, entirely or mostly written by a developer shall have the full name of that developer alone in the copyright line.
- If an existing source file is significantly modified, refactored, improved, by a different developer, add the name of that developer in the copyright line, after other developers names.
- For small bug fixes or local improvements in a source file, don't alter the copyright line.

I will make this more clear in the developer's guide.

piotrserafin, 2024-11-09:
Hi @lelegard,

I will, thanks,. Currently on holidays with my family but coming back on Monday.

Have a nice weekend.

lelegard, 2024-11-09:
@piotrserafin, enjoy your holidays 🍹

lelegard, 2024-12-16:
@piotrserafin, if you are still working on this, please note that the API for descriptor registration and display has been slightly changed recently.

This change was necessary because TSDuck was too DVB-centric and implementing private MPEG descriptors based on preceding registration descriptors became impossible. Similarly, the internal management of ATSC and ISDB descriptors was based on a shameful trick. Everything was cleaned up and TSDuck now treats all forms of descriptors equally, with a cleaner interface.

If your code generates errors, have a look at how other descriptors have been changed. Also feel free to ask questions here.

Sorry for the inconvenience.

piotrserafin, 2024-12-16:
Hello @lelegard,

Yeah sill working although unfortunately I have crunch time in work and  busy evenings. Anyway I've already rebased onto https://github.com/tsduck/tsduck/commit/98f69eb45d4e8b394674a64818796a7487727154 couple of days ago and fixed issues. 
I will do another rebase and check. 

I've added tests in tsduck-test for UNM messages and realized that I've made some mistakes in payloadSerialization, after fixing this I will prepare first PR.

Best regards

piotrserafin, 2024-12-17:
Hello @lelegard I've found one issue after rebasing my changes to HEAD of upstream TSDuck. 

Im doint TSTables analysis with

`tstables -p 0x76A ../../../streams/hotbird_11642H.ts`

Result:
```
* DSM-CC UNM, TID 0x3B (59), PID 0x076A (1898)
  Version: 29, sections: 1, total size: 154 bytes
  - Section 0:
    Protocol discriminator: 0x11 (17)
    Dsmcc type: 0x03 (Download message)
    Message id: 0x1002 (DownloadInfoIndication)
    Transaction id: 0xA97D0003 (2843541507)
    Download id: 0x0000000A (10)
    Block size: 0x0FE2 (4066)
    Module id: 0x0001 (1)
    Module size: 0x00000085 (133)
    Module version: 0x7D (125)
    Module timeout: 0x03938700 (60000000)
    Block timeout: 0x03938700 (60000000)
    Min block time: 0x00000000 (0)
    Taps count: 0x01 (1)
    Tap id: 0x0000 (0)
    Tap use: 0x0017 (BIOP_OBJECT_USE (BIOP objects in Modules))
    Tap association tag: 0x000A (10)
    Defining Standards: 0x0001 (1)
    Table Id: 0x3B (59)
    Descriptor List:
    - Descriptor 0: CA (0x09, 9), 5 bytes
      CA System Id: 0x7800 (unknown), CA PID: 0x0001 (1)
      Private CA data (1 bytes): 26
      Reserved bits incorrectly set:
        Byte 4, bit #1 should be '1'
        Byte 4, bit #2 should be '1'
        Byte 4, bit #3 should be '1'
```
It seems it picks up Descriptor (0x09 - CA) from MPEG (0x0001) Standard context which is reasonable because DSMCC UserToNetwork Table is defined in MPEG. My expectation would be that this will be interpreted as [Compressed Module Descriptor](https://github.com/piotrserafin/tsduck/blob/b5368f30277458f1cdd61e255839982330204888/src/libtsduck/dtv/descriptors/dvb/tsDSMCCCompressedModuleDescriptor.cpp#L20). The problem is that those TableSpecific descriptor are [defined in DVB context](https://github.com/piotrserafin/tsduck/blob/b5368f30277458f1cdd61e255839982330204888/src/libtsduck/dtv/signalization/tsDID.h#L191) (and [names](https://github.com/piotrserafin/tsduck/blob/b5368f30277458f1cdd61e255839982330204888/src/libtsduck/dtv/signalization/tsDID.names#L285C1-L301C57)) according to eg. EN 301 192. 

[image]

or maybe it is defined in MPEG context but I misunderstand specification? Assuming I'm correct how to deal with it shall I pass proper context like this?

```
void ts::DSMCCUserToNetworkMessage::DisplaySection(TablesDisplay& disp, const ts::Section& section, PSIBuffer& buf, const UString& margin)
{
...
            DescriptorContext context(disp.duck(), section.tableId(), ts::Standards::DVB);
            disp.displayDescriptorList(section, context, false, buf, margin, u"Descriptor List:", u"None", user_info_length);
...

}
```
instead of passing ` section.definingStandards()` to [DescriptorContext constructor](https://github.com/piotrserafin/tsduck/blob/b5368f30277458f1cdd61e255839982330204888/src/libtsduck/dtv/tables/mpeg/tsDSMCCUserToNetworkMessage.cpp#L565)?

Latest changes on [feature/dsmccoc](https://github.com/piotrserafin/tsduck/tree/feature/dsmccoc) branch.

lelegard, 2024-12-17:
Hi @piotrserafin

You are here at the core of what was modified: the identification and context of descriptors has been made more general. There is a merge of implementation (C++ class) and external naming (.names files).

Have a look at the file `src/libtsduck/dtv/signalization/tsDID.names`. I let you absorb the guidelines in the initial comments. Then, you should add a section for DSM-CC table-specific descriptors.

In your case, you need a line like this:
~~~
0x0002_03_FFFFFF_3B_09 = Compressed Module
     ^  ^         ^  ^
     |  |         |  |
     |  |         |  Descriptor id
     |  |         Table id where the descriptor is valid (up to 4 table ids over 32 bits)
     |  Descriptor type, 03 = table-specific
     Standards bit mask, 0002 = DVB
~~~

- The `.names` files list everything that is known, including things that we don't implement (yet).
- The C++ classes implement the structures we fully support.

When an implementation is provided, the two must be consistent. The 64-bit identifier in `tsDID.names` should match the 64-bit integer in the `EDID` object which is build in you C++ class. Look into `src/libtsduck/dtv/signalization/tsEDID.h`.

If you want to have a look at how your identifier is identified by TSDuck, you can dump the internal state of the PSI Repository using the command `tsversion --xdump-psi-repository` (wide screen display, pipe it into `less -S`).

Hope this helps...

piotrserafin, 2024-12-17:
Yes I did exactly as described in first place 
**tsDID.names:**
```
#
# DVB table-specific descriptors.
# Valid in a DSM-CC UNM (0x3B, DSM-CC User-to-Network Message, ETSI EN 301 192)
#
0x0002_03_FFFFFF_3B_01 = DSM-CC U-N Message Type
0x0002_03_FFFFFF_3B_02 = DSM-CC U-N Message Name
0x0002_03_FFFFFF_3B_03 = DSM-CC U-N Message Info
0x0002_03_FFFFFF_3B_04 = DSM-CC U-M Message Module Link
0x0002_03_FFFFFF_3B_05 = DSM-CC U-N Message CRC32
0x0002_03_FFFFFF_3B_06 = DSM-CC U-N Message Location
0x0002_03_FFFFFF_3B_07 = DSM-CC U-N Message Estimated Download Time
0x0002_03_FFFFFF_3B_08 = DSM-CC U-N Message Group Link
0x0002_03_FFFFFF_3B_09 = DSM-CC U-N Message Compressed Module
0x0002_03_FFFFFF_3B_0A = DSM-CC U-N Message SSU Module Type
0x0002_03_FFFFFF_3B_0B = DSM-CC U-N Message Subgroup Association
0x0002_03_FFFFFF_3B_70 = DSM-CC U-N Message Label
0x0002_03_FFFFFF_3B_71 = DSM-CC U-N Message Caching Priority
0x0002_03_FFFFFF_3B_72 = DSM-CC U-N Message Content Type
```
 
Your case is different as both UNT and INT and terget_IP_address_descriptor are defined in DVB
**tsTID.names:**
```
#
# DVB-defined tables:
#
0x0002_00_40 = NIT Actual
0x0002_00_41 = NIT Other
0x0002_00_42 = SDT Actual
0x0002_00_46 = SDT Other
0x0002_00_4A = BAT
0x0002_00_4B = UNT (Update Notification)
0x0002_00_4C = INT (IP/MAC Notification)
```
**tsDID.names:**
```
#
# DVB table-specific descriptors.
# Valid in an UNT and INT (ETSI TS 102 006 and ETSI EN 301 192):
#
0x0002_03_FFFF_4C_4B_06 = UNT/INT Target Smartcard
0x0002_03_FFFF_4C_4B_07 = UNT/INT Target MAC Address
0x0002_03_FFFF_4C_4B_08 = UNT/INT Target Serial Number
0x0002_03_FFFF_4C_4B_09 = UNT/INT Target IP Address
0x0002_03_FFFF_4C_4B_0A = UNT/INT Target IPv6 Address
```
In my case DSM-CC UserToNetwork Table and Compressed Module Desriptor are defined in different standards (MPEG and DVB).
**tsTID.names:**
```
#
# MPEG-defined tables:
#
0x0001_00_00 = PAT
0x0001_00_01 = CAT
...
0x0001_00_38 = DSM-CC 0x38
0x0001_00_39 = DSM-CC 0x39
0x0001_00_3A = DSM-CC MPE
0x0001_00_3B = DSM-CC UNM
0x0001_00_3C = DSM-CC DDM
...
```

**tsDID.names:**
```
#
# DVB table-specific descriptors.
# Valid in a DSM-CC UNM (0x3B, DSM-CC User-to-Network Message, ETSI EN 301 192)
#
...
0x0002_03_FFFFFF_3B_07 = DSM-CC U-N Message Estimated Download Time
0x0002_03_FFFFFF_3B_08 = DSM-CC U-N Message Group Link
0x0002_03_FFFFFF_3B_09 = DSM-CC U-N Message Compressed Module
0x0002_03_FFFFFF_3B_0A = DSM-CC U-N Message SSU Module Type
...
```

piotrserafin, 2024-12-17:
In `void ts::DSMCCUserToNetworkMessage::DisplaySection` if I'm using explicit standard ''ts::Standards::DVB' instead of `section.definingStandards()` it works correctly.

```
DescriptorContext context(disp.duck(), section.tableId(), ts::Standards::DVB);
disp.displayDescriptorList(section, context, false, buf, margin, u"Descriptor List:", u"None", user_info_length);
```
Result
```
* DSM-CC UNM, TID 0x3B (59), PID 0x076A (1898)
  Version: 29, sections: 1, total size: 154 bytes
  - Section 0:
    Protocol discriminator: 0x11 (17)
    Dsmcc type: 0x03 (Download message)
    Message id: 0x1002 (DownloadInfoIndication)
    Transaction id: 0xA97D0003 (2843541507)
    Download id: 0x0000000A (10)
    Block size: 0x0FE2 (4066)
    Module id: 0x0001 (1)
    Module size: 0x00000085 (133)
    Module version: 0x7D (125)
    Module timeout: 0x03938700 (60000000)
    Block timeout: 0x03938700 (60000000)
    Min block time: 0x00000000 (0)
    Taps count: 0x01 (1)
    Tap id: 0x0000 (0)
    Tap use: 0x0017 (BIOP_OBJECT_USE (BIOP objects in Modules))
    Tap association tag: 0x000A (10)
    Descriptor List:
    - Descriptor 0: DSM-CC U-N Message Compressed Module (0x09, 9), 5 bytes
      Compression Method: 0x78 (120)
      Original Size: 0x00000126 (294)
    Module id: 0x0002 (2)
...
```
For now I can leave it with proper comment as a workaround.

lelegard, 2024-12-17:
The cause is probably the difference of standards, how table-specific descriptors are selected based on the table and the standard. 

In theory, we should say "descriptor X of standard A is specific to table Y of standard B". However, there is not enough room in 64 bits. Or we could rearrange the layout, limiting the number of tables per specific descriptor to 2 instead of 4. So far, I have seen 2 max: INT and UNT. All other table-specific descriptors are specific to one table only.

Do you see a problem with limiting the number of possible tables to 2 per table-specific descriptor?

lelegard, 2024-12-18:
It seems a bit more complicated than that. This is a matter of principle, not a matter of implementation in TSDuck.

Previously, TSDuck was too DVB-centric. DVB was the default for everything, until something different was found or the standard was forced on the command line using `--atsc` or `--isdb` (there was no `--dvb` since it was a univeral default).

This started to be an issue with streams without clear regional standard (not DVB, not ATSC, not ISDB) where all "private" descriptors were based on MPEG-defined registration descriptors. The DVB default interfered with the analysis of the signalization.

This was the core reason for the recent big change. Now, DVB is no longer a default. Only MPEG is assumed and, just like ATSC or ISDB, DVB is selected either when a DVB-specific table is found (a SDT for instance) or when the new option `--dvb` is used. Note that `--dvb` means "DVB only". This makes a difference because ISDB is based on a moving subset of DVB. Since a SDT is used by both, finding a SDT means "it looks like DVB but it may turn out as ISDB later" while `--dvb` means "it's pure DVB, not ISDB".

So, coming back to your case, you have a MPEG-defined DSM-CC table containing a DVB-defined table-specific descriptor. The descriptor reuses the tag 0x09 which is already used by a MPEG-defined descriptor (the very common CA descriptor). This is where the context matters.

- If the context is MPEG-only, then the MPEG-defined descriptor takes precedence. This is normal. We have no idea if DVB, ATSC or ISDB is relevant. An ATSC user will not want a MPEG-defined CA_descriptor be interpreted as your DVB-defined table-specific descriptor.
- If the context includes DVB, then, within the MPEG+DVB context, the table-specific descriptor takes precedence, when found in that table. Precisely because it is "more specific".

So, to translate this in practice:

- If you build a small test where DVB is not asserted, for instance a text XML file containing only the MPEG-defined DSM-CC table, the descriptor tag 0x09 will be interpreted as the MPEG-defined CA descriptor.
- However, when a DVB context is found, either because there a DVB-defined table in the XML file or because option `--dvb` is specified, then the descriptor tag 0x09 will be interpreted as the DVB-defined table-specific descriptor.

To demonstrate this, I have added the following line to my `.names` file to define a DVB table-specific descriptor with tag 0x09 in the PMT:
~~~
0x0002_03_FFFFFF_02_09 = Foo test in PMT
~~~

Note that the PMT is a MPEG-defined table. So, we have the same situation as yours. Let's try to view the influence of the context using the following file.
~~~
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <PMT service_id="1">
    <CA_descriptor CA_system_id="0" CA_PID="100"/>
  </PMT>
</tsduck>
~~~

Using the above XML file, we reproduce the same situation as yours. We need to force a DVB context using `--dvb` to get the DVB table-specific descriptor.
~~~
$ tstabcomp test2.xml 
$ tstabdump test2.bin

* PMT, TID 0x02 (2)
  Section: 0 (last: 0), version: 0, size: 22 bytes
  Program: 0x0001 (1), PCR PID: none
  Program information:
  - Descriptor 0: CA (0x09, 9), 4 bytes
    CA System Id: 0x0000 (unknown), ECM PID: 0x0064 (100)

$ tstabdump --dvb test2.bin

* PMT, TID 0x02 (2)
  Section: 0 (last: 0), version: 0, size: 22 bytes
  Program: 0x0001 (1), PCR PID: none
  Program information:
  - Descriptor 0: Foo test in PMT (0x09, 9), 4 bytes
    0000:  00 00 E0 64                                      ...d

$ 
~~~

However, in the following XML file, the presence of the SDT asserts a DVB context:
~~~
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <SDT transport_stream_id="1" original_network_id="1"/>
  <PMT service_id="1">
    <CA_descriptor CA_system_id="0" CA_PID="100"/>
  </PMT>
</tsduck>
~~~

And the DVB-defined table-specific descriptor is correctly interpreted from the context:
~~~
$ tstabcomp test3.xml 
$ tstabdump test3.bin

* SDT Actual, TID 0x42 (66)
  Section: 0 (last: 0), version: 0, size: 15 bytes
  Transport Stream Id: 0x0001 (1)
  Original Network Id: 0x0001 (1)

* PMT, TID 0x02 (2)
  Section: 0 (last: 0), version: 0, size: 22 bytes
  Program: 0x0001 (1), PCR PID: none
  Program information:
  - Descriptor 0: Foo test in PMT (0x09, 9), 4 bytes
    0000:  00 00 E0 64                                      ...d

$ 
~~~

As a conclusion, I would say that the new behaviour is the right one. We, DVB users, simply must have the same humility as ATSC or ISDB users previously. If the context is not precise, we must say `--dvb`.

So, I suggest that you revert your code in the display of the DSM-CC section, removing the reference to DVB. This is a MPEG-defined table and non-DVB users who use it don't want to have a forced DVB interpretation. However, when you run small test cases where the DVB context is not implicit, use option `--dvb`.

If you don't like typing `--dvb` all the time, there is a little known feature of TSDuck named the [user's configuration file](https://tsduck.io/download/docs/tsduck.html#chap-userconfig) which allows you to define implicit options for some commands. There is a complete example for Japanese users who have a lot of defaults to set (ISDB, ARIB, JST, character sets).

However, in reference test suites, make sure to explictly use `--dvb` so that anyone can run it outside your user's context.

@paulhiggs, you may be interested in this discussion too.

piotrserafin, 2024-12-18:
Hello @lelegard,

The first approach you proposed, limiting the number of possible tables - i think it's another workaround, and I didn't seen any limit in any specification I'v read. 

I've check your proposal with explicit context by passing `--dvb` flag and I can confirm it is working correctly. I understand the your reasoning and agree with it although probably there will be need to be very explicit in the docs that there is no assumption about DVB context. 

Thank you I will revert workaround and consider this topic as solved.

piotrserafin, 2024-12-21:
Hello,

Small update, with newest [change](https://github.com/tsduck/tsduck/commit/efc17dbb93c2542fdb0b8f807926c1d64c74a5db) serialization is fully supported for UNM messages. I've done a lot of test ant **bin <-> xml** transformation works well and always get correct results. 

Next steps:

- [ ] Test also DownloadDataMessage 
- [ ] Code cleanup, removal of magic numbers
- [ ] Cover couple of edge cases eg. in DSI::BIOP_Profile_Body::TAG_ConnBinder I assume that there is only one Tap which is not necessary true. 
- [ ] PRs to upstream both for tsduck and tsduck-test

lelegard, 2024-12-21:
Hi @piotrserafin 

You probably already setup this, but I recommend that you add a dedicated test in the TSDuck test suite. See the [contributor workflow](https://tsduck.io/download/docs/tsduck-dev.html#contributor) in the developer's guide.

Pick up the last free test number (`test-178` as of today) and implement your test. In the case of new signalisation, the test code is simple:
~~~
#!/usr/bin/env bash
source $(dirname $0)/../common/testrc.sh
test_cleanup "$SCRIPT.*"
source "$COMMONDIR"/standard-si-test.sh $SCRIPT.xml
~~~

Your test is only the `input/test-XXX.xml`. Place all possible configurations of tables here, including corner cases. The `standard-si-test` will crunch your input in all possible ways (compilation, decompilation, binary, XML, JSON). The first time, please check carefully the result. The idea is that if you fork under your name the two repo `tsduck` and `tsduck-test`, the Github CI will test your code against your version of the test suite. After merging the code, your test will stay forever, each push.

Sorry for the repetition if you already did it.

piotrserafin, 2025-01-07:
Thanks! Works perfectly now

```
----  test-179: Test test-179.compile.bin passed
----  test-179: Test test-179.compile.log passed
----  test-179: Test test-179.decompile.xml passed
----  test-179: Test test-179.decompile.log passed
----  test-179: Test test-179.decompile.strict.xml passed
----  test-179: Test test-179.decompile.strict.log passed
----  test-179: Test test-179.decompile.json passed
----  test-179: Test test-179.decompile.json.log passed
----  test-179: Test test-179.compile.json.bin passed
----  test-179: Test test-179.compile.json.log passed
----  test-179: Test test-179.cmp.json.log passed
----  test-179: Test test-179.pack.ts passed
----  test-179: Test test-179.pack.log passed
----  test-179: Test test-179.tstables.txt passed
----  test-179: Test test-179.tstables.xml passed
----  test-179: Test test-179.tstables.log passed
----  test-179: Test test-179.tstables.strict.xml passed
----  test-179: Test test-179.tstables.strict.log passed
----  test-179: Test test-179.tstabdump.txt passed
----  test-179: Test test-179.tstabdump.log passed
```

piotrserafin, 2025-01-08:
Hello @lelegard,

With following change: https://github.com/piotrserafin/tsduck/commit/c2738a79bf5efa86a7ee094c1703a57aee1e39ba serialization works for both tables - User-to-Network (DSI/DII) and DownloadDataMessage. 

I've create initial tests here: https://github.com/tsduck/tsduck-test/compare/master...piotrserafin:tsduck-test:feature/dsmccoc

From feature point of view, **Part 1 - Initial PR: Tables/Descriptors Support** is finished. 

Today/tomorrow I will focus on cleanup and refactoring (I have now the way to check for regressions), so expect Part 1 PRs later this week.

Next step will be:

**Part 2 - DSM-CC Plugin Support**
-  Enable downloading, optional decompression, and saving of DSM-CC Object Carousel/Data Carousel content to disk.

piotrserafin, 2025-01-11:
## Part 2 - DSM-CC Plugin Support plan

Initial implementation plan

- [ ] plugin will be structure similar way as it is done for teletext plugin 
  - [ ] new folder under `tsduck/src/libtsduck/dtv/dsmcc` containing:
    - [ ]  BIOP Messages classes
    - [ ]  HandlerInterface
    - [ ]  Utilities/FileOps
    - [ ] Maybe dedicated Demux based on tsSectionDemux
- [ ] new `tsDSMCCPlugin` class under `tsduck/src/libtsduck/plugins/plugins`
- [ ] new library dependency - **zlib** for uncompressing modules
- [ ] Tests

Assumptions:

- Plugin should allow to pretty print BIOP Messages and carousel hierarchy
- Input params:
  - PID
  - Carousel Type (Data Carousel, ObjectCarousel, SSU) - for now only support for ObjectCarousel
- Output:
  - tar.gz/zipped file with OC content (dir structure)

lelegard, 2025-01-11:
Hi @piotrserafin,

Thank you for your contribution so far. And thank you for going on with part 2. About this part, there are two points that should raise your attention. Because they are important and may be useful later to other contributors, let me present them with some clarity.

### Embedded or independent plugin

The first point is the status of your plugin (or plugins if more than one may handle DSM-CC). The idea of TSDuck is that all generic features are available to all applications and plugins in the form of C++ classes in libtsduck. The applications are separate executables with source code in `src/tstools`. The plugins are separate shared libraries with source code in `src/tsplugins`.

In some specific cases, either for good technical reasons of library dependency or for stupid historical reasons, a _limited number of plugins_ are embedded in the TSDuck library with source code in `src/libtsduck/plugins/plugins`.

In your check list, you mention the two options: _"Create a source file named tsplugin_dsmcc.cpp in the tsplugins subdirectory, Run Python script scripts/build-project-files.py"_ and later _"new tsDSMCCPlugin class under tsduck/src/libtsduck/plugins/plugins"_.

The two options are incompatible, you have to choose one. I would recommend the external shared library approach in `src/tsplugins`. Of course, any generally useful DSM-CC C++ class should be placed in the library and `tsduck/src/libtsduck/dtv/dsmcc` is a natural choice. These classes will be used by the external plugin.

### Dependency to an external library

The second point the new dependency to an external library, zlib. I have no objection about that but there are a number of steps to follow. TSDuck is supposed to be OS-agnostic and provides a consistent interface on all operating systems, Linux, macOS, BSD, Windows. In the case of an external library, the first question is "how to handle the build on each OS" and the second question is "how to handle the deployment and run on each OS". There is no general rule, it is always case by case.

#### Build requirements on all operating systems

On the build side, TSDuck is willing to be easy to build on all OS. I have suffered too much with open source projects which were easy to build on Linux and a nightmare elsewhere. I don't want this. The scripts `install-prerequisites` (`.sh` on UNIX systems, `.ps1` on Windows) must transparently install the required development libraries on each OS. So, you need to find out how to install the zlib development library on all of them. The UNIX script is a large if-else-if structure with 14 different flavours of systems, with sometimes some variants depending on the OS version.

If you can tell me want you need on one distro, I can help for the rest. My humble Windows laptop runs 15 virtual machines using VirtualBox, all different operating systems, 11 Linux distros and 4 BSD. I can update `install-prerequisites.sh`.

On macOS, the first question is "does it come with macOS or with HomeBrew". In the latter case, the second question is "how do I update the HomeBrew formula" which is in the HomeBrew repository, not in the TSDuck project. Again, I can help on that.

The worst problem is (as usual) on Windows. How do you interface with zlib on Windows? Do you have to use the open source library (and then how do you bring its binary on Windows?) or is there an equivalent native Microsoft library on every Windows? As an example, for specific protocols such as SRT or RIST, I had to bring the open source library on Windows. However, for general-purpose features such as Web client and cryptography, I opted for the native option: libcurl and OpenSSL on UNIX systems, WinInet and BCrypt on Windows. For a generic feature such as compression, I wouldn't be surprised if a native library exists on Windows. But which one and how to use it?

#### Build option to disable the dependency

Any external dependency may be a problem for people building TSDuck on small or embedded system without everything. Therefore, there must be a `make` option to disable it ([see details here](https://tsduck.io/download/docs/tsduck-dev.html#buildopt)). In your case, it should be something like `NOZLIB`.

Note that you should care only on UNIX systems. On Windows, the vast majority of people use the binary which are built my me or the GitHub CI and we don't care about reduced versions without dependencies.

Of course, removing a dependency eliminates all TSDuck features which depend on it. For instance, building with `NOCURL` eliminates everything that access the Internet, starting with plugins `http` and `hls`. That's the user's choice. Usually, when they build TSDuck for a reduced system, they have a limited number of usages in mind. In your case, using `NOZLIB` will eliminate part or all of DSM-CC support. That's life...

So, you have to carefully isolate the calls to the external library in some classes with conditional compilation inside the body. Typically, when `NOZLIB` is defined, this class should return errors for all calls. You may search for `NOSRT` and `NORIST` as examples.

#### Packaging issues

TSDuck is also meant to be simple to deploy and install. Several packaging formats are proposed. When you introduce a dependency on an external library, there are consequences on packaging. In your case, this is mainly `rpm` and `deb`. There is also some support for `pkgconfig` in `src/libtsduck/config`.

[... cut ...]  `NORIST`.

As a conclusion, I would say that the new dependency on zlib is probably to worst part of your work. You may write thousands of lines of code, this is your code, your responsibility, your talent. But adding a dependency to an external library means dealing with tons of shit which were accumulated by various people with various skills in many different environments.

I have some experience with that and I can help you on this matter. Just explain clearly what you need and I will help.

piotrserafin, 2025-01-12:
Hello @lelegard ,

Thank you for clarification. 

> The two options are incompatible, you have to choose one. I would recommend the external shared library approach in `src/tsplugins`. Of course, any generally useful DSM-CC C++ class should be placed [...]

Yes, I've only briefly checked and misunderstood purpose of `src/libtsduck/plugins/plugins`, now it's clear and will follow your suggestion. 

When in comes to zlib, thank you for very details approach description it will be very useful in later stages. Zlib is available on all platforms, but this is not the only option, there are header-only DEFLATE implementations that can also be used.

lelegard, 2025-01-12:
Do you need deflate only or deflate and inflate ?

Could you tell what is your main development distro and which package you need to build with zlib ? Then, I could check how to generalize it.

lelegard, 2025-01-13:
@piotrserafin, I have investigated the packages for zlib and it is possible to have it everywhere, using the pristine zlib on Linux, macOS, BSD, and using a header-only implementation on Windows.

I have saved my research in https://github.com/lelegard/zlib-everywhere

A test program validates that zlib and the header-only implementation are interoperable.

I can provide a Zlib API inside the TSDuck library with two services: deflate and inflate, one-shot, processing a complete message. Is this sufficient for your implementation of DSM-CC?

piotrserafin, 2025-01-13:
Hello @lelegard ,

Thank You, you saved me a ton of time. 

I would say that we will need both. Deflate for compressed modules and inflate in two potential use cases:

- dsmcc plugin could store carousel content as single compressed file
- in distant future 😅 I would like also to extend dsmcc support with carousel generation capabilities where support for compressing modules would be nice.

In summary, what you proposed is sufficient.

lelegard, 2025-01-15:
Hello @piotrserafin,

You now have zlib-compatible compression and decompression services in class [`Zlib`](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/base/zip/tsZlib.h).

piotrserafin, 2025-01-15:
Hello @lelegard ,

Thank you, I will perform the test with an  uncompressing real module data.

Meanwhile, progress status:

- I've done some refactor, and added new 'AbstractDSMCCTable' with common parts for both UNM and DDM tables.
- I'm moving some inner classes (IOR, Tap) from UNM to an external location. Those will be also used by BIOP messages
- I've created a skeleton dsmcc plugin, now I can provide a PID parameter and the plugin can catch all necessary tables.

That's all for now.

piotrserafin, 2025-01-17:
Hello @lelegard,

I have a question regarding the setup of demux and filtering to ensure the proper processing order of DSM-CC messages. While this is not strictly required, the cleanest approach would involve receiving and processing UNM (DSI/DII) messages first before collecting DDM messages, as the latter can arrive in any order.

Here is my current approach:

1. Setting Up the Carousel PID:
In the `Plugin::start` method, I set the correct carousel PID using `_demux.addPid(_pid)`.

2. Handling Tables in `Plugin::handleTable`:

- I perform several checks to determine if the table is a UNM message. Based on the results, I set the `_dsi_found` and `_dii_found` flags.
- If the message is a DDM, I return from the method without processing it.

3. Resetting the Demux:
Once both `_dsi_found` and `_dii_found` conditions are met, I call `_demux.reset()`.

4. Processing DDM Messages:
After the reset, the table handler can detect that DSI and DII messages have already been processed, allowing it to proceed with DDM processing. UNM message processing is skipped based on the status of the flags.

I'm wondering what are your thoughts about this approach?

Best regards,
Piotr Serafin

lelegard, 2025-01-17:
> I'm wondering what are your thoughts about this approach?

If the goal is a one-shot capture of a carousel from an endless stream (or at least two carousel cycles long), this should work.

There are possibilities to go further but this may be out of scope, at least for an initial version.

- In terms of optimisation, if you jump into the stream in the middle of a cycle, you receive DDM that you may buffer until you know how to use them, after DSI and DDI. This means that you can grab the whole carousel in one cycle instead of two.
- If you want to monitor new carousels, the reset of the demux may cause any issue. After the reset, you have to ignore the DSI and DDI because you assume that they are the same as the ones you grabbed in the first phase. But maybe not. Maybe it is a new carousel.

Probably out of scope, as I wrote...

> And maybe more general question, is there cannonical way in TSDuck to specify (eg, tid, tid_ext) and attach filter to demux channel?

No, there is not. The demux is just a demux, not a section filter. The filtering shall be done in the application, at the earliest point in methods `handleTable()` or `handleSection()`.

This is what commands such as `tstables` are doing when they need to filter sections. In practice, they use the class `TablesLogger` which uses instances of `TablesLoggerFilterInterface` to filter sections. The purpose of  `TablesLoggerFilterInterface` is to filter sections based on some private criteria and the purpose of `TablesLogger` is to log or display the filtered sections in many possible ways. But this is just an example, too complicated for your case.

In the hardware demux of earlier set-top boxes, for efficiency reasons in these small chips, you can filter tid and tid ext. The section header is analyzed directly inside the TS packets and non-relevant packets are ignored. In the case of a large software running on a regular computer such as TSDuck, this is not necessary, we fully demux all sections and we only use the one we need.

piotrserafin, 2025-01-17:
> * In terms of optimisation, if you jump into the stream in the middle of a cycle, you receive DDM that you may buffer until you know how to use them, after DSI and DDI. This means that you can grab the whole carousel in one cycle instead of two.

This is exactly what I've meant when wrote "While this is not strictly required", thank you, clear for me.  

> * If you want to monitor new carousels, the reset of the demux may cause any issue. After the reset, you have to ignore the DSI and DDI because you assume that they are the same as the ones you grabbed in the first phase. But maybe not. Maybe it is a new carousel.

For this purpose monitoring of transaction Id is required (see [ETSI TR 101 202 V1.2.1 (2003-01) 4.6.5](https://www.etsi.org/deliver/etsi_tr/101200_101299/101202/01.02.01_60/tr_101202v010201p.pdf). 
But for now I agree that one-shot capture is enought for first version, monitoring can come later.

Thanks.

piotrserafin, 2025-01-24:
Hello @lelegard ,

Just letting know that I'm still alive, suffering of lack of time but slowly progressing. Managed to check correctness of a modules after decompression with service you prepared. Works perfectly. I will come back to work in the next couple of days.

lelegard, 2025-05-07:
Hi @piotrserafin 

Just to let you know that, because they are used in several distinct structures, I have refactored a common code and common XML definition for the DSM-CC structures `compatibilityDescriptor()` and `Tap()`. 

The modifications have been propagated to your class `DSMCCUserToNetworkMessage`. The only visible artefact is that all tap structures in the XML now use `<Tap>`.

piotrserafin, 2025-05-08:
Hello @lelegard 

It',s been a while — I've been really busy with work and life. Thanks, the changes look good!

I have some WIP updates where I extracted common structures for dsmcc-oc (e.g., the message header) into an AbstractDSMCC class. I also exposed Tap as a separate class — though I see you've already done that too.

I'll be returning to the plugin soon. The BIOP structures (srg/dir) are currently a work in progress, and I've managed to gather all the necessary tables. What's still missing is parsing the modules, building the object tree, and saving it to disk.

lelegard, 2025-05-08:
Hi @piotrserafin 

I have added the DST table from ATSC A/90 (ATSC data download) which uses the compatibilityDescriptor() and Tap() DSM-CC data structures. This is why I needed these structures are reuseable classes and XML models.

Now, I will try to implement the NRT table, from ATSC A/90 too. This one uses the dsmccResourceDescriptor() DSM-CC data structure. This one is also defined in ISO 13818-6. However, ISO 13818-6 is very unclear about its binary representation.

The definition is:
~~~
dsmccResourceDescriptor {
    commonDescriptorHeader()
    resourceDescriptorDataFields()
}
~~~

The representation of commonDescriptorHeader() is pretty clear and detailed in both ISO 13818-6 and ATSC A/90.

The representation of resourceDescriptorDataFields(), on the other hand, is rather mysterious.
- In ATSC A/90, they say that the _"resourceDescriptorDataFields structures are defined in section 12.3.3, 12.3.4, 12.3.5, 12.3.6 and 12.3.7"_, which are non-existent because the chapter 12 stops after 12.3.2.
- In ISO 13818-6, it is defined in section 4.7.1, table 4-70. However, this diagram uses fields which seem to come out of nowhere. There is a Figure 4-5 which may define them but which is empty.
- ETSI TR 101 202 does not reference resourceDescriptorDataFields() at all.

I wonder how can people implement such poorly specified data structure. 

Would you have by any chance a precise representation of resourceDescriptorDataFields() somewhere?

paulhiggs, 2025-05-08:
I have a copy of the FDIS for the standard from November 1996 if that helps.
There were also some Ammendments and Corrigendum until 2002 but nothing in relation to clause 4.7

lelegard, 2025-05-09:
Hi @piotrserafin 

I have implemented dsmccResourceDescriptor() with resourceDescriptorDataFields() as a raw binary blob for now. Because it is preceded by a length field, not having its precise syntax is not a showstopper. If we can get a precise binary layout of resourceDescriptorDataFields() later, we will update the code and the XML model.

I have completed the implementation of the two ATSC tables, DST and NRT, which contain DSM-CC substructures. I will therefore stop working on DSM-CC structures and let you continue, if you have time.

piotrserafin, 2025-10-06:
Another come back after child birth:

Roadmap: https://github.com/users/piotrserafin/projects/4