tsduck/tsduck #84: IP (Input) - Source specific multicast (SSM) support missing

piotr-serafin-epam, 2018-07-04:
Source Specific Multicast (SSM) requires the multicast group and the source IP to be specified. The format therefore is 
```
udp://<src ip>@<group address>:<port>
```
I was trying to perform it with tsduck
```
tsp -I ip 10.15.216.54@232.5.1.15:49410 -O drop
```
It doesn't understand this format. 

What is working for me is usage of vlc initially for sending proper join messages
```
cvlc udp://10.15.216.54@232.5.1.15:49410
```
and then
```
tsp -I ip 232.5.1.15:49410 -O drop
```
Is there any way to do it with tsduck without vlc?

piotr-serafin-epam, 2018-07-04:
No, unfortunately it is not working.

lelegard, 2018-07-04:
OK, I probably misunderstood what you needed. The command I indicated was intended to receive multicast from group 232.5.1.15, port 49410, for which the source address (the sender) is 10.15.216.54.

I am not familiar with SSM. Is it different from what I described? Does it require specific protocol features?

lelegard, 2018-07-04:
OK, I got it. You are right, SSM is not supported yet. Will do, stay tuned...

lelegard, 2018-07-05:
SSM support has been implemented. Please pull and rebuild to test.

Note that I could not really test it since, as far as I can understand, there is no difference between ASM and SSM if all systems are on the same LAN, without intermediate routers. Am I right? At least, this is what I observe.

So, the price you will have to pay for SSM support is to test it and confirm that it works (or not...) Please test on Windows AND Linux since the two systems behave differently with multicast (the `bind` operation, for instance, does not have the same semantics in the two systems in the case of multicast). So, testing it on one system does not mean it will work on the other. And if you have Macs, I am interested in getting the results too (although I would not be surprised if they are similar to Linux).

piotr-serafin-epam, 2018-07-06:
Hello @lelegard

I've compiled and checked against Windows 10 and Lubuntu 18.04. Works great! Tried to join SSM group dozen of times and always succeeded.

lelegard, 2018-07-06:
Thanks @piotr-serafin-epam 
I plan to release 3.13 next week.
I close this issue. Please open a new one if there is a specific problem about SSM.

closed 2018-07-06 by lelegard