tsduck/tsduck #85: NIT table from scratch

lars18th, 2018-07-04:
Hi,

I hope someone can help me with this problem:

- I'm getting some SPTS from two sources types: tuners, so they come with all PSI tables; and from HLS sources, without DVB PSI tables. The idea is to create one MPTS with all programs and modulate it.

The problem is when "filtering" and "generating" the PSI tables. For example, I'm using this chain for it:

```
tsp -r \
 -I http \"http://192.168.1.100:80/ch1.ts\" \
 -i --ignore-errors --reconnect-delay 100 --connection-timeout 5000 --max-queue 9000 \
 -P pat -v 2 -u \
 -P tsrename -t 100 -o 65000 --add-nit \
 -P nit -v 2 -s 1 -l 1 \
 -P pmt -s 11111 \
 -P svrename 11111 -i 10001 -l 1 -n \"CHANNEL-1\" \
 -P sdt --cleanup-private-descriptors \
 -P regulate -p 32 \
...
```

The Input has only one program from a MPTS without all not needed pids (pid filtering active in the tuner).
So, with this chain I set my channel id to "10001", with the TS-id "100", in the Network "65000". And the output is correct even with the EIT data.
What's the problem? The missing NIT table!
I know that with TSDuck some tools exist for compile PSI tables from XML files. However, I miss some options in the plugins for _generating_ simple tables.

For example, with the `-P pat -u` the NIT table is removed. And I really need to remove any present in the input stream. However, with `-P tsrename --add-nit` I expect to _create an empty_ NIT table. So, with the `-P nit` the idea is to add info for this table.
However, neither the `tsrename` nor the `nit` can create new NIT table and complete data for it.

My objective is processing in the same way inputs with _and_ without PSI tables. And create only a minimal version of them... without the need of store files, but using only processing filters.

This can be achived? Any  idea of how to do it?
Thank you to all!

lelegard, 2018-07-05:
> For example, with the -P pat -u the NIT table is removed.

The `pat` plugin only modifies the PAT. The option `--remove-nit` (why did I used `-u` as short form, can't remember, really) removes the NIT _entry_ in the PAT, not the NIT itself. To remove the NIT, you should simply filter out PID 16.

If you remove the NIT entry from the PAT, the resulting stream is no longer DVB-compliant (although I doubt that this will change anything in receivers).

What I would suggest is the following (in order of plugins) :

- Filter out PID 16 to get rid of a previous NIT (it this is what you want) and replace it with stuffing to preserve bitrate.
- Force the presence of the NIT PID 16 in the PAT (instead of removing it).
- Inject a basic empty NIT from a fixed XML file.
- Then all subsequent plugins will add stuff in your basic empty NIT

But if you want to keep an existing NIT and inject a new one only if the previous one did not exist, then it won't work.

lars18th, 2018-07-05:
Hi @lelegard ,

Thank you! Good explanation. :wink: 

However, one question:
- It's possible to create a simple NIT without injecting a basic NIT from a fixed XML?

The idea is do all stuff running a _large_ command line, instead of mixing files and binaries.

lelegard, 2018-07-05:
If your constraint is to provide one single script file and not some complicated package, you can do something like this:

~~~
NITFILE=/tmp/nit.$$.xml

cat >$NITFILE <<EOF
<?xml version="1.0" encoding="UTF-8"?> 
<tsduck>
  <NIT network_id="0"/>
</tsduck>
EOF

tsp ... \
    -P filter --negate --pid 16 \
    -P inject $NITFILE --pid 16 --stuffing --bitrate 2000 \
    ...

rm -f $NITFILE
~~~

lars18th, 2018-07-05:
Thank you @lelegard !

However, note that only for setting the network id in the NIT table, we need to:
1. Use scripting shell.
2. Generate one XML file.
3. Insert it in the tsp chain.

What you think about this "simple" enhancement of the `nit` plugin?
- `-P nit -c | --create value`: It adds a new NIT table every _value_ milliseconds.

Then using `-P tsrename -o 65000` we can set the original-network-id.

You agree?

lars18th, 2018-07-06:
Hi @lelegard ,

What you think about this idea?

- Include inside the source code of the TSDuck one simple version of each DVB PSI tables in XML format. And provide to filters like `nit`, `pat`, etc. a parameter to **insert** one of such tables at some interval.

The objective is to remove the requirement of _external_ files when modifiying some input stream to be DVB compilant. So using only plugins the stream can be adapted with just the minimal info to pass the specifications.

Do you think it's useful?

lars18th, 2018-07-06:
Hi @lelegard ,

Please, help me with this "simple" problem!

When using your suggestion of `-P inject` from the XML file, I found a problem: The incoming stream doesn't have any NIT table. Futhermore, it doesn't have any padding. So the idea of "filter" the pid 16 to obtain null packets and then replace the null packets with NIT tables doesn't work!

What's the most effective solution to _insert new NIT packets to any VBR stream **without** padding_?

Thank you in advance!

lelegard, 2018-07-06:
If the input bitrate does not matter, I would suggest to use the global option `--add-input-stuffing` to insert an arbitrary amount of stuffing at input level, interspersed with the actual input. You can use that stuffing to insert whatever you want. And a final `filter` plugin can remove the remaining stuffing if you like.

If the input is CBR and this bitrate matters, you must delete something, typically a service or a data PID and replace it with stuffing.

lelegard, 2018-07-06:
>  However, quite brute as I need to add N nulls every M packets.

Brutal but efficient. I use it all the time in my scripts. At least the injection plugins can find stuffing packets to replace quite often.

> And regarding the idea of "simple hardcoded XML tables"... What you think?

Concerning the format, no need for "hardcoded XML". All tables and descriptors are C++ classes. XML is just an external representation of C++ objects. Declaring an object of the appropriate class using the default constructor is enough.

Concerning the feature, yes, it can be added in the future.

lars18th, 2018-07-06:
Hi @lelegard ,

>Brutal but _efficient_. 

I don't doubt it! TSDuck is an impresive project! :+1: 

> Concerning the feature, yes, it can be added in the future.

OK. One more thing to the TODO list. :smile:

lars18th, 2018-07-11:
Hi @lelegard ,

Regarding the creation of a DVB compatible Transport Stream from scratch, I started to use this command:

```
tsp -r --bitrate 16085561 -I null -P regulate \
 -P inject pat.xml=100  --pid  0 -s -b 1024 \
 -P inject nit.xml=2000 --pid 16 -s -b 1024 \
 -P inject sdt.xml=500  --pid 17 -s -b 1024 \
 -P inject tdt.xml=1000 tot.xml=15000 --pid 20 -s -b 1024 \
   -P timeref --start $(date +%Y/%m/%d:%H:%M:%S) \
   -P pat -v 1 -n 16 \
   -P nit -v 1 \
   -P tsrename -t 1 -o 65535 --add-nit \
 ...
```
This creates an "empty" TS with basic tables (despite the bug of tables interval described at #94; and the non-portable run because the use of the `time` command as script commented at #96).

The content of these "basic tables" are in this case:

- PAT.XML:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <PAT transport_stream_id="0"/>
</tsduck>
```
- NIT.XML:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <NIT network_id="0"/>
</tsduck>
```
- SDT.XML:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <SDT original_network_id="0" transport_stream_id="0"/>
</tsduck>
```
- TDT.XML:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <TDT UTC_time="1970-01-01 00:00:00"/>
</tsduck>
```
- TOT.XML:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <TOT UTC_time="1970-01-01 00:00:00"/>
</tsduck>
```

And even though it works, it seems rather cumbersome to me.

So, I suggest this "small" addition to the `inject` plugin:
 ```
 -P inject :empty-pat=100  --pid  0 -s -b 1024 \
 -P inject :empty-nit=2000 --pid 16 -s -b 1024 \
 -P inject :empty-sdt=500  --pid 17 -s -b 1024 \
 -P inject :empty-tdt=1000 :empty-tot=15000 --pid 20 -s -b 1024 \
 ```
That's add five _constants_ as these default tables: `:empty-pat`, `:empty-nit`, `:empty-sdt`, `:empty-tdt` and `:empty-tot`. The use of ":" at the start indicates the use of a hardcoded value, instead of loading a file. This will simplifies the process without the need of any external file.

What you think about this simplification?

lelegard, 2018-08-17:
For consistency with the plugin `cat`, the options  `--create`, `--create-after`, `--bitrate` and `--inter-packet` were just added to plugins `pat`, `pmt`, `bat`, `sdt`, `nit`.

lelegard, 2018-09-11:
Version 3.14 is out, including the new options creating empty tables if they do not exist on input. Closing this issue.

closed 2018-09-11 by lelegard

lars18th, 2018-09-20:
Thank you @lelegard !
This simplifies a lot. :smile: