tsduck/tsduck #86: Errors with tsp with both Ericsson and Elemental IP input

zuzzurro, 2018-07-05:
Hi, I would like to restart the debugging session we had in the #51 issue that drifted away..

I have spent some time setting up a decent test environment free of potential issues and now I restarted testing.

The situation is the following:
Generating a multicast stream, using the various settings you had suggested in #51 still creates issues at the receiver side.
The generator is a linux server connected to the same gigabit network as the receivers (potentially the same switch, but I will have to double check that) and the network is probably very quiet.
In addition to the Ericsson rx8200 receiver that gives me only cryptic errors (TS lock) i now have an Elemental encoder that is quite chatty with error logs that I will paste here.
I also ran tsp in analyse mode and the outputs are attached here.
Finally a similar multicat program works flawlessly with the Elemental (0 error logs).

[analyze_mc.txt](https://github.com/tsduck/tsduck/files/2166189/analyze_mc.txt)
[analyze.txt](https://github.com/tsduck/tsduck/files/2166190/analyze.txt)

analyze_mc is run against the multicat stream.

the relevant command lines are:

sudo chrt -f 99 tsp --max-input-packets 4 --max-flushed-packets 4 -v -I file -i match_61_new3_edit.ts -P pcrbitrate -P regulate -p 7  -O ip 232.0.2.51:5001 -l 172.31.247.13

vs

sudo multicat -i 99 -U match_61_new3_edit.ts 232.0.2.51:5001@172.31.247.13

A sample from the Elemental logs (only the tsp case is provided because there are 0 errors with multicat):

```
2018-07-05 10:21:41 W Dmx[1029.1] [0] PCR to system clock association reset: [340.595] msec, near pkt 15519564, byte 2917678032
2018-07-05 10:21:50 W [sequence-sequence-1770-drc:selector-1030-track-1-drc] crossed frame sync discontinuity. Offset at [00:37:21.412] is [00:34:25.234] previous offset was [00:34:25.467].  RAW 387619200, TA [01:11:46.880], FA [00:37:21.412]
2018-07-05 10:21:50 W [sequence-sequence-1770-drc:selector-1030-track-2-drc] crossed frame sync discontinuity. Offset at [00:37:21.412] is [00:34:25.234] previous offset was [00:34:25.467].  RAW 387619200, TA [01:11:46.880], FA [00:37:21.412]
2018-07-05 10:21:50 W [sequence-sequence-1770-drc:selector-1030-track-3-drc] crossed frame sync discontinuity. Offset at [00:37:21.412] is [00:34:25.234] previous offset was [00:34:25.467].  RAW 387619200, TA [01:11:46.880], FA [00:37:21.412]
2018-07-05 10:21:50 W [sequence-sequence-1770-drc:selector-1030-track-4-drc] crossed frame sync discontinuity. Offset at [00:37:21.412] is [00:34:25.234] previous offset was [00:34:25.467].  RAW 387619200, TA [01:11:46.880], FA [00:37:21.412]
2018-07-05 10:21:50 W [sequence-sequence-1770-drc:selector-1030-track-5-drc] crossed frame sync discontinuity. Offset at [00:37:21.412] is [00:34:25.234] previous offset was [00:34:25.467].  RAW 387619200, TA [01:11:46.880], FA [00:37:21.412]
2018-07-05 10:21:50 W Dmx[1029.1] 1 new PCR errors in the last 5 seconds (78 total), showing 1
2018-07-05 10:21:50 W Dmx[1029.1] [0] PCR to system clock association reset: [-261.903] msec, near pkt 15578191, byte 2928699908
2018-07-05 10:22:09 W [sequence-sequence-1770-drc:selector-1030-track-1-drc] crossed frame sync discontinuity. Offset at [00:37:40.675] is [00:34:24.766] previous offset was [00:34:25.234].  RAW 389331840, TA [01:12:05.909], FA [00:37:40.675]
2018-07-05 10:22:09 W [sequence-sequence-1770-drc:selector-1030-track-2-drc] crossed frame sync discontinuity. Offset at [00:37:40.675] is [00:34:24.766] previous offset was [00:34:25.234].  RAW 389331840, TA [01:12:05.909], FA [00:37:40.675]
2018-07-05 10:22:09 W [sequence-sequence-1770-drc:selector-1030-track-3-drc] crossed frame sync discontinuity. Offset at [00:37:40.675] is [00:34:24.766] previous offset was [00:34:25.234].  RAW 389331840, TA [01:12:05.909], FA [00:37:40.675]
2018-07-05 10:22:09 W [sequence-sequence-1770-drc:selector-1030-track-4-drc] crossed frame sync discontinuity. Offset at [00:37:40.675] is [00:34:24.766] previous offset was [00:34:25.234].  RAW 389331840, TA [01:12:05.909], FA [00:37:40.675]
2018-07-05 10:22:09 W [sequence-sequence-1770-drc:selector-1030-track-5-drc] crossed frame sync discontinuity. Offset at [00:37:40.675] is [00:34:24.766] previous offset was [00:34:25.234].  RAW 389331840, TA [01:12:05.909], FA [00:37:40.675]
2018-07-05 10:22:11 W Dmx[1029.1] 1 new PCR errors in the last 5 seconds (79 total), showing 1
2018-07-05 10:22:11 W Dmx[1029.1] [0] PCR to system clock association reset: [-453.561] msec, near pkt 15722827, byte 2955891476
2018-07-05 10:22:36 W Dmx[1029.1] PCR clock skew [459] msec
2018-07-05 10:23:00 W Dmx[1029.1] 1 new PCR errors in the last 5 seconds (80 total), showing 1
2018-07-05 10:23:00 W Dmx[1029.1] [0] PCR to system clock association reset: [217.203] msec, near pkt 16106254, byte 3027975752
2018-07-05 10:23:36 W Dmx[1029.1] 1 new PCR errors in the last 5 seconds (81 total), showing 1
2018-07-05 10:23:36 W Dmx[1029.1] [0] PCR to system clock association reset: [-63.088] msec, near pkt 16384524, byte 3080290512
2018-07-05 10:25:39 W Dmx[1029.1] PCR clock skew [-256] msec
2018-07-05 10:26:05 W Dmx[1029.1] 1 new PCR errors in the last 5 seconds (82 total), showing 1
2018-07-05 10:26:05 W Dmx[1029.1] [0] PCR to system clock association reset: [-107.698] msec, near pkt 17563193, byte 3301880284
2018-07-05 10:26:12 W [sequence-sequence-1770-drc:selector-1030-track-1-drc] crossed frame sync discontinuity. Offset at [00:41:43.446] is [00:34:24.633] previous offset was [00:34:24.766].  RAW 411139200, TA [01:16:08.213], FA [00:41:43.446]
2018-07-05 10:26:12 W [sequence-sequence-1770-drc:selector-1030-track-2-drc] crossed frame sync discontinuity. Offset at [00:41:43.446] is [00:34:24.633] previous offset was [00:34:24.766].  RAW 411139200, TA [01:16:08.213], FA [00:41:43.446]
2018-07-05 10:26:12 W [sequence-sequence-1770-drc:selector-1030-track-3-drc] crossed frame sync discontinuity. Offset at [00:41:43.446] is [00:34:24.633] previous offset was [00:34:24.766].  RAW 411139200, TA [01:16:08.213], FA [00:41:43.446]
2018-07-05 10:26:12 W [sequence-sequence-1770-drc:selector-1030-track-4-drc] crossed frame sync discontinuity. Offset at [00:41:43.446] is [00:34:24.633] previous offset was [00:34:24.766].  RAW 411139200, TA [01:16:08.213], FA [00:41:43.446]
2018-07-05 10:26:12 W [sequence-sequence-1770-drc:selector-1030-track-5-drc] crossed frame sync discontinuity. Offset at [00:41:43.446] is [00:34:24.633] previous offset was [00:34:24.766].  RAW 411139200, TA [01:16:08.213], FA [00:41:43.446]
2018-07-05 10:26:15 W Dmx[1029.1] 1 new PCR errors in the last 5 seconds (83 total), showing 1
2018-07-05 10:26:15 W Dmx[1029.1] [0] PCR to system clock association reset: [-83.495] msec, near pkt 17635688, byte 3315509344
2018-07-05 10:26:39 W Dmx[1029.1] PCR clock skew [-411] msec
2018-07-05 10:27:17 W [sequence-sequence-1770-drc:selector-1030-track-1-drc] crossed frame sync discontinuity. Offset at [00:42:48.347] is [00:34:24.266] previous offset was [00:34:24.633].  RAW 416968320, TA [01:17:12.981], FA [00:42:48.347]
2018-07-05 10:27:17 W [sequence-sequence-1770-drc:selector-1030-track-2-drc] crossed frame sync discontinuity. Offset at [00:42:48.347] is [00:34:24.266] previous offset was [00:34:24.633].  RAW 416968320, TA [01:17:12.981], FA [00:42:48.347]
2018-07-05 10:27:17 W [sequence-sequence-1770-drc:selector-1030-track-3-drc] crossed frame sync discontinuity. Offset at [00:42:48.347] is [00:34:24.266] previous offset was [00:34:24.633].  RAW 416968320, TA [01:17:12.981], FA [00:42:48.347]
2018-07-05 10:27:17 W [sequence-sequence-1770-drc:selector-1030-track-4-drc] crossed frame sync discontinuity. Offset at [00:42:48.347] is [00:34:24.266] previous offset was [00:34:24.633].  RAW 416968320, TA [01:17:12.981], FA [00:42:48.347]
2018-07-05 10:27:17 W [sequence-sequence-1770-drc:selector-1030-track-5-drc] crossed frame sync discontinuity. Offset at [00:42:48.347] is [00:34:24.266] previous offset was [00:34:24.633].  RAW 416968320, TA [01:17:12.981], FA [00:42:48.347]
2018-07-05 10:27:20 W Dmx[1029.1] 1 new PCR errors in the last 5 seconds (84 total), showing 1
2018-07-05 10:27:20 W Dmx[1029.1] [0] PCR to system clock association reset: [-372.172] msec, near pkt 18145915, byte 3411432020
2018-07-05 10:27:26 W Dmx[1029.1] 1 new PCR errors in the last 5 seconds (85 total), showing 1
2018-07-05 10:27:26 W Dmx[1029.1] [0] PCR to system clock association reset: [273.345] msec, near pkt 18202791, byte 3422124708
[... cut ...]
2018-07-05 10:44:46 W Dmx[1029.1] [0] PCR to system clock association reset: [286.567] msec, near pkt 26385135, byte 4960405380
2018-07-05 10:45:31 W Dmx[1029.1] 1 new PCR errors in the last 5 seconds (118 total), showing 1
2018-07-05 10:45:31 W Dmx[1029.1] [0] PCR to system clock association reset: [-107.723] msec, near pkt 26740714, byte 5027254232
2018-07-05 10:45:38 W Dmx[1029.1] PCR clock skew [-251] msec
2018-07-05 10:46:38 W Dmx[1029.1] PCR clock skew [439] msec
```

I really would love to help you debug this because i really need tsp for my SCTE35 tests....

zuzzurro, 2018-07-05:
Ah, before choosing --max-input-packets 4 --max-flushed-packets 4  I tried with:
--max-input-packets 128 --max-flushed-packets 128 
--max-input-packets 64 --max-flushed-packets 64 
--max-input-packets 32 --max-flushed-packets 32 
--max-input-packets 16 --max-flushed-packets 16 
--max-input-packets 8 --max-flushed-packets 8

I have to say that the error are quite consistent and do not seem to be that affected by these changes

lelegard, 2018-07-05:
~~~
2018-07-05 10:45:31 W Dmx[1029.1] 1 new PCR errors in the last 5 seconds (118 total), showing 1
2018-07-05 10:45:31 W Dmx[1029.1] [0] PCR to system clock association reset: [-107.723] msec, near pkt 26740714, byte 5027254232
2018-07-05 10:45:38 W Dmx[1029.1] PCR clock skew [-251] msec
2018-07-05 10:46:38 W Dmx[1029.1] PCR clock skew [439] msec
~~~
I do not know this Elemental device, but I guess that errors like these indicate that the IP stream is not smooth enough for the Elemental. The PCR values goes back and forth from the device reference clock, -0.2 s, +0.4 s. This device seems to have either limiter input buffer capabilities or requires extremely precise real-time behaviour. 

~~~
2018-07-05 10:44:37 W [sequence-sequence-1770-drc:selector-1030-track-1-drc] crossed frame sync discontinuity. Offset at [01:00:10.355] is [00:34:21.997] previous offset was [00:34:22.497].  RAW 510556800, TA [01:34:32.853], FA [01:00:10.355]
~~~

For these, I do not know. The analysis files you provided demonstrate that there is no discontinuity in the output if tsp. It is probable that the Elemental discards packets when the real-time precision is not sufficient, hence creating its own discontinuities (if the "discontinuities" which are mentioned here are consequences of packet loss).

TSDuck was primarily developed for test, debug, integration, monitoring, demo, etc. I was never designed to work in production. TSDuck can do many things. But it is probably not as good as multicat to do one specific thing extremely well. If you work in a production environment with extremely precise real-time requirements, TSDuck is probably not the best solution.

I assume that commercial inserters for SCTE 35 are much better with real-time requirements. That is why they are commercial and not free...

zuzzurro, 2018-07-05:
Well that may well be right and is in line with my thoughts (there's an issue with precision), but in the end the core of the loop will be to wait with clock_nanosleep for a certain amount of time and then do a send, no? So there should not be that much of variability..
I am using realtime scheduling in both cases to remove as much as possible sources of indetermination.
I also tried using strace to compare the amount of time the two programs sleep in clock_nanosleep but unfortunately tsp uses ABS_TIME and multicat uses relative and the comparison was not immediate.

lars18th, 2018-07-06:
Hi @zuzzurro  and @lelegard ,

I feel the problem can be related to PCR marks. :open_mouth: 

Please, use some MPEG/DVB analyzer for checking the PCR marks. I recommend (as free/demo) the  StreamGuru MPEG Analyzer (demo for 15 days). You can run it in a Windows PC and read the stream from the multicast Network. When openend check the section "PCR Measurement". It's the PCR of the bitstream correct? If not, then the problem with the Ericsson is this: it refuses to process TS streams with inconsistent (not perfect) PCR marks.

And if this is the problem, then my request for a `--rewrite-pcr` in the `regulate` plugin has sense in this scenario: https://github.com/tsduck/tsduck/issues/77#issuecomment-402430128

I hope it helps. I only comment it for this reason.

zuzzurro, 2018-07-06:
I read the #77 but it's not clear to me whether there's a resolution that include the addition of the --rewrite-pcr feature. Did @lelegard add it under a different name? Sorry for being dense...

lars18th, 2018-07-06:
Hi @zuzzurro ,

In #77 I'm only discussing about the problem, and one _possible_ solution.

Please, do first the test I recomend. Have your **tsp** stream correct PCR values?
Check it, please!

zuzzurro, 2018-07-06:
Thanks @lars18th . Unfortunately I already had a copy of StreamGuru on my windows machine and the license had expired. Checking it therefore will take a bit longer ...