tsduck/tsduck #87: SDT: Private descriptor not removed

lars18th, 2018-07-06:
Hi,

I found a very annoying trouble:
- When processing one SPTS from a broadcasted MPTS (from a sat source, with pid filtering activated in the tuner passing only pids corresponding to PSI/SI tables and the pids of this program) and using `-P sdt --cleanup-private-descriptors -p "unkown"`. Then I get in the output stream a SDT table that continues to have inside it an "User Defined" descriptor (0x88). The source has the "Service" descriptor (0x48) plus three "User" descriptors (0x86, 0x88 & 0x8C). And only two of them are removed (the first and last)!

You know where can be the problem?

lelegard, 2018-07-06:
Well... How embarrassing... This is a BUG 😞 
Just fixed it.

lars18th, 2018-07-06:
> Just fixed it.

Thank you! Compiling and testing...

lars18th, 2018-07-06:
Hi @lelegard ,

Fixed! Thank you!

closed 2018-07-06 by lars18th