tsduck/tsduck #88: HTTP input on Windows caching issue

rezna11, 2018-07-07:
Would it be possible to change WinINet implementation in TSDuck to make it possible to turn off or limit caching? When running TSDuck with HTTP input, I noticed free disk space going down, then I found a very large mpeg file (see attached screenshot). I think it's TSDuck doint this.
[image]

rezna11, 2018-07-07:
by the way, I tried adding

```
INTERNET_FLAG_DONT_CACHE |
INTERNET_FLAG_NO_CACHE_WRITE |
```

in \src\libtsduck\windows\tsWebRequestGuts.cpp

and it seems to solve the problem

lars18th, 2018-07-09:
:open_mouth: :open_mouth: :open_mouth: 
This is very embarrassing behavior!
Neither processing tool should generate temporary files. And much less ring buffer stream files. That would collapse the system!

So, please @lelegard review the code. If @rezna11 has targeted this problem and it's true, then fix it.

Thank you all for this impresive project!

lelegard, 2018-07-09:
Integrated. 

`INTERNET_FLAG_NO_CACHE_WRITE` is sufficient. The other value is the same (and no longer documented).
Excerpt from WinInet.h :

~~~
#define INTERNET_FLAG_NO_CACHE_WRITE    0x04000000  // don't write this item to the cache
#define INTERNET_FLAG_DONT_CACHE        INTERNET_FLAG_NO_CACHE_WRITE
~~~

closed 2018-07-09 by lelegard

rezna11, 2018-07-09:
thanks for a very fast fix, looks good!

commit f2a67883 (2018-07-09): Fixed #88