tsduck/tsduck #89: LNB Tuning for a Single LO LNB

joolzg, 2018-07-08:
I am trying to analyse some transponders on QuetzSat 1, 77W but I am having difficulties trying to get it scanning using the correct parameters

-I
dvb
--adapter
2
--frequency
12448000000
--symbol-rate
20000000
--fec-inner
auto
--spectral-inversion
auto
--polarity
vertical
--delivery-system
DVB-S
--modulation
QPSK
--pilots
off
--roll-off
0.35
--lnb
12200,12700,11250
--satellite-number
0

Now the satellite setup is
DBS
	Expressvu, North America
	12200 to 12700 MHz
	Single LO, 11250 MHz

Any idea how i can fix this so I can use your scripts to do a full network scan.

joolz
ps. Fantastic program by the way.

lelegard, 2018-07-09:
Hello @joolzg 

I think that your `--lnb 12200,12700,11250` is incorrect. If I believe the data from [here](https://en.wikipedia.org/wiki/Low-noise_block_downconverter#North_America_DBS_LNB), a DBS LNB do not have low/high band, just one single band. In that case, the option should be `--lnb 11250`.

You need to specify three values when there is a dual band only. For a single band LNB with local oscillator frequency Flo, the option should be `--lnb Flo`.

lelegard, 2018-07-09:
And I updated the documentation which was not very clear.

joolzg, 2018-07-09:
I have tried your suggestion and no luck
tsp -v @dvb.txt -P until --seconds 15 -P analyze -o out.txt -O drop
* tsp: buffer failed to lock into physical memory (12: Cannot allocate memory), risk of real-time issue
* dvb: using /dev/dvb/adapter2 (DVB-S)
* dvb: tuned to transponder --frequency 12428000000 --symbol-rate 20000000 --fec-inner auto --spectral-inversion auto --polarity vertical --delivery-system DVB-S --modulation QPSK --pilots off --roll-off 0.35 --lnb 11250 --satellite-number 0
* Error: dvb: no input DVB signal lock after 5000 milliseconds

Working one
tsp -v @dvb.txt -P until --seconds 15 -P analyze -o out.txt -O drop
* tsp: buffer failed to lock into physical memory (12: Cannot allocate memory), risk of real-time issue
* dvb: using /dev/dvb/adapter2 (DVB-S)
* dvb: tuned to transponder --frequency 11778000000 --symbol-rate 20000000 --fec-inner auto --spectral-inversion auto --polarity vertical --delivery-system DVB-S --modulation QPSK --pilots off --roll-off 0.35 --lnb 9750,10600,11700 --satellite-number 0
* dvb: input bitrate is 32,254,990 b/s

lelegard, 2018-07-09:
You mention that there is a working one. What is the difference between the two ?
Does the working case contain the expected transport stream (from the analysis file) ?
What is your tuner model ?

joolzg, 2018-07-09:
i had to offset the frequency by 650mhz, as the working one uses 10600 as its offset, instead of 11250

working uses frequency 11778 and non working uses 12428.

working uses frequency 10600 and non working uses 11250.

joolzg, 2018-07-09:
Sorry missed the card model

debug: Frontend "TurboSight TBS 6904 DVBS/S2 frontend" supports:

lelegard, 2018-07-09:
> i had to offset the frequency by 650mhz, as the working one uses 10600 as its offset, instead of 11250

I do not really understand. Do you mean that the local oscillator frequency of your LNB is 10600 (instead of 11250) ? What is your `--lnb` option in the working case? 

> working uses frequency 11778 and non working uses 12428.
> working uses frequency 10600 and non working uses 11250.

Since the DBS band, as I can read on Wikipedia, is 12.20 to 12.70 GHz, I assume that these values are local oscillator frequencies, right?

I am not familiar with this. Here, in Western Europe, I use only universal LNB's pointing to low/high band satellites.

lelegard, 2018-07-09:
If it works without `--lnb` option, this means that we use a universal LNB, ie. `--lnb 9750,10600,11700`, which is the default.

So, in that case, assuming that you are in a 12+ GHz range, above the 11.700 switch frequency, you effectively use 10.600 GHz as local oscillator. 

Given the initial description of your problem, I assumed you had a North American DBS dish. But it seems that your LNB is a universal one, just like here in Europe.

joolzg, 2018-07-14:
I fixed this partially by changing the --lnb to "--lnb 9750,11250,11700"

closed 2018-07-14 by joolzg