tsduck/tsduck #90: Remove all unused pids inside a SPTS

lars18th, 2018-07-09:
Hi @lelegard ,

As I see, the commit https://github.com/tsduck/tsduck/commit/e909cdfd4ce1ad412c529fb61ab7a708f65f5a04 fixes all clean-up of private descriptors in all tables. So, I request about how to achieve for removing all unused pids of a SPTS?

The objective is to remove (filter out) automatically all pids that are not PSI/SI tables, video, audio, teletext... in fact all that is not known. For example, user pids or pids related to CA's not in use.

You feel this can be a good idea?

Remember: My objective is to "clean" incoming SPTS from different sources and after "add" to them (convert) all DVB mandatory stuff. This will be done previous to muxing (merge). So, I prefer to have a simple plugin to clean-up the stream.

I hope you found this interesting.
Regards.

lelegard, 2018-07-09:
Do you need something different from what the plugin `rmorphan` already does ?

lars18th, 2018-07-09:
:smile: 

I feel not. Thank you for pointing to it!
If it works, then I'll close this issue.

Regads.