tsduck/tsduck #91: Remove User Private Data from PMT

lars18th, 2018-07-10:
Hi,

Inserting this plugin `-P pmt --cleanup-private-descriptors` in the processing chain of a SPTS from a broadcasted MPTS, I like to remove from the PMT all entries pointing to User Private Data. I see that some `--remove-descriptor` exists, but it implies to use a value. So, at time I need to add  manually `-P pmt -r XXX` for each entry that I see in the table. But the idea it to remove all automatically.

I can achieve this in a direct way?
Thank you!

lelegard, 2018-07-10:
Technically, what do you mean by _entries pointing to User Private Data_ ?

lelegard, 2018-07-10:
Just guessing, if you mean _remove all private descriptors_, then use:
~~~
--remove-descriptor 0x5F --cleanup-private-descriptors 
~~~
The first one removes all private_data_specifier_descriptors. As a consequence, all private descriptors becomes invalid. Meaning that the second option cleans them all.

lars18th, 2018-07-10:
Hi @lelegard ,

> Technically, what do you mean by entries pointing to User Private Data ?

In the PMT I see entries for the different pids (substreams) associated to the program: Video, Audio, TXT, etc. Each of these pids has _StreamType_ with a known value (0x1B, 0x06, etc.). But I see other pids with _StreamType_ with values 0x80 and up. So, they are "User Private Data". And that's the entries that I like to remove from the PMT.

FYI, after this `-P pmt` I put another `-P rmorphan` to remove all unused stuff. But first I need to clean the PMT. At time, I do it reading the table, and adding `-P pmt -r XXX -r YYY` where XXX and YYY correspond to pids with _StreamType_ >= 0x80.

lars18th, 2018-07-10:
Hi @lelegard ,

> The first one removes all private_data_specifier_descriptors. As a consequence, all private descriptors becomes invalid. Meaning that the second option cleans them all.

I'll try it!

lars18th, 2018-07-10:
Hi @lelegard 

It doesn't work! :disappointed:

lelegard, 2018-07-10:
My guess was wrong, it was about removing private descriptors in descriptor lists, not PID entries based on stream type.

lars18th, 2018-07-10:
> not PID entries based on stream type.

Just that!

You know a method to do it?

lelegard, 2018-07-10:
To achieve what you want, it would be better to have an option to remove entries with a specified range of stream types. Not everything above 0x80 is user private. Or at least it depends on what we call "user".

AFAIK, stream types 0x86 = SCTE 35 splice info tables, 0x81 and 0x87 = AC-3 and Enhanced-AC-3 in ATSC systems (and probably other values for AC-4).

lars18th, 2018-07-10:
OK, so perhaps you can support both:

- `--remove-unkown`: This will remove all pids with unkown stream type.
- `--remove-range value1-value2`: This will remove the pids with stream type in the indicated range.

You agree?

lelegard, 2018-07-10:
There is no such thing as an _unknown stream type_. It depends on the various standards and it changes over time. I keep the second option only.

lelegard, 2018-07-10:
Just added `--remove-stream-type value[-value]`

Specify one stream type value or a range of values (`0x80-0xFF` in your case).

lars18th, 2018-07-10:
Hi @lelegard 

Perfect! Compiled it and it's works! :+1:

closed 2018-07-10 by lars18th