tsduck/tsduck #92: Merge to Dektec stability issue

rezna11, 2018-07-10:
Hello. I'm experiencing an issue while merging 4 HTTP inputs and outputting to a Dektec modulator. Everything works for some 6+ hours but then I notice the picture is stopping and then bursting every couple of seconds and the audio is usually missing completely. As usual, TSDuck restart brings things back to normal. But I would really love to know what might be causing this problem.

My command looks like this:

```
C:\mux>tsp --bitrate 39200000 -I null -P regulate -P merge -p 18 -p 20 "tsp -I http -i --reconnect-delay 2000 http://10.0.0.206:8001/1:0:19:66:65:12:5A0000:0:0:0:  -O file" -P merge "tsp -I http -i --reconnect-delay 2000 http://10.0.0.206:8001/1:0:19:67:65:12:5A0000:0:0:0:  -O file" -P merge "tsp -I http -i --reconnect-delay 2000 http://10.0.0.206:8001/1:0:19:69:65:12:5A0000:0:0:0:  -O file" -P merge "tsp -I http -i --reconnect-delay 2000 http://10.0.0.206:8001/1:0:19:6a:65:12:5A0000:0:0:0:  -O file" -P inject -b 32000 -p 0 PAT.xml -P inject -b 32000 -p 16 NIT.xml -P inject -b 32000 -p 17 SDT.xml -O dektec --device 0 --stuffing --modulation 128-QAM --frequency 220000000 --bandwidth 7
* Warning: http: MIME type is video/mpeg, maybe not a valid transport stream
* Warning: http: MIME type is video/mpeg, maybe not a valid transport stream
* Warning: http: MIME type is video/mpeg, maybe not a valid transport stream
* Warning: http: MIME type is video/mpeg, maybe not a valid transport stream
* Warning: http: MIME type is video/mpeg, maybe not a valid transport stream
* Warning: http: MIME type is video/mpeg, maybe not a valid transport stream
* Warning: http: MIME type is video/mpeg, maybe not a valid transport stream
* Warning: http: MIME type is video/mpeg, maybe not a valid transport stream
```

So at some point, TSDuck reconnected to the HTTP sources since I see 8 times the warning message (normally it's printed 4 times at the beginning). I suspect the point of reconnect might be the problematic spot but I can't confirm that since I wasn't able to reproduce the issue, apart from keep it running for 6-18 hours and then come back to broken channels on my TV.

At this point I'm looking for ideas what could I be missing in my command or if this could be hardware related, either a modulator issue or a network problem.

lelegard, 2018-07-10:
I do not think that the modulator is responsible for this. Dektec modulators are very stable and you use the option `--stuffing` which compensates potential underflows. I do not think that the hardware or network are responsible either.

Maybe the root cause is the PCR adjustment in the plugin `merge`. The plugin restamps the PCR but not the DTS/PTS, which is the right thing to do if PCR are very accurately restamped. Since the problem appears after several hours, given the way the PCR are restamped, it is possible that very small inaccuracies accumulate to the point where DTS/PTS are too far away from the restamped PCR.

Also, is the bitrate in the tsp command fully accurate from the expected output modulation parameters? Since PCR are restamped based on the bitrate, this can another possible cause.

This could probably be improved. For instance, when the restamped PCR value become too far away from the original one, we should reset the synchro and restart with the original value unchanged. This may introduce a small glitch at this time but could be more robust on the long term. I will think about it.

As a test, you may try with `--no-pcr-restamp` in all `merge` plugins. This is probably not ideal and my introduce repeated glitches but it would be interesting to see if the problem occurs on the long term. Let us know.

rezna11, 2018-07-10:
I'm not sure what you mean by

> Also, is the bitrate in the tsp command fully accurate from the expected output modulation parameters? Since PCR are restamped based on the bitrate, this can another possible cause.

I specify bitrate at the input plugin and it looks like `dektec` output correctly uses this since it sets the corresponding symbol rate 6075 from 39.2 Mbps rate (I'm doing DVB-C).

I did try `--no-pcr-restamp` but that lead to constant glitching from start. Also when running with debug logging, I can see that the PCR adjusting in `merge` is doing a lot of work but also the PCR adjustments seem all over the place - I don't know if that's normal. Then when I stop, there's some report of nullified packets - I don't know if that's good or bad either:

```
* Debug: merge: adjusted PCR by -2,984,147 (-110 ms) in PID 0x04F1 (1265)
* Debug: merge: adjusted PCR by -3,862,773 (-143 ms) in PID 0x04F1 (1265)
* Debug: merge: adjusted PCR by -4,743,306 (-175 ms) in PID 0x04F1 (1265)
* Debug: merge: adjusted PCR by -5,623,717 (-208 ms) in PID 0x04F1 (1265)
* Debug: merge: adjusted PCR by -6,503,214 (-240 ms) in PID 0x04F1 (1265)
* Debug: merge: adjusted PCR by 4,305,495 (159 ms) in PID 0x0555 (1365)
* Debug: merge: adjusted PCR by 3,404,202 (126 ms) in PID 0x0555 (1365)
* Debug: merge: adjusted PCR by 2,515,133 (93 ms) in PID 0x0555 (1365)
* Debug: merge: adjusted PCR by 138,368 (5 ms) in PID 0x05B9 (1465)
* Debug: merge: adjusted PCR by -774,116 (-28 ms) in PID 0x05B9 (1465)
* Debug: merge: adjusted PCR by 1,798,401 (66 ms) in PID 0x0555 (1365)
* Debug: merge: adjusted PCR by -1,690,450 (-62 ms) in PID 0x05B9 (1465)
* Debug: merge: adjusted PCR by -2,611,260 (-96 ms) in PID 0x05B9 (1465)
* Debug: merge: adjusted PCR by -3,537,160 (-131 ms) in PID 0x05B9 (1465)
* Debug: merge: adjusted PCR by -3,592,662 (-133 ms) in PID 0x05B9 (1465)
* Debug: merge: adjusted PCR by -4,464,681 (-165 ms) in PID 0x05B9 (1465)
* Debug: merge: adjusted PCR by -5,390,582 (-199 ms) in PID 0x05B9 (1465)
* Debug: merge: adjusted PCR by -6,307,666 (-233 ms) in PID 0x05B9 (1465)
* Debug: merge: adjusted PCR by -7,228,017 (-267 ms) in PID 0x05B9 (1465)
* Debug: merge: adjusted PCR by -8,142,122 (-301 ms) in PID 0x05B9 (1465)
* Debug: merge: adjusted PCR by 1,491,390 (55 ms) in PID 0x0555 (1365)
* Debug: merge: adjusted PCR by -9,066,529 (-335 ms) in PID 0x05B9 (1465)
* Debug: merge: adjusted PCR by -9,989,571 (-369 ms) in PID 0x05B9 (1465)
* Debug: merge: adjusted PCR by 715,614 (26 ms) in PID 0x0555 (1365)
* Debug: merge: adjusted PCR by -180,170 (-6 ms) in PID 0x0555 (1365)
* Debug: merge: adjusted PCR by -1,080,093 (-40 ms) in PID 0x0555 (1365)
* Debug: merge: adjusted PCR by -1,975,711 (-73 ms) in PID 0x0555 (1365)
* Debug: merge: adjusted PCR by -2,881,023 (-106 ms) in PID 0x0555 (1365)
* Debug: merge: adjusted PCR by 173,278 (6 ms) in PID 0x04F1 (1265)
* Debug: merge: adjusted PCR by -707,291 (-26 ms) in PID 0x04F1 (1265)
* Debug: merge: adjusted PCR by -1,588,449 (-58 ms) in PID 0x04F1 (1265)
* Debug: merge: adjusted PCR by -2,466,785 (-91 ms) in PID 0x04F1 (1265)
* Debug: merge: adjusted PCR by -3,342,430 (-123 ms) in PID 0x04F1 (1265)
* Debug: merge: adjusted PCR by -4,222,091 (-156 ms) in PID 0x04F1 (1265)
* Debug: merge: adjusted PCR by -5,102,079 (-188 ms) in PID 0x04F1 (1265)
* Debug: merge: adjusted PCR by -5,681,424 (-210 ms) in PID 0x04F1 (1265)
* Debug: merge: adjusted PCR by -5,979,965 (-221 ms) in PID 0x04F1 (1265)
* Debug: merge: adjusted PCR by -6,860,663 (-254 ms) in PID 0x04F1 (1265)
*****     tsp: user interruptsttttp: ussp: user inser tes,inrrupt, ter tep: urminser inteatpterrupt, teingmrminati..: user inter.rnin
rurupt,gpt, terminating....ating...
. t*.
e
rDmienabtinugg.:. .nu
ll: input thread aborted after 1,918,918 packets
* dektec: terminating DTA-111 output
* Debug: inject: packet processing thread aborted after 1,829,678 packets, 1,829,678 passed, 0 dropped, 0 nullified
* Debug: inject: packet processing thread aborted after 1,829,678 packets, 1,829,678 passed, 0 dropped, 0 nullified
* Debug: inject: packet processing thread aborted after 1,829,678 packets, 1,829,678 passed, 0 dropped, 0 nullified
* Debug: merge: receiver thread completed
* Debug: merge: packet processing thread aborted after 1,829,678 packets, 1,810,611 passed, 0 dropped, 19,067 nullified
* Debug: merge: receiver thread completed
* Debug: merge: packet processing thread aborted after 1,829,678 packets, 1,810,510 passed, 0 dropped, 19,168 nullified
* Debug: merge: receiver thread completed
* Debug: merge: receiver thread completed
* Debug: merge: packet processing thread aborted after 1,829,678 packets, 1,829,531 passed, 0 dropped, 147 nullified
* Debug: merge: packet processing thread aborted after 1,829,678 packets, 1,810,628 passed, 0 dropped, 19,050 nullified
* Debug: regulate: packet processing thread aborted after 1,873,986 packets, 1,873,986 passed, 0 dropped, 0 nullified
* dektec: DTA-111 output terminated
* Debug: dektec: output thread aborted after 1,829,678 packets (1,829,678 output)
* Debug: Report logging thread terminated
```

lelegard, 2018-07-10:
> I specify bitrate at the input plugin and it looks like dektec output correctly uses this since it sets the corresponding symbol rate 6075 from 39.2 Mbps rate (I'm doing DVB-C).

OK, your command was so long that I did not notice that you did not specify **all** modulation options and let the modulator find the missing ones from the bitrate. So, this is fine.

> I did try --no-pcr-restamp but that lead to constant glitching from start.

That is normal. The point was to see if the hang occurred or not with this. But we cannot have the answer before several hours.

> Also when running with debug logging, I can see that the PCR adjusting in merge is doing a lot of work but also the PCR adjustments seem all over the place - I don't know if that's normal.

It is normal and expected. The packets are inserted in the right order and on time _on average_ but certainly not with the very same packet interval as in the original stream. So, yes, restamping almost all PCR's is normal.

Otherwise, I did not notice the presence of the `regulate` plugin. I have no experience of using it for many hours. It may accumulate small accuracies as well. But, more important, I wonder if `regulate` is necessary here. It would be definitely necessary with an unregulated output plugin such as `ip` but here you have a modulator which naturally acts as a regulator. Consequently, you have two regulators in the chain, a software one and a hardware one. If they are not _perfectly_ synchronised, some underflow may occur after hours. I think that at most one regulator should be present in a chain to avoid problems.

I suggest to retry without `regulate` and let the modulator do the unique regulation. To increase the responsiveness, also try to reduce `--buffer-size-mb`, `--max-flushed-packets`, `--max-input-packets`.

lars18th, 2018-07-11:
Hi @rezna11 ,

I'm doing similar actions... and I feel I have the same objetive: muxing with TSDuck.
So, let me to comment something perhaps you missed:
- The `http` plugin can generate troubles in some cases: if the connection drops, if the server slowdowns the connection, etc. After some testing I see that I need to tweak the plugin to work flawlessly. In my environment I need to use: `tsp -r I http "http://192.168.1.100:80/ch1.ts" -i --ignore-errors --reconnect-delay 100 --connection-timeout 5000 --max-queue 9000`. So you miss `-r` for realtime and `--max-queue` which is definitely required to avoid jitter problems (for example when the server sends a large number of packets). Finally, please explain: why you use `-O file` in the merge chains? By default if no output is indicated the PIPE STDOUT is the output, no?

Hi @lelegard ,

Regading the muxing using TSDuck (or with external tools, but injecting after to TSDuck for modulation), I observe several times the problem of the slowdown of the stream. So I have some questions:
- When using a modulator like the HiDes what is the best "bitrate" value to use? I feel it's impossible to have the _exact_ value, so I don't know if the "tsp driver" removes the excess of null's or it adds the missing null's.
- The `regulate` plugin is required before muxing to "smooth" the inputs. Particularly with signals than can fail, for example from one HTTP source. So as we can't remove it... it's possible to it for working in synch when using it with a modulator device?
- Would it be possible for the `merge` plugin to detect a slowdown and adjust the difference with a small and puntual acceleration? Perhaps when the buffer increases, or when the PCR marks drift a lot from the DTS/PTS values.

I'm continuing my tests. Wait for more feedback!

rezna11, 2018-07-11:
thanks for suggesting `--max-queue`, I'll consider it during other tests if the one I'm running since last night is not sucessful (removing regulate and decreasing `--buffer-size-mb`,` --max-flushed-packets`, `--max-input-packets`)

I'm not sure why would I need to specify `-r` since my plugins are realtime ones, I think!

> Finally, please explain: why you use -O file in the merge chains? By default if no output is indicated the PIPE STDOUT is the output, no?

I haven't thought about that, you're right I could leave that part out but I don't think it should do any harm, although it's unnecessary. Thanks for pointing this out.

lars18th, 2018-07-11:
Hi @rezna11 ,

> I'm not sure why would I need to specify -r since my plugins are realtime ones, I think!

From what I readed in the Documentation, the realtime mode is automatically selected when some output needs it. In your case the output dektec enables it. However, I don't know about the "sub tsp chains" used with the `merge` plugin. So, I prefer to insert it in any case in all chains.

> thanks for suggesting --max-queue, I'll consider it during other tests if the one I'm running since last night is not sucessful (removing regulate and decreasing --buffer-size-mb,--max-flushed-packets, --max-input-packets)

I'm checking these parameters too. However, for HTTP inputs the default _buffer_ value is very small. For this reason I recommend to increase it with `--max-queue`. I feel this is the only parameter required to handle when using HTTP instead of other inputs. For example, with multicast inputs I need too to add `--buffer-size 83886080` to not lost packets. So, I feel the key value is to have sufficient buffering in the input; without taking into account the internal _processing_ buffer (with a default of 16MB) that I feel we don't need to touch.

And @lelegard , if we like to increase the input buffer for PIPE (standard input) what parameter should we use?

rezna11, 2018-07-11:
I'd think http input plugin is also considered real-time but of course I might be wrong.

Anyway, `--buffer-size` also sounds interesting and it's something I'll try as well. The reason I went for HTTP source when merging was the fact that it worked much more stable than UDP. But I have perhaps a stupid question... what's the 83886080 number representing here? Could it be 83MB? Wouldn't that be too high, introducing a huge delay?

lars18th, 2018-07-11:
Hi @rezna11 ,

1. I don't know if the `http` input is real-time. I doubt it, beacuase it can be used for offline (for example, serving files from a remote server).
2. If you really like to do an action in real-time then it's best to enable it in all cases.
3. The use of the `--buffer-size` option for the multicast input is required in my environment. I'm sure it's the same with `http`: the default input buffer is very small for real-time processing.
4. The value `83886080 ` is the maximum for the socket option in the Windows platform. For Linux, I need to increase the system value with `sysctl net.core.rmem_max` and then I can use this number. This represents 80MB, that's ~64K packets of 1316 bytes (7*188). So it's a common value and only apply to burst transfers.

In any case, I hope @lelegard can explain about how to tweak PIPE inputs. As now I know how to increase the buffer for HTTP and Multicast, but I still have problems with pipe channels.

Regards.

lelegard, 2018-07-11:
> if we like to increase the input buffer for PIPE (standard input) what parameter should we use?

There is no "input buffer" in general. The input plugin is called by `tsp` when there is some free space in the global buffer (the total shared size of which is set by `--buffer-size-mb`). The input plugin is called for at most `--max-input-packets` at a time to make sure that the input plugin, whichever it is, does not wait longer than this number of packets. But the input plugin may choose to return earlier with less packets.

There are some asynchronous plugins such as `ip` which may loose packets if they are not reactive enough. This means that these specific input plugins have an intermediate dedicated buffer to store packets which arrive in burst while there is no room in the global buffer. Playing with this buffer size can influence the correct behavior of the processing since input packets can be lost if the buffer is too small.

Some other plugins such as `http` are fully synchronous _but_ their internal design is a "push" model while the standard model of a tsp input plugin is a "pull" one. The internal input buffer of the plugin is just an adaptation of the push <=> pull models. So, the size of this buffer can influence the performances but not the correctness of the processing. See the intermediate abstract classs `PushInputPlugin` if you are interested on how it works. In the case of `http`, if the buffer is too small, the TCP regulation applies and no packet is lost (unless the remote server decides to drop packets in case its own write operation blocks).

For the standard input, this is the `file` plugin. And a file is a file. Using standard input or a disk file makes no difference. And if the standard input is a pipe or a disk file does not make any difference either. The input plugin `file` just reads from the file descriptor when `tsp` calls it and it writes the packets directly into the global buffer. There is no "input buffer" here.

lars18th, 2018-07-12:
Hi @lelegard ,

From what you said, I understand that:
- IP Input: It uses a buffer because is asynchronous.
- HTTP Input: It uses a buffer to adapt the push <=> pull model.
- FILE Input: no buffer.

Then, let me to comment some topics:
- When using the HTTP input, if the server runs in real-time (that's true for almost all MPEG-TS servers) it drops packets when the TCP output buffer is full. And almost a lot of them use a small buffer for overcome the TCP retransmission troubles (here the buffer is the TCP/IP socket buffer; and not an internal application buffer). So, a buffer inside the HTTP input plugin of TSDuck is a must have. Without it packets are lost (I checked it out.).
- When using PIPE's, and not disk files, a similar behaviour is true. The PIPE buffer is very small (in general) and when the source process can't write on it then it discard packets. This is a problem when running real-time applications. For sure, in an off-line model this behaviour is false.

So, when running `tsp -r` I feel an input file buffer for reading (and perhaps for outputing too) is required.

Please, think on it.

rezna11, 2018-07-12:
OK so back to the original issue - more than a day after running without `regulate` and with `tsp --bitrate 39200000 --buffer-size-mb 12 --max-flushed-packets 500 --max-input-packets 500 -I null -P merge ...`, I didn't notice the same problem where the output becomes totally unwatchable and completely out of sync but there still are problems. What happens is that the video breaks up or pauses for a brief moment every now and then (can happen every 5 seconds or it can be OK for a minute or so). It seems that things get worse when there's more motion on the screen, meaning higher bitrate. The audio remains without interruptions, only video is affected. I didn't try setting `--max-queue` and `-r` yet.

Anyway, now I'm not sure what's causing the errors in the video. Not really knowing much about this, I at least tried to analyze a recording and compare it with another one that's not showing those errors. What seemed a bit suspicious was high DTS-PCR and PTC-PCR variation (almost up to 3500 ms) but I don't know if it's a problem or still OK. Other than that, there are clearly some PCR errors - see attached pictures, it could help.

[image]
[image]

lars18th, 2018-07-12:
Hi @rezna11 ,

> What seemed a bit suspicious was high DTS-PCR and PTC-PCR variation

Sure! This is the typical problem when packets suffer **buffer** troubles. A huge time variation indicates that packets aren't received at the correct speed. So the buffers will fill up or be completely empty. This is the reason of the PCR marks in a Transport Stream.

So you need to "smooth" your TS. Read the issue #77 and check the `regulate` plugin with the "new" parameter `--pcr-synchronous`. I think it'll help you!

rezna11, 2018-07-12:
so you'd add `regulate` for each of the merged streams? Something like this?

`tsp --bitrate 39200000 --buffer-size-mb 12 --max-flushed-packets 500 --max-input-packets 500 -I null -P merge -p 18 -p 20 "tsp -r -I http --max-queue 9000 -i --reconnect-delay 2000 http://10.0.0.206:8001/1:0:19:66:65:12:5A0000:0:0:0: -P regulate --pcr-synchronous" -P merge "tsp -r -I http --max-queue 9000 -i --reconnect-delay 2000 http://10.0.0.206:8001/1:0:19:67:65:12:5A0000:0:0:0: -P regulate --pcr-synchronous" -P merge "tsp -r -I http --max-queue 9000 -i --reconnect-delay 2000 http://10.0.0.206:8001/1:0:19:69:65:12:5A0000:0:0:0: -P regulate --pcr-synchronous" -P merge "tsp -r -I http --max-queue 9000 -i --reconnect-delay 2000 http://10.0.0.206:8001/1:0:19:6a:65:12:5A0000:0:0:0: -P regulate --pcr-synchronous" -P inject -b 16000 -p 0 PAT.xml -P inject -b 8000 -p 16 NIT.xml -P inject -b 16000 -p 17 SDT.xml -O dektec --stuffing --device 0 --modulation 128-QAM --frequency 220000000 --bandwidth 7`

with this I'm getting constant noticable, mostly audio, glitches. But I tried on another TV now so maybe this one shows errors on audio while the other one would experience video issues, I have to check that, but still I'm not getting smooth stream

closed 2026-04-23 by github-actions[bot]