tsduck/tsduck #93: Problem building rpm for Centos 7.

Philip-Wells, 2018-07-10:
I'm having a problem building an rpm for Centos 7 - none of the tools 
and shared objects are being added to the rpm.  I've done this same 
procedure on my Fedora dev system and the rpm does include the 
binaries.

The target deployment is Centos 7 so I built a Centos 7 vm as a build 
system, installed git and the required build files, copied the repo to 
the local Centos drive and did a 'make rpm'. 

uname -a
Linux localhost.localdomain 3.10.0-862.3.3.el7.x86_64 #1 SMP Fri Jun 15 04:15:27 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

cat /etc/centos-release
CentOS Linux release 7.5.1804 (Core)

df /root
Filesystem                        1K-blocks      Used     Available  Use%   Mounted on 
/dev/mapper/centos-root  51474912 2316840    46520248     5%   /

rpm -qlp tsduck-3.13-814.el7.x86_64.rpm
/etc/security/console.perms.d/80-tsduck.perms
/etc/udev/rules.d/80-tsduck.rules
/usr/bin/tsduck.dvb.names
/usr/bin/tsduck.oui.names
/usr/bin/tsduck.so
/usr/bin/tsduck.xml
/usr/share/doc/tsduck-3.13
/usr/share/doc/tsduck-3.13/CHANGELOG.txt
/usr/share/doc/tsduck-3.13/LICENSE.txt
/usr/share/doc/tsduck-3.13/tsduck.pdf

yum install gcc-c++ doxygen dos2unix graphviz curl pcsc-tools pcsc-lite-devel cppunit-devel libcurl libcurl-devel rpmdevtools
	Loaded plugins: fastestmirror
	Loading mirror speeds from cached hostfile
	 * base: mirror.math.princeton.edu
	 * epel: mirror.cs.princeton.edu
	 * extras: mirror.wdc1.us.leaseweb.net
	 * updates: mirrors.centos.webair.com
	Package gcc-c++-4.8.5-28.el7_5.1.x86_64 already installed and latest version
	Package 1:doxygen-1.8.5-3.el7.x86_64 already installed and latest version
	Package dos2unix-6.0.3-7.el7.x86_64 already installed and latest version
	Package graphviz-2.30.1-21.el7.x86_64 already installed and latest version
	Package curl-7.29.0-46.el7.x86_64 already installed and latest version
	Package pcsc-tools-1.4.25-1.el7.x86_64 already installed and latest version
	Package pcsc-lite-devel-1.8.8-7.el7.x86_64 already installed and latest version
	Package cppunit-devel-1.12.1-11.el7.x86_64 already installed and latest version
	Package libcurl-7.29.0-46.el7.x86_64 already installed and latest version
	Package libcurl-devel-7.29.0-46.el7.x86_64 already installed and latest version
	Package rpmdevtools-8.3-5.el7.noarch already installed and latest version
	Nothing to do

make rpm >make-rpm.log 2>>make-rpm.log
<make-rpm.log inserter below>

make -C build rpm
rm -rf /root/tsduck/installers/tmp
mkdir -p /root/tsduck/installers/tmp/tsduck-3.13-814
tar -C /root/tsduck --exclude '.git' --exclude '.svn' --exclude '*.zip' --exclude '*.gz' --exclude '*.tgz' --exclude '[Dd]ebug*' --exclude '[Rr]elease*' --exclude 'doxy' --exclude 'doxygen' --exclude '*.t2d' --exclude '*.o' --exclude '*.so' --exclude '*.ko' --exclude '*.a' --exclude '*.lib' --exclude '*.obj' --exclude '*.exe' --exclude '*.dll' --exclude '*.gcov' --exclude '*.dep' --exclude '.kdbgrc.*' --exclude 'tmp' --exclude '*.tmp' --exclude '*.out' --exclude '*.log' --exclude '*.tlog' --exclude 'core' --exclude 'core.*' --exclude 'vgcore.*' --exclude '*.stackdump' --exclude '.vs' --exclude 'ipch' --exclude '*.ncb' --exclude '*.suo' --exclude '*.idb' --exclude '*.sdf' --exclude '*.opensdf' --exclude '*.vcproj.*.user' --exclude '*.user.*' --exclude '*.vcxproj.user' --exclude '*.dll.embed.manifest.dll' --exclude '*.dll.embed.manifest.ilk' --exclude '.tmp_versions' --exclude '*.VC.db' --exclude '*.VC.dbopen' --exclude '*.VC.opendb' --exclude '*.deb' --exclude '*.rpm' --exclude '*.bz2' --exclude '*.7z' --exclude '*~' --exclude '*.bak' --exclude '*.autosave' --exclude 'cov-int' -cpf - . | tar -C /root/tsduck/installers/tmp/tsduck-3.13-814 -xpf -
make -C /root/tsduck/installers/tmp/tsduck-3.13-814 distclean
  [CLEAN] /root/tsduck/installers/tmp/tsduck-3.13-814
  [CLEAN] /root/tsduck/installers/tmp/tsduck-3.13-814/build
  [CLEAN] /root/tsduck/installers/tmp/tsduck-3.13-814/build/msvc2017
  [CLEAN] /root/tsduck/installers/tmp/tsduck-3.13-814/build/qtcreator
  [CLEAN] /root/tsduck/installers/tmp/tsduck-3.13-814/build/qtcreator/libtsduck
  [CLEAN] /root/tsduck/installers/tmp/tsduck-3.13-814/dektec
  [CLEAN] /root/tsduck/installers/tmp/tsduck-3.13-814/doc
  [CLEAN] /root/tsduck/installers/tmp/tsduck-3.13-814/images
  [CLEAN] /root/tsduck/installers/tmp/tsduck-3.13-814/installers
  [CLEAN] /root/tsduck/installers/tmp/tsduck-3.13-814/sample
  [CLEAN] /root/tsduck/installers/tmp/tsduck-3.13-814/src
[... 980 lines cut ...]
Processing files: tsduck-devel-3.13-814.el7.x86_64
Executing(%doc): /bin/sh -e /var/tmp/rpm-tmp.zmcpR8
Provides: tsduck-devel = 3.13-814.el7 tsduck-devel(x86-64) = 3.13-814.el7
Requires(rpmlib): rpmlib(CompressedFileNames) <= 3.0.4-1 rpmlib(FileDigests) <= 4.6.0-1 rpmlib(PayloadFilesHavePrefix) <= 4.0-1
Checking for unpackaged file(s): /usr/lib/rpm/check-files /root/rpmbuild/BUILDROOT/tsduck-3.13-814.el7.x86_64
Wrote: /root/rpmbuild/SRPMS/tsduck-3.13-814.el7.src.rpm
Wrote: /root/rpmbuild/RPMS/x86_64/tsduck-3.13-814.el7.x86_64.rpm
Wrote: /root/rpmbuild/RPMS/x86_64/tsduck-devel-3.13-814.el7.x86_64.rpm
Executing(%clean): /bin/sh -e /var/tmp/rpm-tmp.fWH9vJ
Executing(--clean): /bin/sh -e /var/tmp/rpm-tmp.SqZkLx
cp -uf /root/rpmbuild/RPMS/*/tsduck-3.13-814.el7.*.rpm /root/tsduck/installers
cp -uf /root/rpmbuild/RPMS/*/tsduck-devel-3.13-814.el7.*.rpm /root/tsduck/installers
cp -uf /root/rpmbuild/SRPMS/tsduck-3.13-814.el7.src.rpm /root/tsduck/installers
+ /usr/bin/mkdir -p /root/rpmbuild/BUILDROOT/tsduck-3.13-814.el7.x86_64/usr/share/doc/tsduck-devel-3.13
+ cp -pr LICENSE.txt /root/rpmbuild/BUILDROOT/tsduck-3.13-814.el7.x86_64/usr/share/doc/tsduck-devel-3.13
+ exit 0
+ umask 022
+ cd /root/rpmbuild/BUILD
+ cd tsduck-3.13-814
+ rm -rf /root/rpmbuild/BUILDROOT/tsduck-3.13-814.el7.x86_64
+ exit 0
+ umask 022
+ cd /root/rpmbuild/BUILD
+ rm -rf tsduck-3.13-814
+ exit 0

lelegard, 2018-07-10:
TSDuck does compile on CentOS 7.5. I do not publish the corresponding packages but I do this routinely at work for my colleagues.

I have noticed the following output line which interrupts before installing the plugins:
~~~
[CXX] tsplugin_olesmarker.cpp
~~~

I assume that this plugin is yours. Not a standard TSDuck plugin anyway.

I would suggest an incremental approach. First, do a clean `git clone` from GitHub on the CentOS system, creating a pristine source tree. Then, try `make rpm` to see if it works (it should). Then, add your plugins and see the differences.

That being said, I would recommend to separate your plugins from the original TSDuck source tree to segment problems. I do this at work as well for a few private plugins. This is a separate project which uses the headers and libraries which are installed by `tsduck-devel`. Then, a separate RPM is created with a dependency to `tsduck`.

lars18th, 2018-07-11:
Hi @lelegard ,

Just out I was curious: It's possible to compile separate plugins and use them with `tsp` without recompile the entire `tsduck` project?

lelegard, 2018-07-11:
> It's possible to compile separate plugins and use them with tsp without recompile the entire tsduck project?

Yes, definitely. This is another objective of TSDuck : being extensible, even by users or third-parties.
You don't even need to get the TSDuck sources. This is the purpose of the "TSDuck development environment". It is an option in the Windows installer and in package `tsduck-devel` or `tsduck-dev` on Linux. See details [here](https://tsduck.io/doxy/usinglibrary.html).

lars18th, 2018-07-11:
Umm... I feel like I don't get it! :unamused: 

The "TSDuck development environment" is to use the TSDuck as a library and generate third-party tools, right? So my question is then: _after_ compile the entire TSDuck project, the `tsp` binary can be used _at runtime_ with new plugins (`.so` files)? If yes, how to do it?

My question is because you pointed to @Philip-Wells to remove his own plugin `olesmarker`. So, if he included it in the TSDuck project then it will be because he likes to use it with `tsp -P olesmarker`, no?

I'm confused.

Philip-Wells, 2018-07-11:
Following up on lars18th -

First off, I found my bug - it is now producing rpm with binaries included.  Thanks you.

With regard to creating my plugin -

I cloned the repo, did a make rpm, installed both the tsduck and tsduck-devel kits, created a new folder with my original source (with new includes as paths have changed), created the Makefile as suggested and built it.  It appears to want to build an image vs a shared object.  I tried to change the target to  tsplugin_olesmarker.so but that only confused it with no rule to make a ".so".

<Makefile>
include /usr/include/tsduck/tsduck.mk
default: tsplugin_olesmarker

<log>
$ make
g++ -I/usr/include/PCSC -I/usr/include/tsduck  --std=c++11  -Wl,-rpath,'$ORIGIN',-rpath,/usr/bin  tsplugin_olesmarker.cpp  /usr/bin/tsduck.so -lcurl -lpcsclite -lpthread -lrt -ldl -lm -lstdc++ -o tsplugin_olesmarker
/usr/lib/gcc/x86_64-redhat-linux/8/../../../../lib64/crt1.o: In function `_start':
(.text+0x20): undefined reference to `main'
collect2: error: ld returned 1 exit status
make: *** [<builtin>: tsplugin_olesmarker] Error 1

Philip-Wells, 2018-07-11:
I think I resolved my issue.  Builds and runs.

include /usr/include/tsduck/tsduck.mk
default: tsplugin_olesmarker.so

tsplugin_olesmarker.so: tsplugin_olesmarker.cpp
	$(CXX) $(CXXFLAGS) $^ -shared -fPIC -o $@

closed 2018-07-11 by Philip-Wells

lelegard, 2018-07-11:
> The "TSDuck development environment" is to use the TSDuck as a library and generate third-party tools, right? So my question is then: after compile the entire TSDuck project, the tsp binary can be used at runtime with new plugins (.so files)? If yes, how to do it?

The answer is "RTFM as usual" :smiley: 

In the user's guide, in the reference for `tsp`, section "Plugin activation options", the way `tsp` searches and loads plugins is explained. Basically, the two main solutions are :
- Copy the .so or .dll in the same directory as the `tsp` executable.
- Or define the environment variable `TSPLUGINS_PATH` to include all directories containing additional plugins.

I updated and pushed the sample plugin in subdirectory `sample/sample-plugin`. I completed and tested the scripts for Linux and Windows. There is a sample makefile for Unix and a sample VS project file for Windows. There are sample test scripts to run `tsp` with the sample plugin. 

Of course, these files and scripts are just examples. You may adapt to your environment. But, at least, they work.

lars18th, 2018-07-11:
Great job @lelegard ! :+1: 

I feel it be interesting to develop new plugins and share them as PR's .