tsduck/tsduck #94: Inject plugin: time interval

lars18th, 2018-07-10:
Hi @lelegard ,

Maybe I'm a bit dumb, but I can't find the option to insert PSI tables at fixed interval (PAT at 100ms, NIT at 2000ms, SDT at 500ms, etc.). In the `inject` plugin I see only `--bitrate` that represents a bitrate and not an interval. In some other plugins, like `spliceinject` the option `--inject-interval` exists. Why not for PSI table insertion?

Regards.

lelegard, 2018-07-10:
The general syntax is:
~~~
tsp -P inject [options] input-file[=rate] ...
~~~

The repetition rate in milliseconds is per input file. This is because there is no reason to have the same repetition rate for all tables. Typically, a NIT or SDT "actual" will have a higher repetition rate than an "other" one, all being in the same PID.

lars18th, 2018-07-11:
OK, Thank you!

But, `=rate` and `--bitrate` aren't exclusive?
For example, can I use `tsp -P inject "pat.xml"=100 --pid 0 --bitrate 1024` ? Which is then the recommanded bitrate for regular PSI Tables?

lars18th, 2018-07-11:
Hi @lelegard ,

Sorry, but it doesn't work, or I'm doing something wrong:

```
tsp -r --bitrate 16085561 -I null -P regulate \
 -P inject pat.xml=100  --pid  0 -s -b 1024 \
 -P inject nit.xml=2000 --pid 16 -s -b 1024 \
 -P inject sdt.xml=500  --pid 17 -s -b 1024 \
 -P inject tdt.xml=1000 tot.xml=15000 --pid 20 -s -b 1024 \
 ...
```

The results are:

1. Each table is inserted at just the same interval, despite the `=value`.
2. If I remove the `-b` (bitrate) parameter, then only the PAT table is injected just one time.
3. I don't know how to use quotes: `-P inject "pat.xml"=100` or `-P inject "pat.xml=100"` is correct? The reason for the quotes are the spaces in the path of the file.

I feel some bug exists when injecting tables in a plain TS with only nulls.
Please, can you help me?

lars18th, 2018-07-12:
Hi @lelegard ,

Please, I'm doing something wrong, or a real bug exists?

lars18th, 2018-07-12:
Hi @lelegard ,

I discovered that using a higher bitrate it works as expected:
```
tsp -r --bitrate 16085561 -I null -P regulate \
 -P inject pat.xml=100  --pid  0 -s -b 32000 \
 -P inject nit.xml=2000 --pid 16 -s -b 32000 \
 -P inject sdt.xml=500  --pid 17 -s -b 32000 \
 -P inject tdt.xml=1000 tot.xml=15000 --pid 20 -s -b 32000 \
 ...
```
However, I still don't know the correct "bitrate" values for these tables. Besides, I don't know why it doesn't work without the bitrate parameter.

Please, can you explain it more in deep in the documentation?
Thank you!

lars18th, 2018-07-13:
Hi @lelegard ,

I write this after a careful reading of the _good_ documentation on the `inject` plugin:

- The use of the "bitrate" parameter with the "rate" parameter isn't clear. The problem that I detected is bacause the bitrate is very small. *Only* if the bitrate is sufficient then the "rate" parameter works as expected. This should be indicated.
- Futhermore, each PSI table should have a minimun"bitrate" value (32Kb/s or more). So I recommend to put some default values and use them when no "bitrate" is indicated. You can complete this with a print log of the bitrates used, and print a warning when the bitrate is insufficient.

I suggest this because without this information it's impossible to create a valid Transport Stream.

You agree with it?

lelegard, 2018-08-29:
I have modified the `inject` plugin to make the usage of bitrates and repetition rates more clear and more consistent (commit 31b2ca1e2e932ebde2a312fcc272621cf5014818). I hope that it solves the issues in this thread.

Originally, when a new PID was created (no `--replace`), the bitrate of the new PID had to be specified, either explicitly (`--bitrate`) or implicitly (`--inter-packet`).

Then, the repetition rates were added per file so that several tables may have different repetition rates in the same PID. For instance, in PID 16, it is legitimate to repeat the NIT Actual faster than the various NIT Other. The same idea applies in PID 17 with SDT Actual, SDT Other and the various BAT's. But the repetition rates only influenced the relative bitrates of the various tables inside the same PID, within a global known bitrate for the PID. For instance, it was usual to give a specific repetition rate for the NIT Actual and let all NIT Other use the remaining bandwidth of the PID 16.

The modification I just made is that _when all input files have an explicit repetition rate_, it is no longer necessary to specify a bitrate for the PID. It will be automatically computed from the sizes and repetition rates of all sections.

lelegard, 2018-09-11:
Version 3.14 is out, including the `inject` updates. Closing this issue.

closed 2018-09-11 by lelegard

lars18th, 2018-09-20:
Hi @lelegard ,

I'll try it, but I hope the problem is fixed.
Thank you!