tsduck/tsduck #95: tsrename: rename Network Name

lars18th, 2018-07-10:
Hi,

I'm creating one MPTS from a dummy stuff (-I null), and some actions are still missing. For example, with the plugin `tsrename` it's impossible to assign (or replace) the Network Name (descriptor 0x40) present in the NIT table.

So please can you add support for it?
Thank you!

lelegard, 2018-07-10:
OK. But this will be done in the `nit` plugin, not `tsrename`. Changing the network name is a pure local operation in the NIT. Plugins such as `tsrename` or `svrename` are designed to perform a global operation which needs a consistent set of modifications in several tables.

lelegard, 2018-07-11:
Implemented in commit 9f4c25d6d4811b08594b2a78ac67951c9a6f3bcf

closed 2018-07-11 by lelegard

lars18th, 2018-07-12:
>Implemented in commit 9f4c25d

I suggest to update the file in the tsduck.io section as the PDF is quite old:
https://tsduck.io/download/docs/tsduck.pdf
:wink: