tsduck/tsduck #96: timeref: insert current time

lars18th, 2018-07-10:
Hi @lelegard ,

Another missing functionality:

- The plugin `timeref` doesn't have any option to use the _current time_. That's to insert the time value of the current machine. Futhermore, it's missing the option for "creating" the TDT/TOT tables. Another option like `--create-at-interval value` will be useful to generate the packets instead of replace them.

You think it's a valuable function?

lelegard, 2018-07-10:
> Another missing functionality

There are an infinite number of missing functionalities, you will never find all of them....

The current time is a scripting issue:
~~~
tsp -P timeref --start $(date +%Y/%m/%d:%H:%M:%S)
~~~

To create a set of TOT/TDT, create template XML files for them, with some random time value. Inject them, followed by a `timeref`to set the right time.

Generally speaking, after more than 10 years of TSDuck usage, I never wanted to add everything in TSDuck. If TSDuck is still robust, this is because it has almost nothing in excess. Too many software fall in pieces after years because they are bloated with excessive features.

When I can do something in a script using the current version of TSDuck, I write a script. I modify TSDuck only when it is impossible or at least very hard to do in a script. I recommend that, for any problem you have, first think hard to check how to do it in a script. And only request features when there is no other solution.

Just an example: A long time ago, in a now defunct company I worked for, I wrote a very large script which performed the complete analysis of virtually everything in a complete television network. Think about all those `--normalized` options. What do you think they are for? I added a simple `crontab` entry and for a budget of one desktop computer and a 50€ tuner I did almost the same thing as multi-thousands € professional software.

lars18th, 2018-07-11:
Hi @lelegard ,

Please, let me say before any futher comments: **Thank you for this impressive project! I really apreciate your work.** :+1: 
I say this with all respect.

Regading the example... YES! It works! Thank you!
I suggest to include it in the documentation, as it's quite difficult to think about this solution directly (as far as I'm concerned).

However, after reading your _qualified_ view on the use of scripts (which I agree with after all), I see a limitation in this particular case: The use of scripts isn't _portable_. So, this solution only works in Linux, but not in Windows.

Why not then support this simple new parameter `-P timeref --start-sys-time`? I feel it's very easy. For sure, it will be more powerfull to support `--start-sys-time[+offset]`, but as you mentioned the simplicity is a good quality. :smiley: 

So can you consider it, please?
Regards.

lars18th, 2018-07-11:
Hi @lelegard ,

Thank you to consider it (even later)!

Why I try to not use scripts from _some_ tsp chains?
Because I like to run them inside containers.

What scripts languages I use? NodeJS, LUA, bash-SHELL and .BAT (yes! the quite old _CMD_). I hate PowerShell; and sometimes I use Cygwin but I started to move to WSL. I agree with the idea of combine TSDuck with scripting. It's very powerfull. However, some _simple_ tasks ("simple" in the sense of "clear") can be done best in a portable way without scripting.

Regards!

lelegard, 2018-08-06:
Closing after @lars18th's contribution in #102.

closed 2018-08-06 by lelegard