tsduck/tsduck #98: tsrename: --add-nit doesn't work

lars18th, 2018-07-12:
Hi @lelegard ,

Doing the actions described in #85 (specifically https://github.com/tsduck/tsduck/issues/85#issuecomment-404099611) I see that the action `-P tsrename -t 1 -o 65535 --add-nit` doesn't insert the current TransportStream (the one to rename) in the NIT list. So the current NIT table only includes the data of the network, but neither of transport streams.

It's this the expected behaviour?

I need it because after the `merge` plugin the Service needs to be included in the Network (NIT table).

lelegard, 2018-07-12:
`tsrename` renames something existing. It the TS exists in the NIT, its entry is renamed. With option `--add-nit`, the entry is duplicated. Assume that you have TS id 1 and you rename it as 2. If the NIT contains an entry for TS 1, this entry is renamed as 2, meaning that entry 1 no longer exists. With `--add-nit`, entry 1 remains in the NIT and a copy of it is created with TS id 2.

But `--add-nit` does not create a non-existing entry. What should it create in the descriptor list ?

The documentation is not very clear and the name `--add-nit` is maybe misleading.

lelegard, 2018-07-12:
The user's guide has been updated. 

For now, I need my time to develop features that cannot be done outside TSDuck. If something can be implemented in a script with inlined XML files are I described in #85, it won't be implemented inside TSDuck.

That being said, if someone feels like developing it, I will review PR's.

lelegard, 2018-08-27:
Hi @lars18th 
Please close this issue if you consider that the new options in #115 match the request.

closed 2018-09-16 by lelegard