tsduck/tsduck #99: merge+http: quit problem

lars18th, 2018-07-12:
Hi @lelegard ,

When using in the inner chain of the merge the HTTP input with **two** merge, then I see these errors when interrupting the program (^C):
```
^C* tsp: user interrupt, terminating...
* tsp: user interrupt, terminating...
* Debug: file: output thread aborted after 34,240 packets (34,240 output)
* Debug: http: Web transfer is interrupted by application
* Error: http: download error, Failed writing received data to disk/application, Failed writing body (0 != 16384)
* http: downloading from http://192.168.1.100:8080/ch1.ts
* http: MIME type: application/octet-stream, expected size: unknown
* Warning: http: MIME type is application/octet-stream, maybe not a valid transport stream
* Debug: http: HTTP header: HTTP/1.1 200 Ok
* Debug: http: HTTP header: Server: Nginx
* Debug: http: HTTP header: Cache-Control: no-cache
* Debug: http: HTTP header: Pragma: no-cache
* Debug: http: HTTP header: Content-Type: application/octet-stream
* Debug: http: HTTP header: Connection: close
* Debug: http: Web transfer is interrupted by application
...
```
And the messages continues repeating the "reconnection action" until I repeat **two more times** the interrupt (^C + ^C).

This issue only appears when the HTTP input is used inside the merge and two consecutive merge are present, like `-P merge "tsp -r -I http ..." -P merge "tsp -r -I http ..."`. When using other inputs the first interrupt closes all threads without errors.

Testing with only one merge I don't see this problem. Perhaps the problem is with multiple instances of the cURL class at the same time.

Have you seen this problem too?

lars18th, 2018-07-16:
Hi,

I confirm that this problem sometimes appears when using only one http input with the merge plugin.

I don't know where the problem is specifically right now. However, some glitch or bug exists when using ^C to stop the process.

lars18th, 2018-07-20:
Hi @lelegard ,

I isolated the problem with the `http` plugin.

Try to create any tsp command including `-P http -i --ignore-errors --reconnect-delay 100 ...`. This will do an infinite loop. And you can exit from it without problems. But if the _HTTP library_ generates errors because it can't connect (for example if you use an incorrect URL), then in this case using the ^C the tsp process isn't closed (you need to kill it).

So I feel you only need to catch all errors when killing the http object.

lelegard, 2018-08-08:
What is the operating system? Windows, Linux?
In the case of Windows, what is the shell? DOS, PowerShell, bash (cygwin)?

lars18th, 2018-09-20:
Hi @lelegard ,

I noticed this in different platforms. I'll try to isolate the problem when I have more time to run more tests.

So please, don't close this issue.
Regards.