tsduck/tsduck #100: CAT not detected by tspsi or psi plugin

marti566, 2018-07-13:
I have a ts file that i'm trying to verify tables being correctly sent out. The PAT, PMT, NIT, SDT all are detected correctly but the CAT isn't found despite being present in the capture when examined with wireshark. the capture is of a dvb-t modulation. capture included.
[recording.ts.tar.gz](https://github.com/tsduck/tsduck/files/2190794/recording.ts.tar.gz)

lelegard, 2018-08-06:
Hello @marti566 

The CAT sections in PID 1 have the bit _current_next_indicator_ set to zero, meaning "not yet applicable" or "next". Consequently, the table is ignored. 

All other PSI tables in your file have this bit set to one, meaning "currently applicable" or "current". This is the expected value. 

Honestly, I never met a PSI table with this flag set to "next". This is clearly an error of the equipment which generates the CAT.

lelegard, 2018-08-06:
To make this kind of problem more easily detectable, I have added an option `--include-next` to various commands and plugins.

Sample output using your file as input. Note the annotation "next (not yet applicable)" to the CAT, and only the CAT.
~~~
$ tspsi rec.2018_07_13_00_33_12.ts --include-next

* PAT, TID 0 (0x00), PID 0 (0x0000)
  Version: 8, sections: 1, total size: 20 bytes
  - Section 0:
    TS id:     118 (0x0076)
    Program:     1 (0x0001)  PID: 1000 (0x03E8)
    NIT:         0 (0x0000)  PID:   16 (0x0010)

* PMT, TID 2 (0x02), PID 1000 (0x03E8)
  Version: 0, sections: 1, total size: 38 bytes
  - Section 0:
    Program: 1 (0x0001), PCR PID: 1011 (0x03F3)
    Program information:
    - Descriptor 0: CA (0x09, 9), 4 bytes
      CA System Id: 0x0112 (Canal Plus), ECM PID: 1013 (0x03F5)
    - Descriptor 1: CA (0x09, 9), 4 bytes
      CA System Id: 0xFEFE (unknown), ECM PID: 1014 (0x03F6)
    Elementary stream: type 0x02 (MPEG-2 Video), PID: 1011 (0x03F3)
    Elementary stream: type 0x04 (MPEG-2 Audio), PID: 1012 (0x03F4)

* CAT, TID 1 (0x01), PID 1 (0x0001)
  Version: 0, sections: 1, total size: 18 bytes
  - Section 0, next (not yet applicable):
    - Descriptor 0: CA (0x09, 9), 4 bytes
      CA System Id: 0x0112 (Canal Plus), EMM PID: 1015 (0x03F7)

* SDT Actual, TID 66 (0x42), PID 17 (0x0011)
  Version: 8, sections: 1, total size: 38 bytes
  - Section 0:
    Transport Stream Id: 118 (0x0076)
    Original Network Id: 9018 (0x233A)
    Service Id: 1 (0x0001), EITs: yes, EITp/f: yes, CA mode: free
    Running status: running
    - Descriptor 0: Service (0x48, 72), 16 bytes
      Service type: 0x01 (Digital television service)
      Service: "Channel_1", Provider: "Pico"
~~~