tsduck/tsduck #101: Modulation problem

lars18th, 2018-07-16:
Hi @lelegard ,

I hope you can help. This is related to #57...

This is the environment:
- The source is a Network Tuner that receives one DVB-T multiplex. It filters the pids of one program and streams it to a multicast address. All checks with this SPTS don't show any error. All is clean.
- Using this SPTS I modulate it directly (so SPTS):
```
tsp.exe -r --debug -I ip 239.0.0.1:12345 --buffer-size 8388608  \
 -P zap 101 \
 -P tsrename -t 1 -o 65535 \
 -P svrename 101 -i 1 -l 1 -n "Test" \
 -P regulate -p 16 --wait-min 20 --pcr-synchronous \
 -P fork "./tsp -O ip 239.0.0.2:12345" \
 -O hides -b 8 -c QPSK -f 498000000 -g 1/32 -h 3/4 -t 8K
```
- The _second_ output to a multicast address is for checking the created TS. No erros are found on it.

As you can see the concept is quite simple: read from a multicast, do some simple changes (rename service and transport stream), and modulete it. Futhermore the `-P regulate --pcr-synchronous` is required to transfer the SPTS (a VBR program) at correct speed.

But, every seconds (between 4-20) some packets (a small amount, less than 10) are corrupted in the modulated TS. I receive it with a tuner attached to the same computer that does the modulation. And analyzing the output TS from the multicast I don't see any problem. However, from the tuner I see these glitches.

You know where can be the problem?

lars18th, 2018-07-16:
Hi @lelegard ,

I do another test with the objective of remove any timing or creation error regarding the TS. So I used this command:
```
tsp.exe -r --debug -I ip 239.0.0.1:12345 --buffer-size 8388608 \
 -P zap -s 101 \
 -P tsrename -t 1 -o 65535 \
 -P svrename 101 -i 1 -l 1 -n "Test" \
 -P fork "./tsp -O ip 239.0.0.2:12345" \
 -O hides -b 8 -c 64-QAM -f 498000000 -g 1/4 -h 2/3 -t 8K
```

In this case the entire **original** TS structure is preserved. So, I don't need to use `-P regulte`. And the secondary stream (@239.0.0.2:12345) is perfect. However, in this case I still see the same problem: time to time, some errors (CC discontinuity) appears in the reception side.

Any idea?

lars18th, 2018-07-16:
FYI,

After moving the HiDes UT-100C from an USB 2.0 port connected directly to the motherboard, to one USB 3.0 of a **POWERED** USB 3.0 HUB the problem has gone.

So, from my experience: **If you use the HiDes UT-100C device then use a powered USB HUB.**

Please, @lelegard add this info to the documentation about this device.

closed 2018-07-16 by lars18th

lars18th, 2018-08-06:
Hi @lelegard ,

After a lot of wated hours doing tests, I _really_ suggest to add this simple advise to the documentation:

> **If you use the HiDes UT-100C device then use a powered USB HUB.**

I hope you agree. :smiley:

lelegard, 2018-08-06:
Done, the documentation has been updated.

It is always good to provide that kind of advice with greedy USB devices. That being said, the product documentation from HiDes [here](http://www.hides.com.tw/product_cg74469_eng.html) states that the required power is only 390mA even when transmission and reception are active at the same time. So, it should fit in a USB 2.0 port because it is required to provide at least 500 mA. So, either the HiDes documentation is incorrect or your USB 2.0 port simply failed to provide 390 mA.