tsduck/tsduck #103: Suggestion: M2TS input and output support

lars18th, 2018-07-19:
Hi @lelegard ,

I see in some files and in the documentation some references to the M2TS format. I suspect the current support is limited to read TS packets with the length of 192 bytes, stripping the 4-bytes header.

However, this format is very convenient when you need to "replay" files. In addition, the header is very simple and easy to handle. So, if we can write files with timestampped marks in this format, and after read the file using these time marks, then we can reproduce any real-time environment with _identical_ instant bitrate.

What you think about this idea?

lelegard, 2018-08-06:
Currently, the only support for M2TS files is in `tsresync` which removes the packet headers to convert the file into a TS file (but losing the time stamp information).

My understanding is that if both input and output for `tsp` are M2TS files, you would like that the time stamp of each input packet is exactly copied in the corresponding output packet. Do I understand correctly?

In that case, the situation applies only when the input and output plugins are `file` _and_ both files have M2TS format. You have to understand that this would change the entire internal organisation of `tsp` regarding the global packet buffer. This would also break the interface of the input and output plugins. I mean the generic API of all input and output plugins, not only the `file` plugin.

closed 2018-09-16 by lelegard

lars18th, 2018-09-20:
Hi @lelegard ,

The user case for the request of this enhancement is this:

- Imagine you catch a MPTS with CBR to a file doing some pid filtering.
- After that you like to replay this file.
- Futhermore, you like to support the replay consistency even when incomming errors appear.

So, the best solution in this case is write timestamps in the file; and the time marks are internally generated to overcome a signal loss. And this exactlly what does the M2TS format: MPEG-TS packets with added timestamps.

You think you can support some kind of this?