tsduck/tsduck #106: compile error on linux-lite

allegfede, 2018-07-24:
linux lite is a stripped down ubuntu based distro. I got a 64 bit version that is equivalent to the ubuntu 14.04

lsb_release -a:
Distributor ID:	Ubuntu
Description:	Linux Lite 2.8
Release:	14.04
Codename:	trusty

now i follow the guide (https://tsduck.io/doxy/building.html)

but i got a lot of errors when the make tries to download the decktek sdk ...

i have done this:
sudo apt-get install g++ dpkg-dev doxygen dos2unix graphviz curl pcscd libpcsclite-dev libcppunit-dev libcurl3 libcurl3-dev
then i make a src folder in my home
cd src
git clone https://github.com/tsduck/tsduck.git
make

and now i got those error:

curl -D /dev/stdout 'https://www.dektec.com/products/SDK/DTAPI/Downloads/LatestLinuxSDK' | grep -i 'Location:' | sed -e 's/.*: *//' -e 's/.*\///' -e 's/\r//g' >LinuxSDK.version
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl -L 'https://www.dektec.com/products/SDK/DTAPI/Downloads/LatestLinuxSDK' -o 
curl: option -o: requires parameter
curl: try 'curl --help' or 'curl --manual' for more information
make[1]: *** [download] Error 2

is that normal? i miss something?

thanks

lelegard, 2018-08-06:
Hello @allegfede 

This is certainly not normal. I do not have "linux lite", just Ubuntu, Fedora and CentOS.

First, let's see what's going wrong. The first curl command uses the generic URL ending in "LatestLinuxSDK". This is a generic URL which is maintained by Dektec and redirects to the latest file. The first curl command does not redirect and gets the actual name of the lastest file using option -D. The name of this file is saved in LinuxSDK.version. The second curl command performs redirections and saves the output using the file name in LinuxSDK.version.

In your case, LinuxSDK.version seems to be empty. The first curl command failed to get the latest file name.

Right now, it works on my Ubuntu system. However, Dektec released a new version in July. Maybe you simply tried at a time where the redirection to the new version was not yet properly set by Dektec. Can you try right now?

allegfede, 2018-08-06:
trying

allegfede, 2018-08-06:
have to run the "make" a pèair of times before it start downloading something ... after the process started, this is the result ....

[CXX] tsPacketizer.cpp
In file included from /usr/include/string.h:640:0,
                 from /usr/include/c++/4.8/cstring:42,
                 from tsPlatform.h:638,
                 from tsMPEG.h:37,
                 from tsPacketizer.h:36,
                 from tsPacketizer.cpp:34:
In function 'void* memset(void*, int, size_t)',
    inlined from 'bool ts::Packetizer::getNextPacket(ts::TSPacket&)' at tsPacketizer.cpp:204:44:
/usr/include/x86_64-linux-gnu/bits/string3.h:81:32: error: call to '__warn_memset_zero_len' declared with attribute warning: memset used with constant zero length parameter; this could be due to transposed parameters [-Werror]
       __warn_memset_zero_len ();
                                ^
cc1plus: all warnings being treated as errors
make[2]: *** [release-x86_64/tsPacketizer.o] Error 1

have no idea of the problem :-(

lelegard, 2018-08-08:
After checking, Ubuntu 14.04 had GCC 4.8.2. I assume that the corresponding version of Linux Lite had the same.

This GCC is too old. It is not even C++11 compliant (in 2018...). Even GCC 4.8.5 (RHEL/CentOS 7.5) is not fully C++11 compliant and TSDuck contains a few #ifdef to support it. But not older versions.

You have to use a decently recent version of GCC, sorry.

allegfede, 2018-08-08:
OK now i have info on the direction to work on. Updating gcc.

Thanks