tsduck/tsduck #108: Retiming the PAT/PMT/SDT

joolzg, 2018-08-17:
Im mixing 6 programs into one and noticed the PAT/PMT are coming every 58ms, which is pretty fast.

is there a way to retime this so they come at a slower rate say 500ms

lelegard, 2018-08-17:
Hi @joolzg,

In a broadcast / multicast environment, the typical cycling time for PAT and PMT is 100 ms. 500 ms is too long. It has a negative impact on the user experience because of the increased channel acquisition time.

Now, to answer your technical question, I do not see any immediate way to do so on live streams with PAT and PMT's potentially changing their content / version over time.

If the PAT and PMT are static (their content do not change), you can extract them once and store them in binary files, one per table, using the plugin `tables`. Then, in the live stream processing, first replace the PAT and PMT PID's by null packets (plugin `filter`) and then insert the binary files in their respective PID's with a lower bitrate (plugin `inject`, one instance per PID).

If the PAT and PMT can potentially change over time, there is one possible track to follow, although not complete enough to exactly achieve what you want. The scenario is similar. In the live processing chain:

- Extract the PAT and PMT tables in binary files (plugin `tables`).
- Replace their PID with null packets (plugin `filter`).
- Use plugin `insert`, one per PID, with a different bitrate and option `--poll-files`. This option does not complain when input files do not exist and reloads the content of the files each time they are externally modified.

This would do what you want with one exception: In the plugin `tables`, the output binary file remains open all the time and new sections are appended. We could add one option in `tables` to open/write/close the output file for each new table (each _new version of a table_). With this option, the scenario is complete to achieve what you want. Will think about it.

joolzg, 2018-08-18:
Sounds great. Looking forward to the enhancement.

lelegard, 2018-08-18:
Options `--rewrite-binary` and `--rewrite-xml` have been added to the plugin `tables`.

These options will be in version 3.14. You may pull and rebuild TSDuck if you'd like to try them now.

Here is a working example
```
# Bitrate for 500 ms repetition rate, assuming one TS packet per table.
BITRATE=$((188 * 8 * 2))

rm -f $TMPDIR/pat.bin $TMPDIR/pmt.bin

tsp -v \
    -I file $SOURCEDIR/astra192E-ts1120-2018-05-11.ts \
    -P regulate \
    -P analyze --table-analysis -o $TMPDIR/test-in.txt \
    -P tables -p 0x00 -t 0x00 -b $TMPDIR/pat.bin --rewrite-bin \
    -P tables -p 0x64 -t 0x02 -b $TMPDIR/pmt.bin --rewrite-bin \
    -P filter -n -p 0x00 -p 0x64 -s \
    -P inject --poll-files $TMPDIR/pat.bin=500 -p 0x00 -b $BITRATE -s \
    -P inject --poll-files $TMPDIR/pmt.bin=500 -p 0x64 -b $BITRATE -s \
    -P analyze --table-analysis -o $TMPDIR/test-out.txt \
    -O drop
```

Here we simulate a live reception using a previous capture from the French DTTV, [file here](https://tsduck.io/streams/france-astra-19.2E/astra192E-ts1120-2018-05-11.ts). We play the file once but we regulate the transmission at the playout speed (plugin `regulate`).

The input repetition rate of the PAT and all PMT's is 100 ms. In file `test-in.txt`, analysis before modification, we can see:
```
|=============================================================================|
|  PID: 0x0000 (0)                                                       PAT  |
|-----------------------------------------------------------------------------|
|  0x00 (0, PAT), TID ext: 0x0460 (1120)                                      |
|      Repetition: ...... 99  ms  Section cnt: ...... 615                     |
|      Min repet.: ...... 91  ms  Table cnt: ........ 615                     |
|      Max repet.: ..... 102  ms  Version: ........... 25                     |
|=============================================================================|
...
|=============================================================================|
|  PID: 0x0064 (100)                                                     PMT  |
|  Service: 0x2649 (9801) F3 BNORMANDIE                                       |
|-----------------------------------------------------------------------------|
|  0x02 (2, PMT), TID ext: 0x2649 (9801)                                      |
|      Repetition: ...... 99  ms  Section cnt: ...... 614                     |
|      Min repet.: ...... 85  ms  Table cnt: ........ 614                     |
|      Max repet.: ..... 107  ms  Version: ............ 2                     |
|=============================================================================|
|  PID: 0x00C8 (200)                                                     PMT  |
|  Service: 0x264A (9802) F3 PAYS DE LOIRE                                    |
|-----------------------------------------------------------------------------|
|  0x02 (2, PMT), TID ext: 0x264A (9802)                                      |
|      Repetition: ..... 100  ms  Section cnt: ...... 612                     |
|      Min repet.: ...... 86  ms  Table cnt: ........ 612                     |
|      Max repet.: ..... 203  ms  Version: ............ 2                     |
|=============================================================================|
```

We implement the proposed scenario:

- Extract PAT and one PMT (PID 0x64) in binary files, rewriting the file of each table using the new option `--rewrite-binary`.
- Remove PID's 0x00 (PAT) and 0x64 (PMT), replacing them with stuffing.
- Inject PAT and PMT from the extracted binary files, recreating the previous PID's, using a new bitrate corresponding to a 500 ms repetition rate.

OK, I wrote that 500 ms is not appropriate. But this is just an example and the input was already at 100 ms, so we need something different on output to show that it works...

Excerpt from the log:
```
* tables: creating /Users/thierry/tmp/pmt.bin
* tables: creating /Users/thierry/tmp/pat.bin
* inject: file /Users/thierry/tmp/pat.bin created
* inject: loaded 1 sections from /Users/thierry/tmp/pat.bin, repetition rate: 500 ms
[... 8 lines cut ...]
|-----------------------------------------------------------------------------|
|  0x00 (0, PAT), TID ext: 0x0460 (1120)                                      |
|      Repetition: ..... 500  ms  Section cnt: ...... 121                     |
|      Min repet.: ..... 497  ms  Table cnt: ........ 121                     |
|      Max repet.: ..... 502  ms  Version: ........... 25                     |
|=============================================================================|
...
|=============================================================================|
|  PID: 0x0064 (100)                                                     PMT  |
|  Service: 0x2649 (9801) F3 BNORMANDIE                                       |
|-----------------------------------------------------------------------------|
|  0x02 (2, PMT), TID ext: 0x2649 (9801)                                      |
|      Repetition: ..... 500  ms  Section cnt: ...... 121                     |
|      Min repet.: ..... 496  ms  Table cnt: ........ 121                     |
|      Max repet.: ..... 503  ms  Version: ............ 2                     |
|=============================================================================|
|  PID: 0x00C8 (200)                                                     PMT  |
|  Service: 0x264A (9802) F3 PAYS DE LOIRE                                    |
|-----------------------------------------------------------------------------|
|  0x02 (2, PMT), TID ext: 0x264A (9802)                                      |
|      Repetition: ..... 100  ms  Section cnt: ...... 603                     |
|      Min repet.: ...... 86  ms  Table cnt: ........ 603                     |
|      Max repet.: ..... 203  ms  Version: ............ 2                     |
|=============================================================================|
```

lelegard, 2018-09-11:
Version 3.14 is out. Closing this one.

closed 2018-09-11 by lelegard

jdwiegman, 2019-06-20:
@lelegard I am trying to run similar code to this as well (live content from a pipe), but I get `Error: inject: input bit-rate unknown or too low, specify --inter-packet` unless I add a -b BITRATE to the root tsp command. Is the intended behaviour? It seems like the filter command initially receives a zero bit-rate and fails out.

lelegard, 2019-06-22:
Hi @jdwiegman 

Note that this topic is old and closed. So, it is likely that no one watches it.

To insert a tables, the plugin needs to know the TS bitrate. How can `tsp` know the bitrate? There are several solutions:
1. If the input device has a fixed rate input, such as a demodulator or ASI, then the bitrate is known.
2. PCR's are analysed. If there is no PCR in any PID or the PCR's are weird and unsynchronised, then no bitrate can be computed. This is probably the cause of your problem.
3. Specify the bitrate on the `tsp` command line.

If you think your TS contains PCR's but the bitrate is unknown, then open a new issue. Please provide the output of `tsanalyze` from your stream.