tsduck/tsduck #112: tsstuff over estimating input bit rate and overstuffing file

marti566, 2018-08-24:
I'm trying to stuff a transport stream that has 100 programs in it and i'm trying to stuff it to about 42Mb/s. output from 
`tsbitrate -a originalFile.ts` is 
`TS bitrate: 33,156,038 b/s (188-byte), 35,977,828 b/s (204-byte)` 
using tsstuff `tsstuff -b 42283562 o
[originalFile.zip](https://github.com/tsduck/tsduck/files/2319910/originalFile.zip)
riginalFile.ts > stuffed.ts`
gives the error `tsstuff: input bitrate higher than target bitrate, cannot stuff`
and over stuffs the stream to
`tsbitrate stuffed.ts`
`TS bitrate: 55,091,052 b/s (188-byte), 59,779,652 b/s (204-byte)`

lelegard, 2018-08-27:
Hi @marti566 

Your stream is a bit weird. Its average bitrate is 33 Mb/s but it is extremely erratic, much more than a standard variable bitrate stream.

I have enhanced the `tsstuff` command to display more accurate messages:
```
using PID 101 (0x0065) as reference
tsstuff: input bitrate locally higher (86,134,080 b/s) than target bitrate, cannot stuff
tsstuff: input bitrate locally higher (206,153,280 b/s) than target bitrate, cannot stuff
tsstuff: input bitrate locally higher (99,264,000 b/s) than target bitrate, cannot stuff
tsstuff: input bitrate locally higher (86,134,080 b/s) than target bitrate, cannot stuff
tsstuff: input bitrate locally higher (203,446,080 b/s) than target bitrate, cannot stuff
...
```

These messages are the same as the ones you mention, except that the local bitrate (between two PCR's) is displayed, explaining why stuffing is not locally possible.

The way `tsstuff` works is the following:

- Find a reference PID with PCR (system clock).
- Work on segments which are defined by two successive PCR timestamps.
- Consider all segments in the file, one by one.
- Evaluate how many null packets should be inserted in this segment to reach the target bitrate.
- Insert this amount of null packets all along the segment.
- Account the remaining bits (less than one full packet) for next segment.

In your file, the bitrate of each segment is erratic, sometimes exceeding the target bitrate by far.

Moreover, most video PID's are incorrectly stamped. There are 100 services and 100 video PID's. The PMT of each service indicates its video PID as PCR PID. However, only the PID 101 seems correctly stamped (1519 PCR values). The PID 151 contains only 18 PCR values. All other 98 video PID's have zero PCR values. So, at least the signalization is incorrect (PCR PID's in PMT's do not contain PCR's).

The final bitrate value includes parts which were stuffed to 42 Mb/s and parts which were already much higher than this, giving an average which is higher than 42 Mb/s.

Anyway, given the extremely erratic nature of the stream, average bitrate values do not really mean anything here.

lelegard, 2018-08-27:
Some more debug message:

```
segment: 33 ms, 181 packets, 8,166,720 b/s
segment: 33 ms, 718 packets, 32,396,160 b/s
segment: 33 ms, 298 packets, 13,445,760 b/s
segment: 33 ms, 218 packets, 9,836,160 b/s
segment: 33 ms, 218 packets, 9,836,160 b/s
segment: 33 ms, 200 packets, 9,024,000 b/s
segment: 33 ms, 9 packets, 406,080 b/s
segment: 33 ms, 1,909 packets, 86,134,080 b/s
segment: 33 ms, 233 packets, 10,512,960 b/s
segment: 33 ms, 255 packets, 11,505,600 b/s
segment: 33 ms, 213 packets, 9,610,560 b/s
segment: 33 ms, 200 packets, 9,024,000 b/s
segment: 33 ms, 188 packets, 8,482,560 b/s
segment: 33 ms, 216 packets, 9,745,920 b/s
segment: 33 ms, 4,569 packets, 206,153,280 b/s
segment: 33 ms, 2,200 packets, 99,264,000 b/s
segment: 33 ms, 218 packets, 9,836,160 b/s
segment: 33 ms, 222 packets, 10,016,640 b/s
segment: 33 ms, 20 packets, 902,400 b/s
segment: 33 ms, 255 packets, 11,505,600 b/s
segment: 33 ms, 213 packets, 9,610,560 b/s
segment: 33 ms, 200 packets, 9,024,000 b/s
segment: 33 ms, 1,909 packets, 86,134,080 b/s
segment: 33 ms, 200 packets, 9,024,000 b/s
segment: 33 ms, 182 packets, 8,211,840 b/s
...
```

The PCR in PID 101 have a steady interval of 33 ms but the number of total packets between two of them is not constant. There are bursts and holes. The most frequent pattern is about 200 packets per PCR segment, giving a plausible bitrate of 9 Mb/s. But there are bursts of thousands of packets, making the stream impossible to smoothen.

marti566, 2018-08-27:
I see, well i'll close the issue as it seems a problem with my source file rather than the software, thank you.

closed 2018-08-27 by marti566