tsduck/tsduck #113: merge: EMM PID conflict

rezna11, 2018-08-26:
Hello.

I'm running into a problem when merging. I have 2 multicast streams, each from a different DVB-S tuner.
I'm trying to take these 2 streams and merge them into ASI output:

`tsp -v --bitrate 68000000 -I null -P regulate  -P merge -p 0 -p 17 "tsp -I ip 239.5.5.5:5500" -P merge "tsp -I ip 239.6.6.6:6600" -O dektec --device 0`

If there's just one merged stream (the 1st one), everything works. But by adding the second source for merging (239.6.6.6:6600), I get the following error, repeated constantly and there's no output.

```
C:\mux>tsp -v --bitrate 68000000 -I null -P regulate  -P merge -p 0 -p 17 "tsp -I ip 239.5.5.5:5500" -P merge "tsp -I ip 239.6.6.6:6600" -O dektec --device 0
* regulate: minimum packet burst duration is 2,000,000 nano-seconds
* null: input bitrate is 68,000,000 b/s
* dektec: using Dektec device 0, output channel 0 (DTA-140 port 2)
* dektec: output fifo size: 8,388,608 bytes, max: 8,388,608 bytes, typical: 8,388,608 bytes
* dektec: initial output bitrate: 68,000,000 b/s
* regulate: regulated at bitrate 68,000,000 b/s
* Error: merge: EMM PID conflict, PID 0x1DCE (7630) referenced in the two streams, dropping from merged stream
* Error: merge: EMM PID conflict, PID 0x1DCE (7630) referenced in the two streams, dropping from merged stream
* Error: merge: EMM PID conflict, PID 0x1DCE (7630) referenced in the two streams, dropping from merged stream
* Error: merge: EMM PID conflict, PID 0x1DCE (7630) referenced in the two streams, dropping from merged stream
* Error: merge: EMM PID conflict, PID 0x1DCE (7630) referenced in the two streams, dropping from merged stream
* Error: merge: EMM PID conflict, PID 0x1DCE (7630) referenced in the two streams, dropping from merged stream
* Error: merge: EMM PID conflict, PID 0x1DCE (7630) referenced in the two streams, dropping from merged stream
```

And this is what's confusing see. I don't understand what it means. The 1st stream (239.5.5.5:5500) has no PID 7630. The second stream (239.6.6.6:6600) does but that's correct, it's an EMM PID which I need. And I don't think there's anything wrong with the command that creates the second stream itself. It only removes some of the services:

`tsp -v -I dvb --adapter 1 --frequency 11,555,000,000 --symbol 29,950,000 --polarity vertical --delivery-system DVB-S2 -P svremove 301 -P svremove 302 -P svremove 304 -P svremove 305 -P svremove 306 -P svremove 307 -P svremove 894 -P svremove 895 -P svremove 900 -P svremove 901 -P svremove 902 -P svremove 904 -P svremove 906 -P svremove 907 -P svremove 908 -P svremove 914 -P svremove 916 -P svremove 918 -P svremove 923 -O ip -l 10.0.0.13 239.6.6.6:6600`

lelegard, 2018-08-27:
Hi Jan,

The message "EMM PID conflict" comes from the merging of the CAT's. The message does not say that the given PID is _present_ in the two streams but that an EMM PID is _referenced_ in the two streams. In other words, this means that the same PID is declared as an EMM PID in the two streams.

But, although referenced, the actual PID is not present in at least one of the streams. Otherwise, when actual packets are found from the same PID, the message is "PID _exists_ in the two streams".

If you manipulate FTA services only, you do not care about EMM's and you may remove all EMM PID's and their reference by first using the plugin `filter` to remove PID 1, followed by the plugin `rmorphan`. The first filter removes the CAT, making all EMM PID's orphan (ie. not referenced anywhere). The second filter removes all orphan PID's, including the PID's which were previously referenced as EMM.

lelegard, 2018-08-27:
About the solution, I missed your remark:

>  The 1st stream (239.5.5.5:5500) has no PID 7630. The second stream (239.6.6.6:6600) does but that's correct, it's an EMM PID which I need.

This means the PID 7630 is still referenced in the CAT of the first stream and you should remove the reference from the CAT in the first stream using `-P cat --remove-pid 7630` (or `--remove-casid` or any other equivalent modification) in the merged `tsp` command.

rezna11, 2018-08-27:
Hi Thierry. I don't know about this. 

If I add `-P tables -p 1 -o cat.txt` to the 1st stream's command before outputting to 239.5.5.5:5500, I don't see 7630 in there, I can only see other EMM PIDs. 

And when I do this for the second stream, I do see PID 7630 in there, as expected.

I don't even think the problem is specific to these 2 input streams. If I swap the merging order, then the error is different, complaining about PID 48, which is EMM PID from the other stream:

```
C:\mux>tsp -v --bitrate 68000000 -I null -P regulate  -P merge -p 0 -p 17 "tsp -I ip 239.6.6.6:6600" -P merge "tsp -I ip 239.5.5.5:5500" -O dektec --device 0
* regulate: minimum packet burst duration is 2,000,000 nano-seconds
* null: input bitrate is 68,000,000 b/s
* dektec: using Dektec device 0, output channel 0 (DTA-140 port 2)
* dektec: output fifo size: 8,388,608 bytes, max: 8,388,608 bytes, typical: 8,388,608 bytes
* dektec: initial output bitrate: 68,000,000 b/s
* regulate: regulated at bitrate 68,000,000 b/s
* Error: merge: EMM PID conflict, PID 0x0030 (48) referenced in the two streams, dropping from merged stream
* Error: merge: EMM PID conflict, PID 0x0030 (48) referenced in the two streams, dropping from merged stream
* Error: merge: EMM PID conflict, PID 0x0030 (48) referenced in the two streams, dropping from merged stream
* Error: merge: EMM PID conflict, PID 0x0030 (48) referenced in the two streams, dropping from merged stream
* Error: merge: EMM PID conflict, PID 0x0030 (48) referenced in the two streams, dropping from merged stream
```

rezna11, 2018-08-27:
perfect, now it works as expected
thank you!