tsduck/tsduck #114: MPEG-TS stream DVB-CSA2 descrambling and playback using VLC

pallavi-girish, 2018-08-29:
Hi,

I am trying to use DVB scrambler and descrambler plugins to encrypt and decrypt MPEG-TS streams using DVB-CSA2 and playback with VLC player.

I use these commands,
$ tsp -I file /home/test.ts -P scrambler --dvb-csa2 --cw 65eea17288b0c494 1 -O file /home/csa2_scrambled_test.ts

$ tsp -I file /home/csa2_scrambled_test.ts -P descrambler --dvb-csa2 --cw 65eea17288b0c494 1 -O play
VLC media player 2.2.4 Weatherwax (revision 2.2.4-3-g2fc51dd)
[0000000104b027e8] core interface error: no suitable interface module
[0000000100212928] core libvlc error: interface "(null),none" initialization failed
[0000000100212928] core libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[0000000104b44838] core playlist: end of playlist, exiting

**[00000001003e1758] core stream error: cannot pre fill buffer**

[0000000104b44838] core playlist: end of playlist, exiting

It never plays audio-video. It never even shows the VLC window!

Regards,
Pallavi

lelegard, 2018-08-29:
Hi @pallavi-girish 

The commands you give are correct. Just to check, I tried them using an SPTS file and they work as expected (except that the version of VLC is 3.0.3, standard version on Fedora 28).

So, there could be a problem with the input file or VLC  or both.

First, to eliminate the less probable cause, how many services are present in `test.ts` ? Just one ? If there are more than one, the one you descramble is service id 1 but is it the one that VLC plays by default ?

Then, I suggest the following troubleshooting scenario until one test fails.

1) Try to play the input file directly with VLC: `vlc /home/test.ts`. Does it play correctly ?

2) Try to play the input file through tsp: `tsp -I file /home/test.ts -O play`.

3) Descramble in a file (`tsp -I file /home/csa2_scrambled_test.ts -P descrambler --dvb-csa2 --cw 65eea17288b0c494 1 -O file /home/csa2_descrambled_test.ts`) and then play the descrambled file with VLC.

4) Compare the descrambled file with the input one: `tscmp -c /home/test.ts /home/csa2_descrambled_test.ts`. The first few packets are probably different and replaced by null packets (this is what the scrambling plugin does until it finds the target service). But all other packets should be identical to the input ones.

Let us know which of these tests fail.

pallavi-girish, 2018-08-29:
Hi lelegard,

The test.ts stream has only one service with service id 1.
I am able to play test.ts directly with VLC and through tsp.
I tried descrambling in a file, but the descrambled file does not play with VLC.

$ tsp -I file /home/csa2_descrambled.ts -O play
VLC media player 2.2.4 Weatherwax (revision 2.2.4-3-g2fc51dd)
[00000001002516f8] core interface error: no suitable interface module
[0000000100230fd8] core libvlc error: interface "(null),none" initialization failed
[0000000100230fd8] core libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[00000001002a19c8] core playlist: end of playlist, exiting
**[000000010d12a5f8] core stream error: cannot pre fill buffer**
[00000001002a19c8] core playlist: end of playlist, exiting

Regards,
Pallavi

lelegard, 2018-08-29:
What about:
```
vlc /home/csa2_descrambled.ts
```
And
```
tscmp -c /home/test.ts /home/csa2_descrambled.ts
```
And, finally, an up-to-date version of VLC (3.0.3).

pallavi-girish, 2018-08-29:
I updated to VLC 3.0.3
Only video playback, no audio in both the cases :
 1. vlc /home/csa2_descrambled.ts
 2. $ tsp -I file /Users/pgirish/Desktop/pgirish/projects/8.XPCA/TSDuck/csa2_scrambled.ts -P descrambler --dvb-csa2 --cw 65eea17288b0c494 1 -O play
VLC media player 3.0.3 Vetinari (revision 3.0.3-1-0-gc2bb759264)
[00007fb67006d8c0] main interface error: no suitable interface module
[00007fb670016bf0] main libvlc error: interface "(null),none" initialization failed
[00007fb670016bf0] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[00007fb6701031c0] videotoolbox decoder: Using Video Toolbox to decode 'h264'
**[00007fb6701031c0] main decoder error: buffer deadlock prevented**
[00007fb670068d30] main playlist: end of playlist, exiting

tscmp -c /home/test.ts /home/csa2_descrambled.ts
* Packet 0 differ at offset 1, 19/20 bytes differ, PID 0/8191, packet 0/0 in PID

The first few packets are different. But all other packets are identical to the input.

lelegard, 2018-08-29:
So, I assume that playing the original file does not work either (`vlc /home/test.ts`)

pallavi-girish, 2018-08-29:
Audio-Video playback is successful with original file.

$ tsp -I file /home/test.ts -O play
VLC media player 3.0.3 Vetinari (revision 3.0.3-1-0-gc2bb759264)
[00007f8f3243a150] main interface error: no suitable interface module
[00007f8f32701a30] main libvlc error: interface "(null),none" initialization failed
[00007f8f32701a30] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[00007f8f378f4e20] videotoolbox decoder: Using Video Toolbox to decode 'h264'
[00007f8f3259e7c0] mpeg4audio packetizer: AAC channels: 2 samplerate: 44100
[00007f8f32615720] main playlist: end of playlist, exiting

lelegard, 2018-08-30:
OK, so:

- The input file plays correctly.
- The descrambled one does not.
- The two files are identical except the first packet (message `PID 0/8191` means that the original packet contained an instance of the PAT and the new one is a null packet).

Having no PAT is clearly a good reason for not being able to play a file. However, the PAT should be repeated, typically every 100 ms. So, missing the first instance of the PAT should only drop the first 100 ms of content. If the descrambled file does not play, the only rational reason I can see is that your input file contains one single instance of the PAT (in packet zero).

You can verify this using this command on the original file:
```
tstables --all-sections --log --packet --pid 0 /home/test.ts
```

If you see only one line, the hypothesis is right. And your file is at least unusual, at worst incorrect.

Now, how to scramble that kind of file (with no repeated PSI/SI) ? The reason why the `scrambler` plugin nullifies all packets before identifying the service is that, as long as the service PMT is unknown, it is impossible to know if a PID belongs to the service to scramble or not. And we do not want to pass clear packets for the service to scramble. In a broadcast environment (including captured files), this makes sense. Probably, this approach is too conservative and the `scrambler` plugin should leave reserved PID's unmodified (PID < 0x20).

lelegard, 2018-08-30:
Since the filtering of PSI/SI in the initial phase was uselessly conservative, I updated the `scrambler` plugin  to let them pass in commit 70d8fdbdbd8f847f0fa3fc30f1477cce5e8a87cc. Other PID's are still nullified to avoid any clear content leak.

If the test from my previous comment demonstrates that your file has only one (non-repeated) PAT in packet zero, then this modification should solve your problem. Please pull, rebuild and let us know if your problem is fixed.

pallavi-girish, 2018-08-30:
Input file contains one single instance of the PAT. I verified using the command :
tstables --all-sections --log --packet --pid 0 /home/test.ts
Packet 0 to 0, PID 0x0000, TID 0x00, TIDext 0x0001, V0: 00 01 E1 00

I pulled the commit 70d8fdb and built on Mac OS. I followed the below steps :
1. git clone https://github.com/tsduck/tsduck
2. git checkout 70d8fdb
3. make
4. source src/tstools/release-x86_64/setenv.sh

**But Audio issue not resolved.**

$ which tsp
/home/tsduck/src/tstools/release-x86_64/tsp

$ tsp -I file /home/csa2_descrambled.ts -O play
VLC media player 3.0.3 Vetinari (revision 3.0.3-1-0-gc2bb759264)
[00007fb8e9585ef0] main interface error: no suitable interface module
[00007fb8eb001240] main libvlc error: interface "(null),none" initialization failed
[00007fb8eb001240] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[00007fb8e95d6a40] videotoolbox decoder: Using Video Toolbox to decode 'h264'
**[00007fb8e95d6a40] main decoder error: buffer deadlock prevented**
[00007fb8e9522a60] main playlist: end of playlist, exiting

lelegard, 2018-08-30:
OK, so now:

1. Generate the descrambled file.
2. Use `tscmp -c` to compare the original and descrambled file. Are they strictly identical? If not, please post the complete output of `tscmp -c`.
3. Does the original file plays correctly using a direct `vlc` command (not `tsp`)?
4. Same question for the descrambled file.

lelegard, 2018-08-30:
And by the way, you don't need to checkout the specific commit. The latest commit from the `master` branch is fine.

pallavi-girish, 2018-09-04:
Yes..Original file (/home/test.ts) plays correctly using direct vlc.
Using direct vlc with descrambled file (/home/csa2_descrambled.ts), 
Video playback is good but there is no audio. csa2_descrambled.ts.

$ tscmp -c /home/test.ts /home/csa2_descrambled.ts  
* Packet 0 differ at offset 1, 19/20 bytes differ, PID 0/8191, packet 0/0 in PID

$ tscmp -c -d /home/test.ts /home/csa2_descrambled.ts
* Packet 0 differ at offset 1, 19/20 bytes differ, PID 0/8191, packet 0/0 in PID
  Packet from /home/test.ts :
      ---- TS Header ----
      PID: 0 (0x0000), header size: 4, sync: 0x47
      Error: 0, unit start: 1, priority: 1
      Scrambling: 0, continuity counter: 0
      Adaptation field: no (0 bytes), payload: yes (184 bytes)
      ---- Full TS Packet Content ----
      47 60 00 10 00 00 B0 0D 00 01 C1 00 00 00 01 E1  G`..............
      00 E8 F9 5E 7D FF FF FF FF FF FF FF FF FF FF FF  ...^}...........
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF              ............
  Packet from /home/csa2_descrambled.ts:
      ---- TS Header ----
      PID: 8191 (0x1FFF), header size: 4, sync: 0x47
      Error: 0, unit start: 0, priority: 0
      Scrambling: 0, continuity counter: 0
      Adaptation field: no (0 bytes), payload: yes (184 bytes)
      ---- Full TS Packet Content ----
      47 1F FF 10 FF FF FF FF FF FF FF FF FF FF FF FF  G...............
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF FF FF FF FF FF FF FF FF              ............
  Differing area from /home/test.ts :
      60 00 10 00 00 B0 0D 00 01 C1 00 00 00 01 E1 00  `...............
      E8 F9 5E 7D                                      ..^}
  Differing area from /home/csa2_descrambled.ts:
      1F FF 10 FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
      FF FF FF FF                                      ....

lelegard, 2018-09-05:
Hi @pallavi-girish,

This is not rational. Your log indicates that packet 0 which originally contained a PAT (PID 0) is now replaced by stuffing (PID 8191). This is precisely what the modification I made was supposed to eliminate. 

I did a similar test with an SPTS file containing a PAT in packet 0 and a PMT in packet 1. After scrambling and descrambling, the initial file and descrambled file are 100% identical, byte by byte. And, of course, the output file plays correctly with VLC, video and audio.
```
tsp -I file cnbc.in.ts  -P scrambler   --dvb-csa2 --cw 65eea17288b0c494 0x272E -O file cnbc.scr.ts
tsp -I file cnbc.scr.ts -P descrambler --dvb-csa2 --cw 65eea17288b0c494 0x272E -O file cnbc.descr.ts

tscmp -c cnbc.in.ts cnbc.descr.ts
cmp -c cnbc.in.ts cnbc.descr.ts
```
First, make sure that you are using TSDuck version 3.14-878 or higher.
Then, could you post a copy of your file (truncate it if it is too large).

pallavi-girish, 2018-09-05:
I verified with commit 70d8fdb. 
Audio-Video playback is successful.
Thank you!

closed 2018-09-05 by lelegard