tsduck/tsduck #115: Licence

alanbirtles, 2018-08-30:
As portions of this project are based on GPL code doesn't the whole project need to be GPL licenced? That's the unfortunate way that GPL works, all derivatives of GPL code must be GPL licenced.

lelegard, 2018-08-30:
Hi @alanbirtles 

Which parts are GPL? If you are talking about telxcc, no code was really _copied_ from the original telxcc code. The Teletext demux of TSDuck was rewritten in C++ but I had read the telxcc code first and I was clearly inspired by it. My reference to telxcc is more a deserved tribute to Petr Kutalek's work (the author of telxcc).

I am not a lawyer and cannot precisely decide if this is a derivative work or not, in the sense of the GPL. If the problem is only the mention to the GPL in the license file, I will remove it. If rewriting code after having read and understood GPL code is a license violation, then I will remove the corresponding code and completely drop Teletext support from TSDuck.

alanbirtles, 2018-08-30:
The licence file states "Telxcc is not fully used or included by TSDuck. However, some code was directly copied and adapted from Telxcc into TSDuck." If code was copied then its a derivative work.

If you looked at it and then made your own implementation that worked in a similar way that's a bit more debatable and probably depends how similar your code is to the original.

alanbirtles, 2018-08-30:
some discussion: https://stackoverflow.com/questions/1723606/should-translations-of-some-gpl-code-to-other-languages-be-licensed-as-gpl https://stackoverflow.com/questions/11939125/ported-gpl-library-should-be-under-gpl-as-well

lelegard, 2018-08-30:
I will modify the documentation and source code to describe more accurately how telxcc influenced the Teletext demux in TSDuck. If some day someone complains and exhibits legal evidences that there is a problem, I will completely drop support for Teletext. As always, GPL brings more problems than it solves...

alanbirtles, 2018-08-31:
I think given ts::TeletextDemux::processTeletextPacket is nearly identical to the original process_telx_packet down to using the same variable names I don't think you can claim its not a copy.

lelegard, 2018-08-31:
I see that you have spent enough time to locate one specific function inside a 200 000 lines of code open source project (TSDuck) and also fetched another open source project (Telxcc) that was abandoned several years ago by its original author (his GitHub repository was even deleted) and the code of which is not easy to read and finally compared the C code of one Telxcc function with the C++ code of one TSDuck function. 

Such effort requires some precise goal. So, please express yourself openly. What is your goal? What is your role?

alanbirtles, 2018-08-31:
I might want to use your library but probably wont be able to if its GPL

alanbirtles, 2018-08-31:
None of the above was hard to find, I just started from your (now deleted) comments that that method was copied from telxcc and then a quick google turned up the original function. Code being abandoned by its author doesn't change the fact that it is copyrighted and under a particular licence

lelegard, 2018-08-31:
TSDuck will never switch to GPL. The BSD license is just fine. BSD allows everyone to have its own conception of "freedom" and still cooperate with each other. GPL forbids cooperation with people who do not share their exact same narrow definition of "freedom".

If any confirmed legal problem arises with this specific function, TSDuck will drop Teletext support and will continue to be distributed under the BSD license. This is a firm statement.

lelegard, 2018-09-04:
Hi @alanbirtles,

I added the possibility to build the binaries, installers and source tarballs without any code from telxcc. On a user perspective, the plugin `teletext` is removed.

Use the following commands:

```
make rpm NOTELETEXT=true         # Fedora, Red Hat, CentOS, etc.
make deb NOTELETEXT=true         # Ubuntu, Debian, etc.
Build-Installer.ps1 -NoTeletext  # Windows
```

I hope that this addresses your concern.

alanbirtles, 2018-09-04:
Partly, this does make it possible to build a non-GPL version but I think you should revert 42d70b76a76bfdd5b650e1297207cff41a6d5bea and instead add a notice that the licence is GPL unless built with the commands you posted above. Just so nobody gets caught out in the future.

e.g. FFMPEG is LGPL by default, you have to pass a parameter to `./configure` to enable GPL modules.

closed 2018-09-17 by lelegard