tsduck/tsduck #116: [Performance] UDP receiver cpu utilization

yaneti, 2018-08-30:
Hi,

just something I noticed while doing a cursory profile of tsp with UDP input.

src/libtsduck/tsUDPReceiver.cpp:336   in ts::UDPReceiver::receive
` report.log(2, u"received UDP packet, source: %s, destination: %s", {sender.toString(), destination.toString()});`

The .toString calls here have around 10% impact in the fast path of the packet processing even when there is no debugging involved

I couldn't figure the best way to code around it so I'll leave that to you :)

lelegard, 2018-08-30:
Hi @yaneti 

Thank you for pointing this. I did not do any profiling on this part. This debug code was once added to toubleshoot strange networking issues. But you are right, the evaluation of the parameters can be costly compared to the rest of this code, even when the debug messages are eventually not issued. I just modified this in commit c4aa74ab2649bc870d3dbeb7a74e355d42da7084. I added a prior double-check of the debug level. Could you please confirm that this fixes your performance issue?

yaneti, 2018-08-30:
Hi @lelegard 

After this change, "toString" is nowhere to be seen in the top >%1 of a 10s perf profile, so I'd say "mission accomplished"

Thanks

closed 2018-08-30 by lelegard