tsduck/tsduck #117: Compiling failing on Mac OSX

tj800x, 2018-08-30:
In the libtsduck directory the link command is failing on my OSX High Sierra 10.13.4 with the following:
Undefined symbols for architecture x86_64:
  "std::__basic_file<char>::is_open() const", referenced from:
      ts::ByteBlock::writeToFile(ts::UString const&, std::_Ios_Openmode, ts::Report*) const in tsByteBlock.o
      ts::ByteBlock::appendFromFile(ts::UString const&, unsigned long, ts::Report*) in tsByteBlock.o
      ts::InputRedirector::~InputRedirector() in tsInputRedirector.o
      ts::OutputRedirector::~OutputRedirector() in tsOutputRedirector.o
      ts::SectionFile::saveBinary(ts::UString const&, ts::Report&) const in tsSectionFile.o
      ts::SectionFile::loadBinary(ts::UString const&, ts::Report&, ts::CRC32::Validation) in tsSectionFile.o
      ts::SubRipGenerator::close() in tsSubRipGenerator.o

```
$ uname -a
Darwin Toms-MacBook-Pro 17.5.0 Darwin Kernel Version 17.5.0: Mon Mar  5 22:24:32 PST 2018; root:xnu-4570.51.1~1/RELEASE_X86_64 x86_64
```

I isolated the link command down to:

`[libtsduck]$  "/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld" -demangle -lto_library /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/libLTO.dylib -dynamic -dylib -arch x86_64 -dylib_install_name @rpath/tsduck.so -macosx_version_min 10.13.0 -o release-x86_64/tsduck.so -L/opt/local/lib -L/usr/local/opt/pcsc-lite/lib/ -L/usr/local/lib release-x86_64/tsAACDescriptor.o release-x86_64/tsAC3Attributes.o release-x86_64/tsAC3Descriptor.o release-x86_64/tsAC4Descriptor.o release-x86_64/tsAES.o release-x86_64/tsAIT.o release-x86_64/tsAVCAttributes.o release-x86_64/tsAVCHRDParameters.o release-x86_64/tsAVCParser.o release-x86_64/tsAVCSequenceParameterSet.o release-x86_64/tsAVCTimingAndHRDDescriptor.o release-x86_64/tsAVCVUIParameters.o release-x86_64/tsAVCVideoDescriptor.o release-x86_64/tsAbstractAVCAccessUnit.o release-x86_64/tsAbstractAVCStructure.o release-x86_64/tsAbstractCASDate.o release-x86_64/tsAbstractDeliverySystemDescriptor.o release-x86_64/tsAbstractDemux.o release-x86_64/tsAbstractDescrambler.o release-x86_64/tsAbstractDescriptor.o release-x86_64/tsAbstractDescriptorsTable.o release-x86_64/tsAbstractMultilingualDescriptor.o release-x86_64/tsAbstractSignalization.o release-x86_64/tsAbstractTable.o release-x86_64/tsAbstractTablePlugin.o release-x86_64/tsAbstractTransportListTable.o release-x86_64/tsAdaptationFieldDataDescriptor.o release-x86_64/tsAncillaryDataDescriptor.o release-x86_64/tsApplicationDescriptor.o release-x86_64/tsApplicationIconsDescriptor.o release-x86_64/tsApplicationNameDescriptor.o release-x86_64/tsApplicationSharedLibrary.o release-x86_64/tsApplicationSignallingDescriptor.o release-x86_64/tsApplicationStorageDescriptor.o release-x86_64/tsApplicationUsageDescriptor.o release-x86_64/tsArgMix.o release-x86_64/tsArgs.o release-x86_64/tsAssociationTagDescriptor.o release-x86_64/tsAsyncReport.o release-x86_64/tsAudioAttributes.o release-x86_64/tsAudioLanguageOptions.o release-x86_64/tsAudioStreamDescriptor.o release-x86_64/tsBAT.o release-x86_64/tsBCD.o release-x86_64/tsBetterSystemRandomGenerator.o release-x86_64/tsBinaryTable.o release-x86_64/tsBitRateRegulator.o release-x86_64/tsBlockCipher.o release-x86_64/tsBouquetNameDescriptor.o release-x86_64/tsByteBlock.o release-x86_64/tsCADescriptor.o release-x86_64/tsCAIdentifierDescriptor.o release-x86_64/tsCASFamily.o release-x86_64/tsCASMapper.o release-x86_64/tsCASSelectionArgs.o release-x86_64/tsCAT.o release-x86_64/tsCOM.o release-x86_64/tsCPDescriptor.o release-x86_64/tsCPIdentifierDescriptor.o release-x86_64/tsCRC32.o release-x86_64/tsCableDeliverySystemDescriptor.o release-x86_64/tsCarouselIdentifierDescriptor.o release-x86_64/tsCerrReport.o release-x86_64/tsCipherChaining.o release-x86_64/tsComponentDescriptor.o release-x86_64/tsCondition.o release-x86_64/tsContentDescriptor.o release-x86_64/tsCopyrightDescriptor.o release-x86_64/tsCountryAvailabilityDescriptor.o release-x86_64/tsCueIdentifierDescriptor.o release-x86_64/tsCyclingPacketizer.o release-x86_64/tsDES.o release-x86_64/tsDIILocationDescriptor.o release-x86_64/tsDTSDescriptor.o release-x86_64/tsDVBCSA2.o release-x86_64/tsDVBCharset.o release-x86_64/tsDVBCharsetSingleByte.o release-x86_64/tsDVBCharsetUTF16.o release-x86_64/tsDVBCharsetUTF8.o release-x86_64/tsDVBHTMLApplicationBoundaryDescriptor.o release-x86_64/tsDVBHTMLApplicationDescriptor.o release-x86_64/tsDVBHTMLApplicationLocationDescriptor.o release-x86_64/tsDVBJApplicationDescriptor.o release-x86_64/tsDVBJApplicationLocationDescriptor.o release-x86_64/tsDataBroadcastDescriptor.o release-x86_64/tsDataBroadcastIdDescriptor.o release-x86_64/tsDataStreamAlignmentDescriptor.o release-x86_64/tsDeferredAssociationTagsDescriptor.o release-x86_64/tsDektecControl.o release-x86_64/tsDektecDevice.o release-x86_64/tsDektecInputPlugin.o release-x86_64/tsDektecOutputPlugin.o release-x86_64/tsDektecUtils.o release-x86_64/tsDektecVPD.o release-x86_64/tsDescriptor.o release-x86_64/tsDescriptorList.o release-x86_64/tsDiscontinuityInformationTable.o release-x86_64/tsDuckProtocol.o release-x86_64/tsECMGClient.o release-x86_64/tsECMGSCS.o release-x86_64/tsECMRepetitionRateDescriptor.o release-x86_64/tsEIT.o release-x86_64/tsEITProcessor.o release-x86_64/tsEMMGClient.o release-x86_64/tsEMMGMUX.o release-x86_64/tsEacemPreferredNameIdentifierDescriptor.o release-x86_64/tsEacemPreferredNameListDescriptor.o release-x86_64/tsEacemStreamIdentifierDescriptor.o release-x86_64/tsEnhancedAC3Descriptor.o release-x86_64/tsEnumeration.o release-x86_64/tsEutelsatChannelNumberDescriptor.o release-x86_64/tsException.o release-x86_64/tsExtendedEventDescriptor.o release-x86_64/tsExternalApplicationAuthorizationDescriptor.o release-x86_64/tsExternalESIdDescriptor.o release-x86_64/tsFatal.o release-x86_64/tsFileNameRate.o release-x86_64/tsForkPipe.o release-x86_64/tsGitHubRelease.o release-x86_64/tsGrid.o release-x86_64/tsGuard.o release-x86_64/tsGuardCondition.o release-x86_64/tsHDSimulcastLogicalChannelDescriptor.o release-x86_64/tsHEVCTimingAndHRDDescriptor.o release-x86_64/tsHEVCVideoDescriptor.o release-x86_64/tsHiDesDevice.o release-x86_64/tsHiDesDeviceInfo.o release-x86_64/tsHierarchyDescriptor.o release-x86_64/tsIBPDescriptor.o release-x86_64/tsIDSA.o release-x86_64/tsINT.o release-x86_64/tsIPAddress.o release-x86_64/tsIPAddressMask.o release-x86_64/tsIPMACGenericStreamLocationDescriptor.o release-x86_64/tsIPMACPlatformNameDescriptor.o release-x86_64/tsIPMACPlatformProviderNameDescriptor.o release-x86_64/tsIPMACStreamLocationDescriptor.o release-x86_64/tsIPSignallingDescriptor.o release-x86_64/tsIPUtils.o release-x86_64/tsIPv6Address.o release-x86_64/tsISO639LanguageDescriptor.o release-x86_64/tsISPAccessModeDescriptor.o release-x86_64/tsInputRedirector.o release-x86_64/tsIntegerUtils.o release-x86_64/tsLNB.o release-x86_64/tsLinkageDescriptor.o release-x86_64/tsLocalTimeOffsetDescriptor.o release-x86_64/tsLogicalChannelNumberDescriptor.o release-x86_64/tsMACAddress.o release-x86_64/tsMD5.o release-x86_64/tsMJD.o release-x86_64/tsMPEDemux.o release-x86_64/tsMPEG.o release-x86_64/tsMPEG4AudioDescriptor.o release-x86_64/tsMPEG4VideoDescriptor.o release-x86_64/tsMPEPacket.o release-x86_64/tsMacPList.o release-x86_64/tsMaximumBitrateDescriptor.o release-x86_64/tsMemoryUtils.o release-x86_64/tsMessageDescriptor.o release-x86_64/tsModulation.o release-x86_64/tsMonotonic.o release-x86_64/tsMultilingualBouquetNameDescriptor.o release-x86_64/tsMultilingualComponentDescriptor.o release-x86_64/tsMultilingualNetworkNameDescriptor.o release-x86_64/tsMultilingualServiceNameDescriptor.o release-x86_64/tsMultiplexBufferUtilizationDescriptor.o release-x86_64/tsMutex.o release-x86_64/tsNIT.o release-x86_64/tsNVODReferenceDescriptor.o release-x86_64/tsNames.o release-x86_64/tsNetworkNameDescriptor.o release-x86_64/tsNullReport.o release-x86_64/tsObject.o release-x86_64/tsOneShotPacketizer.o release-x86_64/tsOutputPager.o release-x86_64/tsOutputRedirector.o release-x86_64/tsPAT.o release-x86_64/tsPCR.o release-x86_64/tsPCRAnalyzer.o release-x86_64/tsPCRRegulator.o release-x86_64/tsPCSC.o release-x86_64/tsPESDemux.o release-x86_64/tsPESPacket.o release-x86_64/tsPIDOperator.o release-x86_64/tsPMT.o release-x86_64/tsPSILogger.o release-x86_64/tsPSILoggerArgs.o release-x86_64/tsPacketizer.o release-x86_64/tsParentalRatingDescriptor.o release-x86_64/tsPartialTransportStreamDescriptor.o release-x86_64/tsPlugin.o release-x86_64/tsPluginR [... cut ...]
      ts::ByteBlock::appendFromFile(ts::UString const&, unsigned long, ts::Report*) in tsByteBlock.o
      ts::InputRedirector::~InputRedirector() in tsInputRedirector.o
      ts::OutputRedirector::~OutputRedirector() in tsOutputRedirector.o
      ts::SectionFile::saveBinary(ts::UString const&, ts::Report&) const in tsSectionFile.o
      ts::SectionFile::loadBinary(ts::UString const&, ts::Report&, ts::CRC32::Validation) in tsSectionFile.o
      ts::SubRipGenerator::close() in tsSubRipGenerator.o

...many more undefined symbols not shown.

Any help would be appreciated.

Thanks,
Tom

lelegard, 2018-08-30:
Hum... I could not upgrade clang to 9.0.1. No update from Apple Store. I explicitly installed _Command Line Tools (macSO 10.13) for Xcode 9.4.1_ from https://developer.apple.com/download/more/ and the clang version is still 9.0.0:

```
$ which clang
/usr/bin/clang
$ clang --version
Apple LLVM version 9.0.0 (clang-900.0.38)
Target: x86_64-apple-darwin17.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
$ 
```
Other more recent versions from Apple Download are flagged as Beta.

Do you install Xcode from Apple Download or Update ? From HomeBrew ?

lelegard, 2018-09-04:
Hi @tj800x 

Any news on this ? Can you confirm that you used a stable version of clang / llvm ?

Your undefined symbol belongs to the standard C++ library, which tends to indicate a problem with the compilation setup.

lelegard, 2018-09-28:
After upgrading to MacOS Mojave and Xcode 10.0, I got Apple LLVM version 10.0.0 (clang-1000.11.45.2). TSDuck is compiled correctly and the complete test suite passes successfully.

closed 2018-10-04 by lelegard