tsduck/tsduck #119: Log timestamp accuracy question

rezna11, 2018-08-31:
Hello.
I'm trying to figure out an issue with TDT/TOT timing. I'm getting the following output from TSDuck:

```
* 2018/08/31 12:53:10 - history: 26: TDT: 2018/08/31 10:53:07 UTC
* 2018/08/31 12:53:12 - history: 27: TOT: 2018/08/31 12:53:08 LOCAL
* 2018/08/31 12:53:25 - history: 28: TDT: 2018/08/31 10:53:22 UTC
* 2018/08/31 12:53:26 - history: 29: TOT: 2018/08/31 12:53:23 LOCAL
```

And what I get first is 3 second difference between logged time and TDT time and 4 second difference between logged time and TOT time.

But then I get 3 second difference from logged time for both - TDT and TOT.

So what I'd like to know is how is the time in the log timestamp (`tsp -t`) calculated - is it the time when the packet was really received or time when TSDuck found some free time to print out the log message, so it could be delayed from the time when the packet was received? 
And is the logged time the real hours:minutes:seconds or are the seconds rounded in some way? What I'm trying to do is find out the real, exact, difference between current UTC time and the timestamp which it arrives in TOT and TDT.

lelegard, 2018-08-31:
> So what I'd like to know is how is the time in the log timestamp (tsp -t) calculated - is it the time when the packet was really received or time when TSDuck found some free time to print out the log message, so it could be delayed from the time when the packet was received?

This is the second option. The time stamp is the time at which the message is printed. The log is handled by a low-priority thread. Each time a plugin has to log a message, it stores the message string in a bounded queue and continues immediately. If the bounded queue is full, the message is silently dropped (unless option  `--synchronous-log` is used). The message queue is processed by the log thread when no other thread is waiting to run.

This design is used on purpose to avoid priority inversions or the "error avalanche effect" which is sometimes seen in applications where the threads actually write their log. The catastrophic scenario is the following: some errors occur; messages are written; text I/O being long, the application threads are delayed; threads being late to process, more errors occur; then more log messages are produced; etc until everything fail...

To avoid this, threads in `tsp` never wait for the log. Message logging is "best effort" only. It never runs at the expense of the application features.

That being said, having a constant 3 seconds shift is probably not caused by this asynchronism for several reasons:

- Noticeable delays occur only when the CPU load is constantly close to 100%.
- Most CPU's are multi-core and parallel threads run on multiple cores.
- 3 seconds is probably too long anyway.
- Having exactly the same 3 seconds every time (not 2 or 4 sometimes) is too regular.

So, I would think that the most probable causes are one of the following (or both at the same time):

- The clock of the system running `tsp` is a few seconds ahead from the broadcast system.
- There is some constant time shift from end to end, head-end processing, transmission, local tuner, `tsp` buffers, etc.

At a test, you may try to reduce the size if the global buffer in `tsp`.

rezna11, 2018-08-31:
Thanks for explaining how the log works. It explains why sometimes I see 3 second difference and sometime 4 second difference. But that brings me to the question whether it wouldn't be useful to have a feature that would show the time when a packet actually arrived? 
I know for a fact that the TDT time in the TS is wrong and it's caused by the multiplexer. But I wanted to know how big is the actual delay, compared to the NTP source. For example, how can I do `-P timeref --add` if I don't know how much or how to compare 2 different multiplexers and their TDT delay if there's no way to measure it.

lelegard, 2018-08-31:
> But that brings me to the question whether it wouldn't be useful to have a feature that would show the time when a packet actually arrived?

Because of the buffering between plugins, it would not be much more accurate to have the plugin generate its own timestamp before enqueuing the log message.

To get the accuracy you need, it would be necessary to get a system time stamp for each packet at the time they are received in the input plugin and keep it all along the plugin chain. That would change quite a lot of things in the `tsp` architecture. So, that will not make it a high priority modification I am afraid.

rezna11, 2018-08-31:
OK, I think everything is clear now so I can close this ticket. Thanks!

closed 2018-08-31 by rezna11