tsduck/tsduck #122: Insert SCTE pid into ts

mhzbg, 2018-09-07:
Apologies if this is noob question but I am trying to do the following: 
Create a ts stream from ffmpeg and send to tsp for SCTE35 splice insert. 

The challenge is that I need to add a stream type of type 0x86 to pid 500 (new pid).
Question: How do I add a pid (500 for example) with a stream type 0x86 ?

My assumption is that after I am able to add the above pid, I can use the spliceinject plugin to add splice events ?
An example will be very much appreciated and a beer at IBC guaranteed. 

Currently I am able to generate a TS stream with ffmpeg and send via local udp unicast to tsp, but not able to add splice events because I dont have the proper scte35 table 
Thanks

lelegard, 2018-09-08:
Hi @mhzbg 

To create the PID, you need to

1. Have enough stuffing so that packets for the new PID can be inserted by replacing null packets
2. Have the new PID correctly declared in the PMT of the service

For point 1, if you work on a real broadcast stream, there is always some stuffing. If you work on a synthetic stream from ffmpeg, there is probably none and you need to create some. In this case, use the tsp option `--add-input-stuffing` which inserts null packets in the middle of the input stream. For instance, `tsp --add-input-stuffing 1/10` adds 1 null packet every 10 input packets, which is more than enough for you. The inserted null packets will be used as replacement, when needed, for the new PID. At the end of the chain, before the output plugin, it is recommended to remove the remaining null packets using the plugin `filter`.

For point 2, the plugin `pmt` modifies a PMT on the fly. You can declare the new PID 500 with stream type 0x86. To fully comply with SCTE 35, it is also required to add a registration descriptor with id 0x43554549. If there are more than one service in the TS (probably not the case for you), you also need to specify the PMT PID or the service name or id.

This gives the following command skeleton:
```
tsp --add-input-stuffing 1/10 \
    -I ... \
    -P pmt --add-programinfo-id 0x43554549 --add-pid 500/0x86 \
    -P spliceinject ... \
    -P filter --negate --pid 0x1FFF \
    -O ...
```
Of course, all options are documented in the user's guide.

lelegard, 2018-09-08:
I also added a new example about SCTE 35 cue information insertion in the [TSDuck user's guide](https://tsduck.io/download/docs/tsduck.pdf), section 5.2.24.

closed 2018-09-17 by lelegard

edwardlai, 2019-07-23:
> Hi @mhzbg
>
> To create the PID, you need to
>
> 1. Have enough stuffing so that packets for the new PID can be inserted by replacing null packets
> 2. Have the new PID correctly declared in the PMT of the [...]

This command add-pid in PMT table . but  how to let the pid (500), keep sending ?
avoid the TR101/290 pid error
"Referred PID does not occur for more than 5 seconds, pkt 133084"

lelegard, 2019-07-31:
> This command add-pid in PMT table . but how to let the pid (500), keep sending ?
avoid the TR101/290 pid error
"Referred PID does not occur for more than 5 seconds, pkt 133084"

See my answer in #375