tsduck/tsduck #126: Plugin TS Encapsulation/Decapsulation

lars18th, 2018-09-20:
Hi @lelegard ,

I’m in return with another request to TSDuck. :smile: 
I hope you can understand the idea. So, please let me first to explain my “user case”:

The objective is: _from a regular DVB MPTS I like to process one program with FFMPEG (in fact, just only one program). The output needs to continue to be a valid DVB MPTS_.

The problem is that the FFMPEG tool is a very bad tool for MPEG-TS manipulation. However, it’s great for video processing (fortunately).

So the idea is to execute these tasks in a processing chain:

1. TSDUCK: Encapsulate all pids from other programs inside a private data stream (data).
2. TSDUCK: Encapsulate all DVB regular stuff inside another private data stream (tables).
3. TSDUCK: Create fake PAT/PMT tables for the target program.
4. FFMPEG: Process the A/V streams and copy the private data streams.
5. TSDUCK: Renumber resulting A/V streams to map the original values.
6. TSDUCK: Restore original DVB stuff from the “tables” private data stream.
7. TSDUCK: Restore other pids from other programs from “data” private data stream.

The outcome is then a regular MPTS (the same TS, DVB conformant) with a “processed” program.
Great, right? :wink: 

As a tip, the FFMPEG command needs to be like this:

    ffmpeg -i <input> ... -map 0:p:10000 -copy_unknown ... -f mpegts <output>

Where the program to process has the id 10000 and it contains only known streams (aka video, audio and teletext). And the parameter `-copy_unknown` does the trick of copying all private data (in our case the encapsulated data).

But, to accomplish this task... **what we need?** In fact, only a plugin that “encapsulate” and “decapsulate”. Here the description of it:

`-P encapsulator -p|--pid value -i|insert value,value -e|extract value,value --clear`
- the pid is the target pid for the stream;
- the insert is the list of pids to include when encapsulating;
- the extract is the list of pids when decapsulating;
- and the clear parameter indicates to clear the packets (free with NULLs) when inserting/extracting (so the default behavior is to copy the packet instead of clear it).

More or less this is similar to the `t2mi` plugin. However, in this case **the encapsulation is raw** (a simple bitstream of TS packets inside the payload of a pid), and it can be used for any processing task that you need to complete. Perhaps the code from the `t2mi` plugin can be reused, but in this case the output isn’t a file and the encapsultation/decapsulation is done in the same stream. Furthermore it can be done on-the-fly as the encapsulation task only consumes two TS packets as maximum. So you only need to write one packet when it is complete, and save the rest bytes for the next packet. Therefore, with only one buffer with the size of two packets is ecough to do the processing.

I hope this explanation is sufficient to understand the idea.
What do you think about that?

lelegard, 2018-09-20:
Hi @lars18th 

Your description is clear. I think I perfectly understand the idea.

It needs two plugins, one to encapsulate, one to decapsulate. There is one small issue. The encapsulated stream is larger than the original one. We need to add 1 extra TS packet every 37 original packets approximately. The encapsulation adds 4 bytes per new packet, plus one "pointer field" in most packets (to mimic the section packetisation and allow resynchronisation). As usual, tsp cannot create new packets, only replace or drop existing ones. So, a pre-requisite is a minimum amount of pre-existing stuffing (or use `--add-input-stuffing 1/36` for instance) and exclude stuffing from the encapsulation.

This is feasible but requests some work and the use case is very limited since it does not implement any standard. I keep the idea open but will likely not implement it in the short term.

lars18th, 2018-09-20:
Hi @lelegard ,

Thank you for the response!

Regarding the concept of the "replacing" null stuff... no problem! It's just my original idea. :wink: 

However, why you think is _required_ to add 4 bytes and one "pointer field"? Perhaps I missed something. But my idea is a brute MPEG-TS encapsulation in a PID. Some other tools do the same with different objectives (DAB ETI streams in MPEG-TS, multiple DVB CBR in one MPEG-TS, etc.). For sure, we have the T2-MI and GSE standards, but a simple encapsulation will be very useful to include an external process in the middle of a TSDUCK chain. The idea is "hide/unhide" some pids for the external tool. You can found a lot of other scenarios where this encapsulation can be useful.

Regards.

lelegard, 2018-09-20:
> However, why you think is required to add 4 bytes and one "pointer field"? 

If we encapsulate a packet, we need to encapsulate the complete packet, header included. Otherwise, essential information would be lost.

A TS packet is 188 bytes long, 4 bytes header, up to 184 bytes payload. So, to encapsulate data, you can get at most 184 bytes per "outer" packet. Since the original "inner" packet is 188 bytes long, it cannot fit into 184 bytes. The remaining 4 bytes will go into the next outer packet, leaving only 180 bytes for the next inner packet, pushing 8 bytes to the next outer packet, etc.

So, in the general case, when you decapsulate an inner packet, you "open" an outer packet and get 184 bytes payload. But the beginning of this payload is rarely the start of an inner packet. It is the middle of an inner packet.

So, assuming that you start in the middle of an encapsulated stream, or you have lost some packets, how do you find the start of the inner packet to decapsulate? Find the next 0x47? Could be a data byte in the middle of a payload. So, you need to find a mechanism to know if the outer packet contains an inner packet start and where it is in the outer payload. No need to reinvent the wheel, MPEG-2 has already invented the PUSI + pointer field mechanism for sections and it perfectly matches our need.

Then, knowing that all inner data are 188 bytes long into an outer payload of 184 bytes, there is a pointer field in almost all outer packets (but not exactly all).

lars18th, 2018-09-21:
Hi @lelegard ,

Good explanation. Thank you!

However, I still don't understand why a pinter is required. See this:

- Regular TS packer header has a _fixed_ size of 4 bytes (when the Adaption field flag is set to 0x01, so payload only and no adaptation field).
- Then the payload has a long of 184 bytes every time.
- In that case, 46 consecutive TS packets (with a size of 188 bytes) can be encapsulated over 47 consecutive TS packets (with a payload of 184 bytes).
- Therefore, you only need to mark the first packet (of the 47) with the PUSI flag; and there you can found the 0x47 start mark of the bitstream in the first byte of the payload (if not, then the bitstream has errors).

I see this technique with some DAB EDI-NI encapsulation inside MPEG-TS, and it's quite simple.

I don't request for a complex encapsulation, only a brute force one. In brief, put in 46 TS packets inside 47 TS packets, and viceversa.

Do you think it's difficult?

lelegard, 2018-09-21:
What you suggest is correct in the absence of packet loss. Transport streams are designed to be resilient to packet loss. Otherwise, what you suggest would also be valid for sections. The pointer field is required only when synchronisation was lost, both for sections and the private encapsulation we are talking about.

There is one possible optimisation. Since the PUSI + pointer field makes synchronisation explicit, we can remove the 0x47 sync byte and use only 187 bytes per packet. This is what is done in T2-MI, make synchronisation explicit and remove the implicit - but not 100% reliable - synchronisation of the 0x47. The overall volume does not change (gain: remove one 0x47 byte, loss: add one pointer field byte) but the synchronisation is more robust.

lelegard, 2018-09-21:
> Some devices and tools use the simple mode (with the 0x47 sync byte), so then the TSDUCK will be compatible with them.

What do you mean here?

If we define a proprietary encapsulation format which is defined nowhere, it does not need to be compatible with anything.

Or do you suggest an already existing encapsulation format? Which one?

Please do not suggest T2-MI. It is too complicated for the use case.

lars18th, 2018-09-21:
> Or do you suggest an already existing encapsulation format? Which one?

It's an _informal_ implementation. Some projects implement a simple MPEG-TS encapsulation using trivial techniques:
- Adding one 0x47 sync byte every 188 bytes (this converts _any_ bitstream in a _pseudo_ MPEG-TS).
- Or, adding a 4 bytes header every 1884 bytes and selecting one pid number (this converts _any_ bitstream to a valid MPEG-TS, that you can mix it after with the corresponding PSI/SI tables and generate a valid DVB stream).

As example, you can see the list of ETI-NI DAB ensembles encapsulated inside DVB-S transponders:
https://github.com/piratfm/eti-tools#dab-ensembles-working-with-eti-tools

For sure, a DAB ensemble isn't a MPEG-TS bitstream. However, the inner bitstream format here isn't the question (in the sense of the approach used to do the encapsulation). Similary other tools encapsulate multiple MPEG-TS CBR bitstreams over one big MPEG-TS. And they do it without the standard T2-MI specification but with this _brute force_ "format".

Futhermore, I feel it's more simple the solution of encapsulating with the sync byte. But, this is only my opinion. You're an experimented developer and your vision is more global. If you prefer to implement a more robust format, then perhaps it's the best option. For my user case, I only need to encapsultate/decapsulate; so the format isn't relevant. I only suggest the simpler because I see it in other implementations. No more, no less. :smile:

lelegard, 2018-09-21:
Hi @lars18th,

I just came up with another concern about your use case : using ffmpeg to process the stream with everything else in a private PID containing an encapsulation of the rest of the TS.

I suppose that you request some A/V processing or reencoding from ffmpeg in the remaining service. The number of output packets for the transformed A/V stream will likely be different from the input, maybe quite different if the bitrates are significantly different.

My concern is : how will ffmpeg re-mux the private PID with the new A/V streams? How are they synchronized? There is a risk that the decapsulated stream will be completely out of sync with the transformed A/V stream.

We could add an option to add PCR's in the new outer PID and specify an existing service (the one you transform using ffmpeg) or PID containing PCR's and synchronize the new PCR's in the new PID with the existing service. Thus, we _may hope_ (no guarantee) that ffmpeg will synchronize the private PID with the transformed service. But no guarantee.

Before implementing the plugins, I would recommend that you test the synchronization of a large private PID with a service which is transformed by ffmpeg. I would not like to implement something which may happen to be useless because of a deficient transport stream processing by ffmpeg.

lars18th, 2018-09-25:
Hi @lelegard ,

I agree with your concern to test before doing any implementation!

As for my user case, take note about these facts:

- The idea is to encapsulate _all_ data of _other_ services, including all original PSI/SI tables. So, in this case, no synchronization with the encapsulated data is required (as far as I know).
- The ffmpeg tool does a pure remux of the private PIDs as _data services_. So, it only copies each incoming packet to the output when it arrives.
- Futhermore, the ffmpeg process generates new PAT/PMT tables, and the PID numbers of A/V streams are changed. So, an additional task is restore the original tables and renumber the outcoming pids.

Then my concern is about another problem: How to maintain the original bitrate. My idea is to encapsulate the null packets too, and after the ffmpeg process do some cleanning. This can be easy with CBR A/V streams, but nearly impossible with VBR.

In any case, incorporating a PCR in the outer stream can be a very useful functionality. Almost as an _option_. I'll run some tests and publish the results here.

Thank you for your interest in (crazy) user requests! :wink:

lelegard, 2018-10-07:
Added plugins `encap` and `decap`. Please check. See the user's guide 5.2.25 for an example.

lars18th, 2018-10-18:
Hi @lelegard ,

I'm testing it now.
I'll post results soon.
Thank you!

lars18th, 2018-10-18:
Hi @lelegard ,

It will be complex to support inside the `encap` plugin the same syntax of the `remap`?
The idea is to support ranges, like in: `tsp -P encap -o 8189 -p 0-31 -p 500-8186`

I think it's a must have!
Regards.

lars18th, 2018-10-18:
And another suggestion, if you feel it useful:

I need to process some TS that's is (already) fully encapsulated inside a pid in a MPTS.
So, it will be possible to add the option `--sync` to the `decap` plugin to support the case (assuming no errors) of the 0x74 sync byte inside the encapsulated (inner) stream?

I suggest this only for convenience. :wink:

lars18th, 2018-10-19:
Hi @lelegard ,

> I need to process some TS that's is (already) fully encapsulated inside a pid in a MPTS.
So, it will be possible to add the option --sync to the decap plugin to support the case (assuming no errors) of the 0x74 sync byte inside the encapsulated (inner) stream?

Please, forgive this suggestion. It doesn't have sense as the decapsulated stream is a _new_ stream. So it can't be done inside a TS processing. The only valid solution is a new utility (like `tsdecap`) for extracting the inner TS from the outher TS, So this utility will output only packets from the inner TS.

Feel free to save this idea for a future utility. But discard it for any `encap` enhancement.

I try to close the testing today. However, please consider to add the range support in the parameters of the filter.

Regards.

lelegard, 2018-10-19:
Yes, you are right, the PID range should have been implemented in the first place. Done in commit dd1a2617f921542bd01e5eab3724728af7d0e4b4.

lelegard, 2018-11-02:
Version 3.15 was released with the two plugins. Closing this issue. Please open a new one in case of specific problem with these plugins.

closed 2018-11-02 by lelegard