tsduck/tsduck #127: PCR continous playout

joolzg, 2018-09-25:
Im playing a 60 seconds file out with the --loop option, but i need to pcr to increase  and not start again when the file loops.

Ive looked in the docs but cant find any command that would make the Xnd loop of the file keep an increasing pcr.

Any ideas

lelegard, 2018-09-25:
There is currently no plugin the sole purpose of which would be recomputing PCR's.

Some plugins such as `merge` restamp PCR's because they slightly change the alignment of packets in the stream and the PCR's need some adjustment. But this is different. The clock is still valid and the position of the packets changes.

In your case, because of the wrap up, the clock is performing some time warp. However, changing the PCR's _only_ will be worse because the PTS and DTS will become completely out of sync. The result is that the audio / video will no longer play at all (believe me, I tried).

So, the PTS and DTS of all PID's which are somehow associated with the PCR's shall be adjusted by exactly the same amount. This is what the plugin `rmsplice` does, for instance, after removing a video segment (splice out/in sequence).

In the general case, the following should be done:

- Analyse the complete service map (PAT and all PMT's) to locate all services which are synchronised on the PCR PID to restamp. Note that some streams have shared PCR PID's. This is why all services must be inspected.
- Recompute all PTS and DTS inside all the identified services.
- But stop doing anything if any service is scrambled at some point.

The last point is required because PCR's are always in the clear (in the adaptation fields) while PTS and DTS are scrambled (in the PES headers). So, the PTS and DTS cannot be adjusted in scrambled services. In that case, modifying the PCR's would make the service unplayable after descrambling.

So, in the general case, recomputing PCR's just because of a time warp is not as easy as it seems. Probably a suggestion for a future new plugin, but with limitations.

joolzg, 2018-09-25:
well if you get a chance it would be great as the only way i know to do this is to run it though ffmpeg which is a pain as the output is not very nice.

rezna11, 2018-10-22:
I think this would be nice. 
But just FYI, if you need this functionality and if you have a dektec card with corresponding license, then you can try Streamxpress, it has some options exactly for this purpose:

[image]

rezna11, 2018-10-22:
and there's also something in DVSStation4 which seems to be free, but when I tried to play the output in VLC, I saw errors in the picture during the loop so I'm not sure if this works - anyway it would be nice to have such feature in the universal tsduck...

[image]

cbernander, 2018-11-13:
I understand that it would take some effort to write the code needed to handle the general case, but I think it would be very useful to (in the mean time) have a simpler plugin implementation where the following assumptions are made:

- There is only one PCR present in the stream
- There are no scrambled services in the stream

Since I suspect this plugin would mostly be used in the case where a (preprocessed) ts input file (containing a single program) is being looped, the initial version could require the offset time to be specified by the user, so that every time the PCR/PTS/DTS wraps around, an incremented multiple of the specified offset is added. That could not only simplify the implementation somewhat, but also serve as a reminder that the plugin is not yet ready for the general case.

I think such an implementation would be sufficient for a lot (if not most) of actual use, and it could still be used as a foundation to build upon and improve later.

lelegard, 2019-01-22:
The plugin `pcradjust` has been implemented. I have opened issue #201 by mistake instead of replying to this one. Sorry about the double issue.