tsduck/tsduck #129: Building without `curl` and `pcsclite` libraries

deech, 2018-09-25:
Hi,
I'd like to disable `curl` and `pcsclite` from being linked, I tried building with `CFLAGS="-DTS_NO_PCSC=1 -DTS_NO_CURL=1 -DTS_NO_DTAPI=1" NOTEST=1 make static` but it still fails with:
```
  [DEP] tstabdump.cpp
  [CXX] tsanalyze.cpp
  [LD] release-x86_64-static/tsanalyze
/usr/bin/ld: cannot find -lcurl
/usr/bin/ld: cannot find -lpcsclite
collect2: error: ld returned 1 exit status
make[3]: *** [../../Makefile.common:910: release-x86_64-static/tsanalyze] Error 1
```
Any help is appreciated.

lelegard, 2018-09-28:
Hi @deech,

The various symbols such as `TS_NO_CURL` were automatically defined by the makefile on platforms which do not support the corresponding libraries. These symbols only affect the C++ source, not the build command in the makefile. So, in your commands, the references to the libraries were actually removed but the link commands were not modified. In your error message, the libraries for curl and pcsclite were not used. However, their simple presence in the command line was a problem because the libraries were not found.

I just modified the makefiles so that you may request the references to the corresponding libraries to be cleanly removed, even on platforms which support them.

With the latest commit, this command works on Linux (tested on Fedora 28):
~~~
make static NOTEST=1 NOPCSC=1 NOCURL=1 NODTAPI=1
~~~

Please close this issue if this works for you.

deech, 2018-10-01:
This works great! Thanks!

closed 2018-10-01 by deech