tsduck/tsduck #132: New plugin to save native teletext pages

nightwatcher74, 2018-10-05:
Hello,

is it possible to get a plugin to save teletext pages?
A few older software packages to save teletext exists but no none of these is able to do this from an IP (unicast or multicast) source.
I need to save one or multiple pages (inclusive subpages) from a ts (filtered mux) so it would be perfect to set the pages as cli parameters, for example 111,112,201-210,310
No conversion is needed. I'm able to do this with a script because i need the full ttxt page with graphic and color commands.
The absolutly best would be that the page will only saved as a new file if something in the page was changed.

Thanks a lot
Frank

lelegard, 2018-10-05:
Hi @nightwatcher74,

The current `teletext` plugin only saves Teletext subtitles in SubRip (SRT) format.

I have no plan to implement more Teletext features because Teletext is a complex binary format and it takes a lot of time to implement such features.

I keep this issue open as a reminder for a hypothetical future enhancement if a developer would like to contribute with such a plugin. But I won't do it myself because of lack of time, sorry.

nightwatcher74, 2018-10-05:
If someone can/will help me... there is a small tool who will do this (for all pages from a ts/configured ts from a mux) and with a hardcoded export dir... but this tool will only work with /dev/dvb....
https://github.com/nxmirrors/dvbtools/tree/master/dvbtext
I need this as plugin for tsduck and the directory and exportd pages can be configured. 

Thanks a lot
Frank

premultiply, 2019-08-30:
Should be very easy to implement on base of the current srt-subtitle-plugin code.
Just throw away all the text filtering and srt output and write the raw page data from PES for each page to a separate page-numbered binary file.

SonnyWalkman, 2023-07-16:
@premultiply,
I'm hoping one day TSDuck to support Teletext data both decode and encode natively.  The vbit2 does all of it in code, just need a method or plugin to inject the stream via UDP into the TS using the video PCR or muxing a complete TS manufactured with PCR and merge the PES as it arrives.

closed 2026-04-23 by github-actions[bot]