tsduck/tsduck #133: Filter MPE packets of specified size

dukey914, 2018-10-07:
Hello, I would like to ask if is able to add some small filter to tsp plugin "mpe" for filtering packets of specified size. For example... with "--log" I see packets with exact size and it would be awesome, if there can be some filter for filtering only packets of specified size (example size 1468 bytes).

Thank you very much for incredible work!!

dukey914

lelegard, 2018-10-08:
Hi @dukey914 

Options `--udp-size`, `--min-udp-size`, `--max-udp-size`, `--net-size`, `--min-net-size`, `--max-net-size` are added to plugin `mpe` in commit 61871ebad728f13e0f4b1102e5eb57de2dc97547 (version 3.15-947).

dukey914, 2018-10-08:
Working perfectly!!! Exactly what I wished!!! Thank you @lelegard !!! Awesome work!!!

closed 2018-10-09 by lelegard