tsduck/tsduck #134: Can't build on CentOS 7

rezna11, 2018-10-13:
OK, so I'm probably missing something trivial but I'm fighting this for 3 days now so I'd like to ask for help. I once managed to build tsduck and Dektec with no issue and then I failed with DVB tuners. So I tried again, installing new CentOS, upgraded kernel from 3.10 to 4.18 and was able to install DVB tuner but building tsduck just didn't work. So again I made a new CentOS install and I wanted to start with TSDuck, then Dektec and then upgrade to kernel 4.18 and install DVB tuner(s). But I'm getting stuck on step 1 - tsduck.

I installed CentOS 7 Minimal from a freshly downloaded ISO.
Then I installed yum group install "Development Tools".
Then I cloned tsduck from git.
Then I installed dependencies.
Then I ran make but it fails with these errors:

```
[root@localhost tsduck]# make
  [CXX] tsApplicationRecordingDescriptor.cpp
tsApplicationRecordingDescriptor.cpp: In member function 'virtual void ts::ApplicationRecordingDescriptor::deserialize(const ts::Descriptor&, const ts::DVBCharset* ':
tsApplicationRecordingDescriptor.cpp:135:52: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
         _is_valid = size >= 1 && size >= data[0] + 2;
                                                    ^
tsApplicationRecordingDescriptor.cpp:146:61: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
     _is_valid = _is_valid && size >= 1 && size >= 1 + data[0];
                                                             ^
tsApplicationRecordingDescriptor.cpp:155:61: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
     _is_valid = _is_valid && size >= 1 && size >= 1 + data[0];
                                                             ^
tsApplicationRecordingDescriptor.cpp: In static member function 'static void ts::ApplicationRecordingDescriptor::DisplayDescriptor(ts::TablesDisplay&, ts::DID, const uint8_t*, size_t, int, ts::TID, ts::PDS)':
tsApplicationRecordingDescriptor.cpp:197:52: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
             valid = size >= 1 && size >= data[0] + 2;
                                                    ^
tsApplicationRecordingDescriptor.cpp:210:53: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
     valid = valid && size >= 1 && size >= 1 + data[0];
                                                     ^
tsApplicationRecordingDescriptor.cpp:222:53: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
     valid = valid && size >= 1 && size >= 1 + data[0];
                                                     ^
cc1plus: all warnings being treated as errors
make[2]: *** [release-x86_64/tsApplicationRecordingDescriptor.o] Error 1
[root@localhost tsduck]#
```
gcc version is gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-28)
I tried some other versions (5, 7) but that didn't help much. So before messing up the system even more - anyone knows what's tsduck missing in my system right now? Why isn't it built?

rezna11, 2018-10-13:
I was able to build after removing -Werror from Makefile.common.
Not sure why I was able to compile in other CentOS installations and not in this Centos Minimal ISO without removing the flag but at least it worked out in the end. So if noone knows what's the reason for all this, we can close this issue.

lelegard, 2018-10-13:
Hi @rezna11,

This certainly not normal. TSDuck is meant to build on the latest version of CentOS since this is the only "production-certified" system for many companies.

My development policy is very strict to chase potential bugs in advance. So, I always activate the most paranoid set of warnings for the each compiler and turn them into errors. I never let warnings of any kind in the code.

Currently, I use three compilers: clang (LLVM) on macOS, gcc on Linux and MSVC on Windows. They have different marginal warning checks and can spot different issues. I can see that your warnings occur on a source file I added last night (so, you must have cloned the latest repository snapshot). As of today, it has been built on macOS only (which is my main development platform). It is possible that additional warnings can be triggered with another compiler. A file is fully certified only when it has been successfully tested on all platforms. I will check. Please retry after the next commit.

Sorry for the inconvenience. This can happen with fresh code snapshots.

lelegard, 2018-10-13:
> Not sure why I was able to compile in other CentOS installations and not in this Centos Minimal ISO without removing the flag but at least it worked out in the end. 

It worked because you cloned the repository before last night, before I added a new file that was tested on macOS only.

rezna11, 2018-10-13:
Great, it all makes sense. Thanks for looking into this.
By the way, CentOS, after updating kernel to 4.18, is so far the best linux I've ever tried... in just 3 days I was able to make it work with 4 different DVB tuners (DVB Sky S960, TBS 5922, 5925, Technotrend TT-4100) at the same time, together with TSDuck and Dektec (I tested DTA-111 and DTA-140).  I was never able to do this on Debian or Ubuntu. I'm not sure if upgrade to 4.18 was absolutely necessary but on 3.10, I could not build TBS drivers.

lelegard, 2018-10-13:
CentOS + kernel 4.18 ?
You should try Fedora, this is RedHat with bleeding edge techno and pre-built integration (no need to mess with kernel versions).

closed 2018-10-13 by lelegard