tsduck/tsduck #135: XML escape characters

marcino, 2018-10-14:
Hi,

I noticed that the created XML files (for example by the tables plugin) do not have escaping characters for special characters used in the XML specification. Parsing such files by external tools raises an error.

For example - not escaped & char:
`<service_descriptor service_type="0x02" service_provider_name="nc+" service_name="RIRE & CHANSONS"/>`

List of XML escape characters: [https://www.w3.org/TR/REC-xml/#dt-escape](https://www.w3.org/TR/REC-xml/#dt-escape)

lelegard, 2018-10-14:
Hi @marcino 

Thank you for pointing this. This is a bug of course. In attribute values, the quotes `'` and `"` were properly escaped but not the ampersand `&`, which is ridiculous since the escaped characters `&apos;` and `&quot` use the ampersand in the escape sequence.

I just fixed this.

However, I think some trade-off is needed. Let me explain this and please provide your feedback.

Originally, the only escaped characters were `'` and `"` in attributes and `<` and `>` in text nodes. This is the minimum required to make the syntax consistent (plus the `&` which was forgotten and was a bug).

Looking into the XML specification, it seems that the 5 characters `<>&'"` should be escaped everywhere, including attribute values and text nodes. I started to fix it this way. But, running and fixing the complete TSDuck test suite after the bug fix, I realised that many text nodes in EIT and some HbbTV descriptors became completely unreadable because they were cluttered with many `&apos;` and some `&quot;`. But, while escaping `<` and `>` in text nodes is syntactically required, escaping `'` and `"` in text nodes is rather pedantic.

So, in the current fix, I came up with the following trade-off:

- In attribute values, escape `<>&'"`
- In text nodes, escape `<>&` only

Although escaping `<>` in attribute values does not seem syntactically necessary, I have read on the net that some XML analysers required it. I wonder if such difficulties exist with unescaped `'"` in text nodes. If it is confirmed that not escaping `'"` in text nodes break well-known tools, then I will also fix this. But the price to pay for the human reader seems too high to do this in the first place.

Please let us know what you think about this.

marcino, 2018-10-19:
I tested your patch. I used the C, Perl and Python libraries for different sets of data. Everything works correctly.
However, in my opinion, you should strictly stick to the XML documentation - maybe you should think about using an additional argument, e.g. --strict-xml?

lelegard, 2018-10-23:
Hi @marcino 

You are rigth, the option `--strict-xml` is the solution. It has been added in commit 98fa2a8ebb1c2a31d6cd59a90d8621c494e04e84 and applies to `tstables`, `tstabcomp` and plugin `tables`.

closed 2018-10-23 by lelegard