tsduck/tsduck #136: Simulcrypt question - where do I get SCG?

rezna11, 2018-10-15:
Hello. I wanted to find out a bit more about how simulcrypt works so I wanted to use TSDuck as the CA system but now I'm not quite sure how to make this work.

For this test, I'm using a multiplexer which allows me to configure SCS, EMMg, ECMg, CWG, EIS and PSIG.
So I configured EMMg and ECMg to use TSDuck's `tsemmg` and `tsemmg` - connection between those and the multiplexer is established.

Then I set a service to be scrambled using CSA Simulcrypt. But it's not scrambled. The multiplexer complains that a SCG is missing. I probably understand that SCG is normally created by EIS. But I don't have any EIS. 

So my question is - is this something TSDuck could also do (with or without a new plugin)? Or do you know about a (free/demo/trial) application that could act as EIS and supply SCG for a simple test, just to make the multiplexer happy?

Thanks

lelegard, 2018-10-15:
:warning: :skull_and_crossbones: :warning:  :skull_and_crossbones: :warning:  :skull_and_crossbones: :warning: 
>  I wanted to use TSDuck as the CA system 

:warning: :skull_and_crossbones: :warning:  :skull_and_crossbones: :warning:  :skull_and_crossbones: :warning: 

Danger here ! TSDuck is not a CA system. There is zero security. The control words are in the clear in the pseudo / fake ECM's. The command `tsecmg` and the generic `descrambler` plugin are meant to provide software stubs only. They are functional and can be used to test, debug and demonstrate end-to-end integration (they were developed for this specific need) or for educational purpose, but __NEVER__ for any kind of production, even a non-critical one.

Now, concerning your specific need, an SCG is a Scrambling Control Group, a set of PID's which must be scrambled using the same CW at the same time. In practice, an SCG is built per service and contains the video PID and all audio PID's of this service. SCG are provisioned by the EIS but they reference service ids or component ids which are defined in the SCS (which is part of the MUX system).

The definition and management of these ids are highly MUX dependent. If your MUX/SCS system is flexible enough, there is probably some mechanism to define static SCG's inside the SCS, without an EIS. You should probably investigate this.

The `tsemmg` command was developed to test and debug the EMMG/PDG<=>MUX integration. It is useless in your case since there is no need for EMM's to make your system work. EMM's are meant to transport subscription rights and hierarchies of encryption keys which, ultimately, allow the decryption of ECM's. But since ECM's are in the clear here, there is no need for EMM's.

Again, beware, TSDuck is an integration toolbox, not a CA system!

lelegard, 2018-10-15:
The relevant standard is [available here](https://www.etsi.org/deliver/etsi_ts/103100_103199/103197/01.05.01_60/ts_103197v010501p.pdf). It used to be my Bible, 10 years ago...

rezna11, 2018-10-15:
Of course I realize that TSDuck is not a real CA system but I'm not running a real TV station so I thought TSDuck would be perfect tool for testing a multiplexer which has the ability to scramble the output. So I wanted to see my own ECM and EMM in the stream, see how the communication works and how the EMMs are multiplexed. 

Unfortunately I don't think I can add SCG manually. Do you think, with your experience, should I try to develop a simple EIS tool to feed a (hopefully simple) command to the multiplexer with an SCG? If I understand the standard right, the EIS<>SCG interface is not proprietary, right?

As for EMMs, I think that I'm just doing something wrong and giving the manuals proper reading might help. Since I see the `tsemmg` connection in the multiplexer, yet I don't see any EMM streams, I think it's just configuration issue and hopefully I'm not missing any other compoment (like the EIS with EMMs)

lelegard, 2018-10-16:
> see how the communication works 

You can use option `--log-protocol` in `tsecmg` and `tsemmg`. You can also use Wireshark which has a dissector named SIMULCRYPT to properly analyze all DVB SimulCrypt messages.

> and how the EMMs are multiplexed.

Inside an EMM PID, the individual EMM's are multiplexed by the EMMG, according to each CA system proprietary rules. The stream of EMM's is passed directly from the EMMG into the TS. In fact, the protocol is named EMMG/PDG<=>MUX, where PDG means "Private Data Generator" because this interface can be used to inject any kind of unformatted data into a PID, not even sections if you use the packet mode.

Each EMM PID is simply declared in a CA_descriptor in the CAT.  That's all. 

> If I understand the standard right, the EIS<>SCG interface is not proprietary, right?

Yes, it is fully documented in chapter 10 of the [DVB SimulCrypt standard](https://www.etsi.org/deliver/etsi_ts/103100_103199/103197/01.05.01_60/ts_103197v010501p.pdf). The basic TLV syntax and the channel/stream principles are identical to other protocols such as ECMG<=>SCS and EMMG/PDG<=>MUX. Only the parameter semantics is specific to the protocol.

> Do you think, with your experience, should I try to develop a simple EIS tool to feed a (hopefully simple) command to the multiplexer with an SCG? 

Developing a simple EIS can be somewhat complicated if you have never done that. If you only need to declare a static SCG and repeat that command each time you need, there is probably a simpler way without programming. Just use a shell script and `ncat`. Build the three EIS<=>SCS commands _channel_setup_, _SCG_provision_ and _channel_close_ (I think they are sufficient). Build them in hexadecimal in a text file using the spec as reference. In your script, convert them in binary using `xxd -r -p` and send them to the SCS using `ncat`. Use Wireshark and its SIMULCRYPT dissector to debug the dialog.

This is brutal and primitive but probably efficient.

> As for EMMs, I think that I'm just doing something wrong and giving the manuals proper reading might help. Since I see the tsemmg connection in the multiplexer, yet I don't see any EMM streams, I think it's just configuration issue and hopefully I'm not missing any other compoment (like the EIS with ECMs)

EMM injection is independent of everything else. `tsemmg` alone is sufficient to inject fake EMM's using various parameters and bitrates. I have done this many times to stress MUXes. See its reference in the TSDuck user's guide.

However, there is usually some specific MUX configuration. At the MUX level, you need to configure an EMM stream with a `data_stream_id` and map it to a PID. This `data_stream_id` shall be specified using the option `--stream-id` in `tsemmg`. This is the way the MUX will know where to send the injected stream. This configuration is MUX-specific, usually through a configuration GUI or low-level commands (or both).

rezna11, 2018-10-17:
Thanks for your help.

I was able to send commands that open channel and send an SCG to the multiplexer, the command content I used was pretty much inspired by wireshark's example.

provision SCG:
```
0x03, 0x04, 0x08, 0x00, 0x45, 0x00, 0x01, 0x00, 0x02, 0x00, 0x01, 0x00, 0x06, 0x00, 0x02, 0x00, 0x01, 0x00, 0x0f, 0x00, 0x02, 0x00, 0x65, 0x00, 0x16, 0x00, 0x02, 0x00, 0x12, 0x00, 0x07, 0x00, 0x04, 0x00, 0x00, 0x00, 0x01, 0x00, 0x0e, 0x00, 0x02, 0x00, 0x67, 0x00, 0x05, 0x00, 0x1b, 0x00, 0x09, 0x00, 0x02, 0x1A, 0x0A, 0x00, 0x08, 0x00, 0x04, 0x01, 0x00, 0x01, 0x00, 0x00, 0x0a, 0x00, 0x04, 0x10, 0x20, 0x30, 0x40, 0x00, 0x10, 0x00, 0x01, 0x01
			_LEN  _EIS_channel_ID___________________  _ECM_Group________________________  _transport_stream_ID______________  _network_ID_______________________  _SCG_reference_ID_____________________________  _service_id_______________________  _ECM_Group____________//_ECM_ID___________________________  _Super_CAS_ID_________________________________  _access_criteria______________________________  _AC_changed_flag____________
```

then the SCG for this channel was visible in the multiplexer which immediately started talking to `tsecmg` and scrambling the channel and tsduck's ECMs were automatically muxed in. 

`tsemmg` also works without issues, thanks for pointing out the importance of `data_stream_id` here...

So I'm happy and closing this question :)

closed 2018-10-17 by rezna11