tsduck/tsduck #137: Some enhancemets (scrambler/descrambler, mpe)

dukey914, 2018-10-16:
Hello. I have some questions for enhancements.

 1. I would like to ask, it there could be scrambler/descrambler with XOR algorithm. I mean... I would like to make some stream, encrypt it with static 184 bytes long XOR key and on another device decrypt it with the same key. When Adaptation Field is present, leave all 184 bytes unencrypted. I know that XOR can be easily hacked, but it's experimental (if it's not hard to add this command).

 2. Can MPE plugin has output for filtered packets throught IP? For now, we can filter MPE packets and save it only to the file with `-o` argument. It would be awesome, if that output can be pushed to IP output. I mean, something like `-O ip (address)`, but when I do that in this case, it will be streaming all MPE packets without any filtering. And last question...

 3. `--min-udp-size` and `--max-udp-size` works perfectly, but I discovered, that sometimes they are sending some garbage text informations in that range. See this example:
 
```
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 12 bytes, fragment: 0x0000	/ garbage text
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 32 bytes, fragment: 0x0000	/ MPEG-TS packets
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 102 bytes, fragment: 0x0000	/ garbage text
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 144 bytes, fragment: 0x0000	/ MPEG-TS packets
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 184 bytes, fragment: 0x0000	/ MPEG-TS packets
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 212 bytes, fragment: 0x0000	/ garbage text
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 216 bytes, fragment: 0x0000	/ MPEG-TS packets
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 336 bytes, fragment: 0x0000	/ garbage text
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 352 bytes, fragment: 0x0000	/ MPEG-TS packets
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 465 bytes, fragment: 0x0000	/ garbage text
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 466 bytes, fragment: 0x0000	/ garbage text
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 467 bytes, fragment: 0x0000	/ garbage text
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 477 bytes, fragment: 0x0000	/ garbage text
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 478 bytes, fragment: 0x0000	/ garbage text
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 480 bytes, fragment: 0x0000	/ MPEG-TS packets
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 652 bytes, fragment: 0x0000	/ MPEG-TS packets
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 728 bytes, fragment: 0x0000	/ MPEG-TS packets
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 840 bytes, fragment: 0x0000	/ MPEG-TS packets
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 1188 bytes, fragment: 0x0000	/ MPEG-TS packets
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 1424 bytes, fragment: 0x0000	/ MPEG-TS packets
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 1460 bytes, fragment: 0x0000	/ MPEG-TS packets
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 1464 bytes, fragment: 0x0000	/ MPEG-TS packets
* mpe: PID 0x1389 (5001), src: 192.168.xxx.xxx:4000, dest: 224.xx.xx.10:4000 (xx:xx:xx:xx:xx:xx), 1468 bytes, fragment: 0x0000	/ MPEG-TS packets
```

So for now, I'm using command with `--min-udp-size 480 --max-udp-size 1468`, but sometimes, I will miss some parts of MPEG-TS packets, due to that packets are smaller than 480 bytes. After long analysing, they are not sending packets bigger than 1468 bytes. If there can be some argument for dropping packets with specific size, it would be awesome, too!! :) For this example something like this: `--drop-udp-sizes 12,102,212,336,465,466,467,477,478`.
Probably, they are using variable sizes for MPEG-TS packets, but fixed sizes for text messages.

Thank you so so much for your powerful TSDuck !!! You're incredible @legard !!!

dukey914

PS: When will be released new binaries?

lelegard, 2018-10-16:
Hi,

1. Your request for an XOR plugin is very specific and experimental for you only. Too specific for a general case. Moreover, and you mentioned it, using XOR for encryption is a joke. I mean literally a joke. In the crypto community, "use an XOR" is the usual joke to say "this is crap". I cannot spend time developing code for one person only if it does not benefit to the community or increase the value of the project. But, you could develop it yourself, it is quite easy.

2. Did you try `-P mpe --udp-forward --redirect address:port`? 

3. The trailing text should be an hexa + ascii dump of the packet. If you think it is incorrect, please post an example stream to reproduce the problem. Only the MPE PID, only a few packets would be sufficient.

New binaries for version 3.15 will be published when the new `tsswitch` utility is sufficiently tested by users, as discussed in issue #73. If you need the latest development, you can build it yourself, see the [instructions here](https://tsduck.io/doxy/building.html).

dukey914, 2018-10-16:
OK, I understand. Thanks for reply anyway! :) I will try `-P mpe --udp-forward --redirect address:port`.

dukey914, 2018-10-16:
Looks like `-P mpe --udp-forward --redirect address:port` works, but (I didn't mentioned that.. sorry) I have to use `--skip 20` argument for droping MPE header, which works only for `-o` output.
`[80 21 FA C5 45 DC 0C F4 28 28 6C 00 01 03 40 40 00 00 00 00] 47 40 00 39 A6 00 FF FF FF FF FF FF FF...`

dukey914, 2018-10-16:
...so if I'm not lost, `mpe` successfully decapsulates UDP from MPE and that header `80 21 ... 03 40 40 00 00 00 00` has nothing in common with MPE - It's UDP header? And... Exists some way how to drop it (live)?

lelegard, 2018-10-17:
Look at the MPE spec in chapter 7 of the [DVB data broadcast spec](https://www.etsi.org/deliver/etsi_en/301100_301199/301192/01.06.01_60/en_301192v010601p.pdf).

- TS packets of the MPE PID contain DSM-CC sections
- DSM-CC sections contain a network datagram, including (in the case of UDP datagram):
-- MAC address info (part of the Ethernet frame header)
-- IP header
-- UDP header
-- UDP payload

You should make a step backward, ie. stop thinking in terms of bytes but in terms of features. With `--udp-forward`, the `mpe` plugin forwards the UDP datagrams on the local network using the original destination IP address (usually a multicast one) from the IP header in the MPE section. Thus, the broadcast network acts as a VPN tunnel for IP multicast. With `--redirect`, it also modifies the destination IP address and UDP port, in case the original multicast address is no longer suitable on the local network or to transform multicast in unicast or vice-versa.

In all cases, the net result of the UDP forwarding is that you receive the original UDP messages when you listen to the UDP port on the local network.

dukey914, 2018-10-17:
OK, Thanks for explanation @lelegard !!
For now, I'm using this way:
`-P mpe --pid 2009 -d "224.20.20.54:6000" --min-udp-size 480 --max-udp-size 1468 --skip 20 -o "tmp01.bin"` and using `ffmpeg` just after I run this command for live streaming `tmp01.bin` throught udp (or hls), but I have to restart `tsp` after some hours, because `tmp01.bin` should mess my HDD.
I just looked for some way, how to not mess my HDD (without restarting `tsp`). How to stream that `-o` output without saving it to HDD. Maybe it's not possible. I understand. :)

lelegard, 2018-10-17:
> How to stream that -o output without saving it to HDD. 

That is exactly what I was trying to explain. Simply replace `-o` with `--udp-forward`.

dukey914, 2018-10-17:
`--skip 20` do the trick, but it works only with `-o`. When I do it with `--udp-forward`, looks like streaming works perfectly, but VLC (or another player) doesn't play it correctly, because they're broadcasting it with that header, which `--skip` skips, so the picture is clear (only 188 bytes long MPEG-TS packets), but when I do it with `--udp-forward` (even with `--skip 20`), `--udp-forward` will not skip that extra bytes, so it will not stream clear 188 bytes packets.

See example:
[image]

`-o` do the trick, but `--udp-forward` not.

lelegard, 2018-10-17:
Again, you should think in terms of formats instead of bytes. What is the formal format of the UDP datagrams from your MPE stream? TS encapsulation inside UDP can be raw (no header), RTP, etc. So, having a header inside the UDP payload before the first TS packet is not unusual.

dukey914, 2018-10-17:
OK. I'm sorry, if I was saying some nonsense. I really enjoy your app, because it's really powerful and can do a lot of things. Looks like TSDuck do everything correctly, but they're just broadcasting it a little bit modified. Later I will post some short TS record of one PID with MPE. Thanks for your efforts. I hope it wasn't waste of time with me.
See you for now @lelegard !!

rezna11, 2018-12-09:
> * Your request for an XOR plugin is very specific and experimental for you only. Too specific for a general case. Moreover, and you mentioned it, using XOR for encryption is a joke. I mean literally [...]

Sorry to bring this old issue up again but I'd just like to point out that XOR is actually used, it seems that Compel and Rosscrypt CA systems use XOR so implementing it in TSDuck would actually allow hackers to gain access to those few channels using this CA system so it's better for TSDuck not to implement this to stay out of trouble.

dukey914, 2018-12-09:
OK. Yes, I noticed that info about Rosscrypt and Compel, too. Now I
understand. Thanks for reply anyway.

See you.

PS: I'm still using MPE plugin. :)

lelegard, 2018-12-09:
> it seems that Compel and Rosscrypt CA systems use XOR

I do not know these CA systems but I seriously doubt about XOR being used as content or key "encryption". This would be less than weak or vulnerable. I would simply call it "over stupid". They would be well-known in the crypto-community, precisely because of using XOR as pseudo-encryption.

These two CA systems are not well referenced on the Web.

- Compel: a [PDF document](http://wegener.com/datasheets/DS_COMPELCAa.pdf) from 2004 mentions _"RSA Data Security has audited COMPEL CA."_. Audited XOR ? No way...
- Rosscrypt: the [Wikipedia page on CA systems](https://en.wikipedia.org/wiki/Conditional_access) mentions that Roscrypt (with one 's') has been introduced in 2006. A [Web archive page](https://web.archive.org/web/20110726100247/http://www.niir.ru/eng/page.php?trid=96) from a Russian federal institute mentions _"Modern encryption methods, openness of the system for appropriate Russian specialists"_. This, in itself, does not mean anything valuable. But Russia is probably in the top 5 countries for cryptography. So, using XOR is not credible at all.

Now, "encrypting" a 184-byte message with a 184-byte key looks like stream cipher or OTP. But, in that case, the security is in the generation of this key stream. And being a "key stream", not a single reusable key, it changes every message.

rezna11, 2018-12-09:
I don't know if the system really is Compel, it's just what Lyngsat says for those 2 horse betting channels that use XOR as encryption with the XOR key broadcast in clear in the video PID. There is no CAT, no ECM and no EMM, so who knows, might be something else.

closed 2019-01-14 by lelegard