tsduck/tsduck #138: How to "copy" the content of a pid?

lars18th, 2018-10-18:
Hi,

I'm impressed to not found any plugin for "copying" packets from one pid!

Using the current `encap` plugin I like to "store" a _copy_ of the PSI tables in some private pid. However, I don't like to discard the current PSI tables, as the external processor (FFmpeg) needs to read these tables.

So, how to achieve this?
Perhaps de current `remap` plugin can include a new parameter `--copy` to create simple copy of packets with the new pid number.

What you think?

lelegard, 2018-10-19:
The principle of `tsp` is to _modify_ packets. The plugin `encap` modifies input packets by replacing them with the outer packets (with a slight overhead requiring 1 null packets every 45 input packets).

Duplicating PID's is possible with one condition and one consequence:
- There must be at least as many null packets as input packets to duplicate.
- The packets to duplicate will be buffered until null packets are available, which will introduce some time shift / desync between the original PID's and their copies.

lars18th, 2018-10-19:
Hi @lelegard ,

I assume the limitations... And I agree with the two restrictions. I them in the `encap` plugin (the buffer with default 1024 bytes long, and the "shift" when not using the PCR).

But the idea is to use it for a simple copy of the PSI tables. Then the bitrate is low, and the shift is not a huge problem.

So, a dumb "copy" for `remap`... you feel it has sense?

It's a simple question, not a request.

lars18th, 2018-10-19:
Hi @lelegard ,

In order to complete the test of the #126  I think I need to fully "copy" the PSI pids, because I don't know how I can store the original PSI tables inside the encapsulted stream.

I see the example in the documentation (**5.2.25 Encapsulating PID’s into a private tunnel**). But the problem is that the external tool (like the FFmpeg does) can "destroy" the original PSI tables. So the idea is to "restore" them _after_ the processing.

How I can achieve this without doing a "copy"?
I wait for your response.

lelegard, 2018-10-19:
OK, added plugin `duplicate`.

closed 2018-10-23 by lelegard