tsduck/tsduck #139: svremove plugin optimization

lars18th, 2018-10-19:
Hi @lelegard ,

I comment this only for convenience.
Please *only* consider it if it's more efficient than a simple repetition of the plugin.

For the `svremove` plugin I suggest these to enhancements (a list of services, and the not operator):

- `tsp -P svremove 100,200,300,400`: This indicates to remove four services at the same time.
- `tsp -P svremove !100`: This instructs to remove _all_ services _except_ the service 100.
  
I feel it's best to do the work in one plugin instance, than repeat it a lot of times for the same objective (more resource intensive).

What you think?

lelegard, 2018-10-19:
Hi, @lars18th,

As you have experienced multiple times, the project is open to suggestions and enhancements. But this time, the answer is a firm "no" because this is against the basic principles of TSDuck: Keep It Simple and Stupid (KISS), implement basic and simple features, provide the ability to combine them in an infinite number of ways. 

If something can be done by using a plugin twice, I do not want to make the plugin more complex by modifying it to do it in one pass. Because if we do this once, we will have to do it everywhere. And the code will become more complex, which means more error-prone, more bugs, spending more time to maintain it. If TSDuck is so reliable after 10 years of maintenance and evolution, it is mainly because of such principles.

I accept to modify a plugin only when something cannot be done in two separate instances, when it needs to be done in once. 

The best example is the good old `grep` command. What do you do it you want to search for FOO _and_ BAR vs. FOO _or_ BAR? The _or_ must be implemented in the command: `grep -e FOO -e BAR` but the _and_ can be achieved with two instances: `grep FOO | grep BAR`. This is why the popular and universal `grep` command never implemented an _and_ in more than 40 years of existence. I apply the same principle.

lelegard, 2018-10-19:
And I forgot to address your second suggestion: `tsp -P svremove !100` is already implemented as `tsp -P zap 100`.

lars18th, 2018-10-19:
Hi @lelegard ,

After think more, I completly agree! Thank you for the good response, and for the tip of `tsp -P zap 100`.

I close this issue.

lars18th, 2019-06-19:
Hi @lelegard ,

I accept your well based opinion, and I don't want to resurrect an issue without a cause. But I have a problem:

- I need to remove _all_ services from a TS _except_ a short list (two or three). And I need to put it inside one script to process a list of TS that have an undefined number of services inside it. Any idea to do this?
```
tsp -P svremove --negate 100,200,300
```
In this example all services are removed _except_ 100 and 200 and 300.

Please, help me as I don't know how to do it with the `zap` plugin as I need to preserve all three.
Thank you!

lars18th, 2019-06-21:
Hi @lelegard ,

> In this example ALL services are removed _except_ 100 and 200 and 300.

Any idea to do that?
Please comment something even if your opinion remains a firm "no".

lars18th, 2019-06-24:
Hi @lelegard ,

I understand your opinion, but please comment about my request of:

> In this example ALL services are removed except 100 and 200 and 300.

This have sense with the example described here: https://github.com/tsduck/tsduck/pull/331#issuecomment-504969384

I really need a command to "remove _all_ services except a list". I understand that for 1 service the current implementation of the `zap` plugin can be used. But when we want to maintain 2 or more services and discard the rest, without explicitly listing them because we don't know what they are; as far as I know, there is no option to do so.

Any ideas? Please!

lars18th, 2019-06-25:
Hi @lelegard ,

The syntax that I suggest is this: `-P svremove --except 100,200,300 0`
- The service `0` is a special id with the sense "all".
- The option `--except` is the list of services to not remove, and it implies to use the service=0 (all).

I hope you reconsider your decision.
Almost please, comment about it.
Thank you!

closed 2026-04-23 by github-actions[bot]