tsduck/tsduck #141: tsp not seeing/getting incoming multicast packets

manuelmann, 2018-10-19:
Hi,

I try to join a SSM stream,
```
tsp -v -I ip 10.15.97.125@232.100.142.1:5000 -P count  -O drop
* ip: joining multicast group 10.15.97.125@232.100.142.1:5000 from local address 10.252.179.57
* ip: joining multicast group 10.15.97.125@232.100.142.1:5000 from local address 10.252.180.57
```
or
```
tsp -v -I ip 10.15.97.125@232.100.142.1:5000 -P analyze -O drop
* ip: joining multicast group 10.15.97.125@232.100.142.1:5000 from local address 10.252.179.57
* ip: joining multicast group 10.15.97.125@232.100.142.1:5000 from local address 10.252.180.57
```

With iftop or tcpdump I can see that packets of the stream are coming in.
But tsp is some how not recognizing it. The streams runs till the router send a query. tsp is not answering so the stream stops.

I tried it on 

Ubuntu 18.10
```
uname -a
Linux check-igmp5 4.18.0-10-generic #11-Ubuntu SMP Thu Oct 11 15:13:55 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.14-894
```

and on Fedora 28
```
uname -a
Linux localhost.localdomain 4.17.4-200.fc28.x86_64 #1 SMP Tue Jul 3 14:06:39 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.14-894
```
Is there something I should do differently? Or is there something wrong?

lelegard, 2018-10-19:
What do your UDP packets contain ? TS packets ?

The first question is : does tsp receive or not the packets ? It may receive them and not recognise or accept the content. Try `tsp --debug=2`. It is very verbose but it should display a message each time a UDP packet is received. With these messages, tsp receive the packet but does not like the content for some reason. Without these messages, tsp does not receive the packets.

lelegard, 2018-10-19:
Strange, support for SSM was added in version 3.13 after a user's request (issue #84). The user reported success on Windows 10 and Ubuntu 18.04.

The problem is that I do not have the infrastructure to reproduce the problem at home. Could you suggest a simple way to generate SSM traffic to test here?

manuelmann, 2018-10-19:
I think generating a SSM Stream is not the problem, the problem is more you need a routing devices which supports that. Otherwise you will get the MC Group with or without a specific source. Because a sending device always sends out the multicast packets with it's ip or a vsip to the multicast group. But only if the router is configured to ssm the source address is relevant.
Do you have access to a managed Layer3-Switch or similar?
If not I have to think about how this could be simulated...

manuelmann, 2018-10-19:
Could it help if I give you a capture of the stream as a TS file?
And do you think this really related to the stream itself? Because igmp joins are working and stream is coming in, which is half of the rent. Couldn't it be that there is something related to the OS or that I use a VLAN-Interface or I don't know 🤔, just guessing?

yaneti, 2018-10-19:
Sounds like you have more than one interface on your machine. Have you turned rp_filter off?
``` # echo 0 | tee `find /proc/sys/net/ -name rp_filter` ```

manuelmann, 2018-10-19:
That's it!!
Thank you very much!!
Now it works.
I never had a problem with that filter. I need to learn more about it.
@lelegard So no problems with your SSM implementation.
tsp looks like a great piece of software! Now I can start testing.
Thanks both of you for your quick help!

closed 2018-10-19 by manuelmann