tsduck/tsduck #142: plugin analyze: simple improvements

lars18th, 2018-10-19:
Hi @lelegard ,

I know the relevance of the KISS model development for the TSDuck. Please, I like only to contribute with my troubles when using this amazing project. Forget this if you don't agree!

Regarding the `analyze` plugin, I found it very useful when doing tests... as I can see if the processing is doing right or not. However, I found two *simple* missing options:

1. Decimal output: with a simple `-P analyze --decimal` the numbers of PIDs can be translated to decimal values (with four digits). I request this because several other tools use decimals instead of hexa (that I prefer, seriously!). So, doing manual translation is a pain.
2. Total of PSI tables without padding: At time the summary is like this:
```
===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 6,427,621, PID's: 8 (clear: 8, scrambled: 0)                   |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C   15,428,731 b/s  |
|  0x0000  PAT .......................................... C        3,168 b/s  |
|  0x0001  CAT .......................................... C        3,168 b/s  |
|  0x0010  DVB-NIT ...................................... C          151 b/s  |
|  0x0011  SDT/BAT ...................................... C          760 b/s  |
|  0x0012  EIT .......................................... C       70,170 b/s  |
|  0x0014  TDT/TOT ...................................... C          100 b/s  |
|  0x0015  Network Synchronization ...................... C        2,467 b/s  |
|  0x1FFF  Stuffing ..................................... C   15,348,744 b/s  |
|=============================================================================|
```
So, I suggest to add before the Stuffing line a new line for "Total PSI data (without stuffing)" like this:
```
|  0x0015  Network Synchronization ...................... C        2,467 b/s  |
|   Total  PSI data (without NULLs)...................... C      132,619 b/s  |
|  0x1FFF  Stuffing ..................................... C   15,348,744 b/s  |
```

I feel this missing data is relevant (almost for someone that likes to encapsulate all PSI data).

What you think?

lelegard, 2018-10-19:
You may also try `analyze --wide`. The display is wider and there is room for the decimal values.

lars18th, 2018-10-20:
> You may also try analyze --wide. The display is wider and there is room for the decimal values.

Interesting! Great output. Thank you! :+1: 

However, why this `--wide` is inside `tsanalyze` tool and not in the plugin, and not documentend?

closed 2018-10-21 by lelegard

lars18th, 2018-10-21:
Hi @lelegard ,

Only for reference (when searching in the issues list)...
The PR #143 implements the suggestion. :wink: